Koenkk/zigbee2mqtt #4839: Zemismart Tubular motor not recognized

rodrigorsdev, 2020-11-01:
The Zemismart Tubular ZM25TQ engine is available on compatible devices, but is not recognized.

I get the message:

Zigbee2MQTT:warn  2020-11-01 12:56:05: Received message from unsupported device with Zigbee model 'TS0601'
Zigbee2MQTT:warn  2020-11-01 12:56:05: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

Koenkk, 2020-11-02:
Can you share the `data/database.db` entry of this device?

rodrigorsdev, 2020-11-04:
this is it?

{"id":16,"type":"Router","ieeeAddr":"0x5c0272fffe8a5a2b","nwkAddr":23355,"manufId":4098,"manufName":"_TZE200_fzo2pocs","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,10,4,5,61184],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"��\u0004'\u0013\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0005\u0000\u0000\u0000\u0000\u0012\u0000\u0000\u0000\u0000\u0013","65506":21,"modelId":"TS0601","manufacturerName":"_TZE200_fzo2pocs","powerSource":1,"zclVersion":3,"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"meta":{}}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1604340041164}

Koenkk, 2020-11-04:
Added, please confirm.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

rodrigorsdev, 2020-12-05:
How can I test de dev version on Hassio?

Koenkk, 2020-12-06:
@rodrigorsdev use the edge version of the addon.

rodrigorsdev, 2020-12-07:
It worked, thanks

closed 2020-12-07 by rodrigorsdev