Koenkk/zigbee2mqtt #4845: Add Ikea Tradfri E27 600lm bulb (warm)white channel support

werra11, 2020-11-02:
## Is your feature request related to a problem? Please describe
The Ikea Tradfri E27 600lm rgb & white bulb can display Warm White. Sadly zigbee2mqtt doesn't support adjusting the white channel on the bulb, only the RGB channels.

## Describe the solution you'd like
The ability to change the (warm)white channel of the bulb

## Describe alternatives you've considered
.

## Additional context
The bulb supports white spectrum according to IKEA: https://www.ikea.com/nl/en/p/tradfri-led-bulb-e27-600-lumen-wireless-dimmable-colour-and-white-spectrum-colour-and-white-spectrum-globe-opal-white-00408612/ & switches to the "Warm White" color when you cut & restore power to the bulb. zigbee2mqtt then display's the color before it turned off.

Koenkk, 2020-11-02:
I'm not sure if this device supports color temperature. Last time I checked it didn't but maybe something changed with recent firmware updates of the bulb. 

You can try by changing https://github.com/Koenkk/zigbee-herdsman-converters/blob/fd76352313e5f255c52949284c2dfbce2d554d61/devices.js#L2412 to `extend: generic.light_onoff_brightness_colortemp_colorxy,` and see if it reacts when sending to `zigbee2mqtt/DEVICE_FRIENDLY_NAME/set` payload `{"color_temp": 250}`

ademobile, 2020-11-03:
Just tried with mine and they did not react to colour temperature changes. I just use shades of RGB white, which I think the real IKEA app does

werra11, 2020-11-03:
> I'm not sure if this device supports color temperature. Last time I checked it didn't but maybe something changed with recent firmware updates of the bulb.
>
> You can try by changing [...]
Sadly I'm not home for the coming days. I will try this as soon as I get home!

> Just tried with mine and they did not react to colour temperature changes. I just use shades of RGB white, which I think the real IKEA app does
Very strange, my bulb has a very clear warm white (when it boots up & after factory reset) and there is no way this is produced by RGB values only. If that were the case I would be able to set it to that color using rgb values but I can;t, no way to reproduce it.

LubosBa, 2020-11-08:
This device definitely supports white channel. After I migrated to zigbee2mqtt from the original Ikea gateway, I can't get the white colors or brightness to to the same level as before. There are multiple discussions around the internet about this issue.

Is there a way for me to test the above with the z2m installed inside HA Supervisor?

PS: The same also applies to the E14 RGB Ikea bulb, that one also doesn't seem to show white channel at all.

E14/E27 model = LED1624G9

Thank you very much.

ammmze, 2020-12-06:
@LubosBa any luck with this? Also, did you intentionally have 2 `l` in the name and flip the `e` and `i`? `zigbee2mqtt/bedroom_cielling_bulb_1` vs `zigbee2mqtt/bedroom_ceiling_bulb_1`. Or was that just a typo and perhaps reason why it didn't respond?

ammmze, 2020-12-17:
Just tested it myself and yea changing the color temp as no effect. 😞

But I did another test where I power cycled the light, which defaults to a nice white color. I went to zigbee2mqtt web ui and hit the refresh button to get the current color xy values, then I set those same xy values. In theory it should remain the same, however the light changed slightly on the first call to set the xy values to their current value. The difference was quite subtle though. I would probably only notice at the time the update happened because its like the leds turn off for a split second then turn back on.

ov1d1u, 2020-12-17:
For those interested, the only way I managed to set the bulb at its full brightness was to reset it (power cycling it might be enough, I can't test this right now) and, after repairing it without doing any color or brightness change, by storing the current scene using the steps described [here](https://www.zigbee2mqtt.io/information/scenes.html). Then, when you want to set the bulb at its full brightness, just recall the scene and you're done.

biau, 2020-12-29:
I've tried to group the bulb with a [https://www.zigbee2mqtt.io/devices/L1528.html#ikea-l1528](url) and it looks like it doesn't work.
But when I remove the group it looks like the bulb sync with  the panel and receive the full brightness and warm color.
I've never seen my bulb with a brightness and warm color like that.
As soon as I made a change in z2m the color come back to a ugly yellow.

LubosBa, 2020-12-29:
No luck on my side, edited the file as per the instruction however the color temperature slider doesn't work. It really looks like the bulb uses a combination of all RGB LEDs rather than a dedicated white LED as they advertise ;)

What I managed to find out is what the others have found out already. It seems that there is a specific color combination which has higher brightness than the standard RGB color when set. So it replicates the brightness/color of white or opal.

In case of "warm white" these are the numbers I have in the state which produce noticeably brighter light.
```
{
    "brightness": 253,
    "color": {
        "x": 0.42,
        "y": 0.411
    },
    "color_temp": 250,
    "linkquality": 166,
    "state": "ON",
    "update": {
        "state": "idle"
    },
    "update_available": false
}
```

MattWestb, 2020-12-29:
Its no need cutting it open because its one RGBW then looking on the FCC certificate for the [LED1624G9](https://fccid.io/FHO-LED1624G9/Internal-Photos/Internal-Photos-3631433.html) but its having one older [zigbee module with external antenna](https://fccid.io/FHO-LED1624G9/Internal-Photos/Internal-Photos-3631436) and RGBW outputs for the led drives.

But if you do im interesting of dump of the flash region 0 and 1 for converting other modules to IKEA RGBW.

sjorge, 2020-12-30:
I looked into this a while ago too, I did notice at startup the colorMode attribute is sometimes reported.

It was set to color_temperature mode, but I’ve never been able to get the bulb in that mode after. There are other things we can try though. E.g. my innr bulbs are brighter when using hue/sat that with xy, perhaps those bulb also take hue/sat.

biau, 2020-12-30:
This is the attributes reported when I group/ungroup the bulb with my float panel:
{
    "brightness": 254,
    "color": {
        "x": 0.458,
        "y": 0.41
    },
    "linkquality": 0,
    "state": "ON",
    "update": {
        "state": "idle"
    },
    "update_available": false
}

R 255 G 232 B 123
H 50 S 100% L 74%
[image]

Then if I change something I can't go back to these values no matter I try to change RGB or HSL:
{
    "brightness": 254,
    "color": {
        "x": 0.38028169014084506,
        "y": 0.4460093896713615
    },
    "linkquality": 0,
    "state": "ON",
    "update": {
        "state": "idle"
    },
    "update_available": false
}
[image]

Davst, 2021-01-10:
We had a discussion on this bulb and I did some experimentation with it that I described in a comment at: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4140#issuecomment-757396662

Maybe that can help you guys as well.

MattWestb, 2021-01-10:
@Davst "Hello Norrköping Vienna is calling".
Sorry but i cant commenting your post in deCONZ git then as payed user of their product im being blocked for using it (company policy ?).
I have doing some sniffing and IKEA is setting up the 3 "default scene" with deCONZ and if you is adding one extra groupe with number 65289 and adding scenes to it they is working with the remote ala ZLL = also then deCONZ is offline :-)))
[image]
The scenes you is added in deCONZ is absolute (its changing all parameters not only the color because the IKEA special commands is not sent by deCONZ then setting them up).
The 3 default scene is restored on firmware update and you can force deCONZ doing on firmware downgrade and then upgrading the bulb and it should adding the scenes (I have not tested on my RGBW but its working on newer GU10WS). 
IKEA is also adding them then pairing one bulb and the value you can finding here: https://github.com/zigpy/zha-device-handlers/issues/561#issuecomment-723041108
Thes always adding scenes with xy and reading them back for verifying that its correct setted up.
I dont knowing if they is setting up the same values for for the RGBW as the WS (the sniffs is made with new version of GU10 WS).

One more thing then using deCONZ for working with light its only the old "Wireless Light Control" that is working well and doing the right things.

The old RGBW is as you have saying one RGBW and the firmware can simulating some TC that is working if doing it right as you have showing and you have doing good findings !!

Mvh  MW

closed 2021-02-18 by github-actions[bot]

GSzabados, 2021-03-14:
@Koenkk, I came across this issue meanwhile searching for a solution how to set the bulb to warm white. 

When I ask Google to set the light to Warm White it does react but sets the light to "x": 0.5, "y":0.389 which is close to 2200K.

The default Warm White on the bulb is "x": 0.458,
"y": 0.41 (confirmed multiple ways), which is close to 2700K.

I cannot find that where the command is get scrambled from Warm White to 2200K. Is it Z2M which does it incorrectly or the Google Integration from HA.

Indeed any other bulb which supports mired ends up being around 2700K when asked to be Warm White.

sjorge, 2021-03-14:
If you enable debug logging, it should show up in the log:
```
Zigbee2MQTT:debug 2021-03-14 14:05:44: Received MQTT message on 'zigbee2mqtt/office/light/mood/set' with data '{"state":"OFF"}'
```

e.g. when I turn on a bulb via homekit, when the color is set you should see the same and can see what color value gets passed to z2m.

GSzabados, 2021-03-14:
> If you enable debug logging, it should show up in the log:
> 

Ok, I did. This is what I got:

`Debug Received MQTT message on 'zigbee2mqtt/Small Room 2/set' with data '{"state": "ON", "color": {"x": 0.5, "y": 0.389}, "brightness": 254}'`

And 

`Debug Received MQTT message on 'zigbee2mqtt/Desk Lamp/set' with data '{"state": "ON", "color_temp": 333}'`

It must be some issue at HA level, and not with Z2M.

sjorge, 2021-03-14:
z2m just sets what it gets send, so in this case its probably whatever is sending it that is causing the diff.

Xenomes, 2021-05-25:
I don't know if this idea is already was mentioned. I understand that there is no color_temp but can there be a tabel of the 255 steps translation kelvin to colorXY values? When send x 0.45986 y 0.41060 to the blub the startup color is almost correct working old and new firmware. I used this site for translation https://www.waveformlighting.com/tech/calculate-cie-1931-xy-coordinates-from-cct

Xenomes, 2021-05-27:
I see that there is a zigbee-herdsman-converters/lib/kelvinToXy.js tabel can it used in this command?
extend: extend.light_onoff_brightness_colortemp({colorTempRange: [250, 454]}),

Xenomes, 2021-05-31:
@Koenkk is it possible to built a colortempkelvin, I was trying to create some code but got not further then this not working code for the toZigbee.js.
```
    light_colortempkelvin: {
        key: ['color_temp_kelvin'],
        convertSet: async (entity, key, value, meta) => {
            const preset = {'warmest': 1800, 'warm': 2400, 'neutral': 2700, 'cool': 3800, 'coolest': 6500};

            if (key === 'color_temp_percent') {
                value = utils.mapNumberRange(value,
                    0, 100,
                    (colorTempMin : 1800), (colorTempMax : 6500),
                ).toString();
            }

            const result = await converters.gledopto_light_colortemp.convertSet(entity, key, value, meta);
                result.state.color = libColor.ColorXY.fromMireds(value).rounded(4).toObject();
                return result;
        },
        convertGet: async (entity, key, meta) => {
            return await converters.light_color_colortemp.convertGet(entity, key, meta);
        },
    },
```
I has also edit the ikea and extend file, i am not a great JavaScript coder 😅