Koenkk/zigbee2mqtt #4846: Securifi Peanut Plug OTA not working yet

airdrummingfool, 2020-11-02:
Hi Koen, the 1.16.0 release notes mention that OTA is now supported on the Peanut Plug. I've tested OTA on two Peanut Plugs (one updated, and one that needs an update) and both fail with the following error:
```
ERROR: Failed to check if update available for '[Peanut Plug]' (Device didn't respond to OTA request)
```

I believe this is because, as mentioned in @walthowd's https://github.com/Koenkk/zigbee-OTA/pull/14, specific, non-standard behavior is required to trigger the OTA update for Peanut plugs:
> To initiate the OTA transfer you have to write attribute 2 (groups) to cluster 5 (scenes) with the value 49502

It seems like this behavior might be able to be added by using a manufacturer-specific OTA file (instead of using `ota.zigbeeOTA`) in `zigbee-herdsman-converters`... I took a look, but I'm not familiar enough with the OTA files or device communication to be able to crank out a PR at this point.

### Debug info
Zigbee2MQTT version: `1.16.0-dev` (ca53013)
Adapter hardware: CC2531

Koenkk, 2020-11-02:
I think we need to add an additional toZigbee converter which does the write attribute 2 (groups) to cluster 5 (scenes) with the value 49502, afterwards the user immediately has to request an ota.

airdrummingfool, 2020-12-03:
This hasn't been implemented yet. (commenting to ward off the stale bot)

airdrummingfool, 2021-01-04:
Unstale

deiger, 2021-01-04:
Is there a way to do this manually?

Koenkk, 2021-01-04:
@airdrummingfool can you experiment with adding a converter for this? in toZigbee.js add:

```js
securifi_ota: {
    key: ['securifi_ota'],
    convertSet: async (entity, key, value, meta) => {
        await entity.write('genScenes', {currentGroup: 49502});
    },
},
```

And add `tz.securifi_ota` to the `toZigbee` list of the device in `devices.js`.

Now you can trigger this converter by sending to `zigbee2mqtt/DEVICE_ID/set` payload `{"securifi_ota": ""}`. I'm not sure what the device will do after that. Maybe you can send https://www.zigbee2mqtt.io/information/ota_updates.html#update-to-latest-firmware after that

airdrummingfool, 2021-01-05:
@Koenkk that worked! I sent the `securifi_ota` message and then immediately requested an update check in the frontend, and it immediately returned that there was an update.

I'd like to do some additional testing before actually running the update, but I can't seem to get z2m (or the Peanut Plug) to "forget" that an OTA update is available. All I could think to do was to unplug/plug in the Peanut Plug and manually change `update_available` and `update:state` in `state.json` while z2m was stopped, but that didn't work. Is there a way to clear the OTA cache or something like that?

Koenkk, 2021-01-07:
Closing this as fix got merged.

closed 2021-01-07 by Koenkk

mdegat01, 2021-01-11:
Does the fix here mean that you no longer need the almond router to update the firmware of the Peanut plugs and can get energy monitoring working just by connecting it to Z2M?

deiger, 2021-01-11:
Yes, it does.