Koenkk/zigbee2mqtt #4848: Sonoff SNZB-03 is identified as SNZB-02

oh2ftu, 2020-11-02:
## What happened
Using HA addon with zigbee2mqtt 1.16.0, a previously paired (and newly paired) Sonoff SNZB-03 is identified as SNZB-02
## What did you expect to happen
I expected it to be detected is SNZB-03.

## How to reproduce it (minimal and precise)
Pairing and thusly fails.

## Debug info
Zigbee2MQTT version: 1.16.0
Adapter hardware: CC2531
Adapter firmware version:unknown
`2020-11-02 15:06:48: MQTT publish: topic 'homeassistant/sensor/0x00124b002248xxx/linkquality/config', payload '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/0x00124b002248xxx/availability"}],"device":{"identifiers":["zigbee2mqtt_0x00124b002248xxx"],"manufacturer":"SONOFF","model":"Temperature and humidity sensor (SNZB-02)","name":"0x00124b00224xxx","sw_version":"Zigbee2MQTT 1.16.0"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/0x00124b002248xxx","name":"0x00124b002248xxxx_linkquality","state_topic":"zigbee2mqtt/0x00124b002248xxxx","unique_id":"0x00124b002248xxxx_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'`

Koenkk, 2020-11-02:
Can you share the `data/database.db` entry of this device?

oh2ftu, 2020-11-02:
This is the one
`{"id":44,"type":"EndDevice","ieeeAddr":"0x00124b002248ba95","nwkAddr":41571,"manufId":0,"manufName":"eWeLink","powerSource":"Battery","modelId":"66666","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1280,1],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"66666","manufacturerName":"eWeLink","powerSource":3,"zclVersion":1,"appVersion":3,"hwVersion":1,"dateCode":"20191107"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00124b0014d967b2","zoneState":1}},"genPowerCfg":{"attributes":{"batteryPercentageRemaining":200}}},"binds":[{"cluster":1026,"type":"endpoint","deviceIeeeAddress":"0x00124b0014d967b2","endpointID":1},{"cluster":1029,"type":"endpoint","deviceIeeeAddress":"0x00124b0014d967b2","endpointID":1},{"cluster":1,"type":"endpoint","deviceIeeeAddress":"0x00124b0014d967b2","endpointID":1}],"meta":{}}},"appVersion":3,"hwVersion":1,"dateCode":"20191107","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1604335668822}`

Koenkk, 2020-11-03:
Fixed.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-11-03 by Koenkk

merloose, 2020-11-19:
Hi all, Is it implemented on the 1.16.1 version ?
I have the same issue there.

Koenkk, 2020-11-20:
@merloose check latest dev branch first (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html), if that doesn't help please post your `data/database.db` entry of your device here

merloose, 2020-11-20:
@Koenkk I did install the dev branch and it fully solves the issue ! Many Thanks !

jamiekris, 2020-11-20:
Is there any idea when this change will be brought into the head/master? I have the same issue, and the procedure to switch to the edge (on hass.io) seems to require some reconfiguration, and opens up update problems too.

Koenkk, 2020-11-20:
@jamiekris 1 december (every 1st of the month)

jamiekris, 2020-11-21:
Thanks! I try to wait it out then :+1:

sporniak, 2020-12-03:
Have zigbee2mqtt in 1.16.1 version on my HA and my SNZB-03 is still recognized as SNZB-02. 

`{"id":13,"type":"EndDevice","ieeeAddr":"0x00124b002247b4f9","nwkAddr":62391,"manufId":0,"manufName":"eWeLink","powerSource":"Battery","modelId":"66666","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":1026,"inClusterList":[0,3,1280,1],"outClusterList":[3],"clusters":{"genBasic":{"attributes":{"modelId":"66666","manufacturerName":"eWeLink","powerSource":3,"zclVersion":1,"appVersion":3,"hwVersion":1,"dateCode":"20191107"}},"ssIasZone":{"attributes":{"iasCieAddr":"0x00212effff053d8b","zoneState":1}}},"binds":[],"meta":{}}},"appVersion":3,"hwVersion":1,"dateCode":"20191107","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1607000513896}`

jamiekris, 2020-12-03:
@sporniak The issue was only fixed in 1.16.2.

sporniak, 2020-12-03:
@jamiekris just new update is available for me now :D received it few minutes ago
[image]

BCsabaEngine, 2020-12-07:
With 1.16.1 the device recognized as snzb-03: eWeLink Sonoff SNZB-03 Battery