Koenkk/zigbee2mqtt #4849: IKEA TRADFRI stopped working as routers with the recent Zigbee2MQTT updates

joaocmfcm, 2020-11-02:
## What happened
TRADFRI plug and TRADFRI light bulb started becoming really unstable (95% of the times they don't turn on/off) and they don't act as router anymore. 
## What did you expect to happen
I expected them to work as they've been working flawlessly for more than a year. I didn't add or changed anything in my network for months besides the ```new_api: true```.  
## How to reproduce it (minimal and precise)
My network used to make use of those 3 routers (hallway_plug, balcony_light, living_room_plug) but now all the devices connect to the coordinator making the network really slow and unresponsive. This all started recently with recent zigbee2mqtt updates.

Also, I don't know why, but on the map there are some devices that seem to not being connect to anything. Despite that, they respond to commands and work (slower than before).
[image]

My configuration is currently:
```data_path: /share/zigbee2mqtt
external_converters: []
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://homeassistant'
  user: ************
  password: ***********
serial:
  port: /dev/ttyACM0
advanced:
  homeassistant_discovery_topic: homeassistant
  homeassistant_status_topic: hass/status
  pan_id: 6754
  channel: 11
  network_key:
    - 1
    - 3
    - 5
    - 7
    - 9
    - 11
    - 13
    - 15
    - 0
    - 2
    - 4
    - 6
    - 8
    - 10
    - 12
    - 13
  availability_blocklist: []
  availability_passlist: []
device_options: {}
blocklist: []
passlist: []
queue: {}
frontend:
  port: 8099
experimental:
  new_api: true
socat:
  enabled: false
  master: 'pty,raw,echo=0,link=/dev/ttyZ2M,mode=777'
  slave: 'tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5'
  restartdelay: 1
  initialdelay: 1
  options: '-d -d'
  log: false
ban: []
whitelist: []
cache_state: true
```
## Debug info
Zigbee2MQTT version: 1.16.0
Adapter hardware: CC2531
Adapter firmware version: 

Upon doing a network scan to generate the network map I get this on the logs: 
```
Zigbee2MQTT:info  2020-11-02 15:48:31: Starting network scan (includeRoutes 'false')
Zigbee2MQTT:error 2020-11-02 15:48:41: Failed to execute LQI for 'balcony_light'
Zigbee2MQTT:error 2020-11-02 15:48:51: Failed to execute LQI for 'living_room_plug'
Zigbee2MQTT:error 2020-11-02 15:49:01: Failed to execute LQI for 'hallway_plug'
Zigbee2MQTT:info  2020-11-02 15:49:01: Network scan finished```

Almost every time I get this errors when setting commands to any of these routers: 
```
Zigbee2MQTT:error 2020-11-02 15:52:20: Publish 'set' 'state' to 'balcony_light' failed: 'Error: Command 0xccccccfffecf9274/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SREQ '--> AF - dataRequest - {"dstaddr":50727,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":17,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,7,0]}}' failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))'``` 

What can I do to make everything work again?

Thanks in advance

Koenkk, 2020-11-02:
`(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))'` is a potential indication that your CC2531 firmware got corrupted (unfortunately not uncommon for a CC2531). I recommend to reflash with the following firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

itavero, 2020-11-02:
For security reasons, it's better _not_ to share your network key online.

netweaver1970, 2020-11-02:
I'm having same issue, totally erratic behavior all of a sudden today. 
My only change was the upgrade of my LAUNCHXL-CC26X2R1 to the just released  Z-Stack_3.x.0 20201026 firmware. 
Before I was on a march-april or so firmware. For Z2M, I was always on the Dev release, updated 1-2 / week, never really had serious issues. I also did that this morning. 

If no visibility on a fix, I'll try to downgrade again to a way older release, as the WAF is below zero for the moment.

The issue didn't happen immediately, at least I didn't notice it immediately as the light are not often on during daytime. For sure by the evening things went 'very funny'. I thought I was getting crazy. 

Almost looks like the Zigbee network is flooded. 
No WIFI channel changes happened. I'm still on channel 6 for 2.4GHz. My Zigbee channel is 13. Always been. 
I did a WIFI signal scan and all neighbors, on WIFI channel 1 are on very low signal strength, as they were before.

This is part of the normal Z2M log, sending an ON state to the EntranceHall group.
EntranceHall is a Zigbee group with 2 lightbulbs : IKEA_Bulb_hall1 and IKEA_Bulb_hall2

```
Zigbee2MQTT:info  2020-11-02 20:02:54: MQTT publish: topic 'zigbee2mqtt/EntranceHall', payload '{"brightness":253,"color":{"x":0.380438429420364,"y":0.376746069841299},"color_temp":150,"state":"ON"}'
Zigbee2MQTT:info  2020-11-02 20:02:54: MQTT publish: topic 'zigbee2mqtt/IKEA_Bulb_hall1', payload '{"brightness":253,"color":{"x":0.380438429420364,"y":0.376746069841299},"color_temp":150,"last_seen":"2020-11-02T20:02:54+00:00","linkquality":144,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-11-02 20:02:54: MQTT publish: topic 'zigbee2mqtt/IKEA_Bulb_hall1', payload '{"brightness":253,"color":{"x":0.380438429420364,"y":0.376746069841299},"color_temp":150,"last_seen":"2020-11-02T20:02:54+00:00","linkquality":144,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-11-02 20:02:54: MQTT publish: topic 'zigbee2mqtt/IKEA_Bulb_hall1', payload '{"brightness":253,"color":{"x":0.380438429420364,"y":0.376746069841299},"color_temp":150,"last_seen":"2020-11-02T20:02:54+00:00","linkquality":144,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-11-02 20:02:54: MQTT publish: topic 'zigbee2mqtt/IKEA_Bulb_hall1', payload '{"brightness":253,"color":{"x":0.380438429420364,"y":0.376746069841299},"color_temp":150,"last_seen":"2020-11-02T20:02:54+00:00","linkquality":144,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-11-02 20:02:54: MQTT publish: topic 'zigbee2mqtt/IKEA_Bulb_hall1', payload '{"brightness":253,"color":{"x":0.380438429420364,"y":0.376746069841299},"color_temp":150,"last_seen":"2020-11-02T20:02:54+00:00","linkquality":144,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-11-02 20:02:54: MQTT publish: topic 'zigbee2mqtt/IKEA_Bulb_hall1', payload '{"brightness":253,"color":{"x":0.380438429420364,"y":0.376746069841299},"color_temp":150,"last_seen":"2020-11-02T20:02:54+00:00","linkquality":144,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-11-02 20:02:54: MQTT publish: topic 'zigbee2mqtt/IKEA_Bulb_hall1', payload '{"brightness":253,"color":{"x":0.380438429420364,"y":0.376746069841299},"color_temp":150,"last_seen":"2020-11-02T20:02:54+00:00","linkquality":144,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-11-02 20:02:54: MQTT publish: topic 'zigbee2mqtt/IKEA_Bulb_hall1', payload '{"brightness":253,"color":{"x":0.380438429420364,"y":0.376746069841299},"color_temp":150,"last_seen":"2020-11-02T20:02:54+00:00","linkquality":144,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-11-02 20:02:54: MQTT publish: topic 'zigbee2mqtt/IKEA_Bulb_hall1', payload '{"brightness":253,"color":{"x":0.380438429420364,"y":0.376746069841299},"color_temp":150,"last_seen":"2020-11-02T20:02:54+00:00","linkquality":144,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-11-02 20:02:54: MQTT publish: topic 'zigbee2mqtt/IKEA_Bulb_hall1', payload '{"brightness":253,"color":{"x":0.380438429420364,"y":0.376746069841299},"color_temp":150,"last_seen":"2020-11-02T20:02:54+00:00","linkquality":144,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-11-02 20:02:55: MQTT publish: topic 'zigbee2mqtt/IKEA_Bulb_hall1', payload '{"brightness":253,"color":{"x":0.380438429420364,"y":0.376746069841299},"color_temp":150,"last_seen":"2020-11-02T20:02:55+00:00","linkquality":144,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-11-02 20:02:55: MQTT publish: topic 'zigbee2mqtt/IKEA_Bulb_hall1', payload '{"brightness":253,"color":{"x":0.380438429420364,"y":0.376746069841299},"color_temp":150,"last_seen":"2020-11-02T20:02:55+00:00","linkquality":144,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-11-02 20:02:57: MQTT publish: topic 'zigbee2mqtt/Aqara PIR LivingRoom', payload '{"battery":100,"elapsed":10001,"illuminance":0,"illuminance_lux":0,"last_seen":"2020-11-02T20:02:47+00:00","linkquality":144,"occupancy":false,"voltage":3005}'
[... cut ...]
Zigbee2MQTT:info  2020-11-02 20:03:27: MQTT publish: topic 'zigbee2mqtt/IKEA_Bulb_hall2', payload '{"brightness":253,"color":{"x":0.380438429420364,"y":0.376746069841299},"color_temp":150,"last_seen":"2020-11-02T20:03:27+00:00","linkquality":57,"state":"ON","update":{"state":"idle"},"update_available":false}'
Zigbee2MQTT:info  2020-11-02 20:03:27: MQTT publish: topic 'zigbee2mqtt/IKEA_Bulb_hall2', payload '{"brightness":253,"color":{"x":0.380438429420364,"y":0.376746069841299},"color_temp":150,"last_seen":"2020-11-02T20:03:27+00:00","linkquality":57,"state":"ON","update":{"state":"idle"},"update_available":false}'
```

Koenkk, 2020-11-03:
@netweaver1970 does downgrading the firmware fix the issue?

tomdee, 2020-11-03:
I'm also seeing that my network has stopped working. I'm also using only Tradfri devices and a CC2531

I'm seeing similar errors to the original issue raiser.

```
Zigbee2MQTT:error 2020-11-03 09:56:31: Publish 'set' 'state' to '0x680ae2fffea21dff' failed: 'Error: Command 0x680ae2fffea21dff/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC no ack' (233))'
Zigbee2MQTT:info  2020-11-03 09:56:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x680ae2fffea21dff' failed: 'Error: Command 0x680ae2fffea21dff/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'MAC no ack' (233))'","meta":{"friendly_name":"0x680ae2fffea21dff"},"type":"zigbee_publish_error"}'
Zigbee2MQTT:error 2020-11-03 09:56:31: Publish 'set' 'state' to '0x680ae2fffea45ee5' failed: 'Error: Command 0x680ae2fffea45ee5/1 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC no ack' (233))'
Zigbee2MQTT:info  2020-11-03 09:56:31: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'state' to '0x680ae2fffea45ee5' failed: 'Error: Command 0x680ae2fffea45ee5/1 genOnOff.off({}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'MAC no ack' (233))'","meta":{"friendly_name":"0x680ae2fffea45ee5"},"type":"zigbee_publish_error"}'
```

And these during a network scan
```
Zigbee2MQTT:error 2020-11-03 09:54:12: Failed to execute LQI for '0x680ae2fffea45ee5'
Zigbee2MQTT:error 2020-11-03 09:54:22: Failed to execute LQI for '0x680ae2fffea21dff'
```

lvesterg, 2020-11-03:
Hi

I have the same erratic behaviour - the IKEA bulbs are sometimes reacting, but usually not. And when do it takes up to 10 seconds.
Also with the same kind of log entries as in the previous posts. 

I use slaesh’s cc2652 stick with the following firmware 

> Zigbee2MQTT:info  2020-11-03 19:22:55: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20200925,"transportrev":2},"type":"zStack3x0"}'

I just recently installed the zigbee network and the bulbs, but it was working very well until the upgrade to 1.16.0.
I haven’t changed the coordinator firmware.

Let me know if there’s anything I can try to help

netweaver1970, 2020-11-04:
> @netweaver1970 does downgrading the firmware fix the issue?

Things got stable again ... very weird. It seems it needed time to re-balance the network again overnight. 
But I didn't add/remove permanent (router like) devices to have to require this rebalancing... 

So no further major issues anymore, apart from the occasional need to physically power cycle the Raspberry. 
And that's a know problem I saw in other issues too. I'll re-pair the Hue outdoor sensor, to see if that issue is now also fixed with the new FW + Z2M combo.

Also another not directly visible 'issue' is the fact I'm getting every Aqara PIR message 2x, with different Elapsed values, sometimes with different Occupancy values, sometimes with the same occupancy values. sometimes with different LQI. 
I guess doing a debounce might solve this but for a PIR sensor you don't want that...

```
Zigbee2MQTT:info  2020-11-04 09:25:54: MQTT publish: topic 'zigbee2mqtt/Aqara PIR LivingRoom', payload '{"battery":97,"elapsed":6578,"illuminance":79,"illuminance_lux":79,"last_seen":"2020-11-04T09:25:54+00:00","linkquality":144,"occupancy":false,"voltage":2995}'
Zigbee2MQTT:info  2020-11-04 09:25:54: MQTT publish: topic 'zigbee2mqtt/Aqara PIR LivingRoom', payload '{"battery":97,"elapsed":11,"illuminance":79,"illuminance_lux":79,"last_seen":"2020-11-04T09:25:54+00:00","linkquality":141,"occupancy":true,"voltage":2995}'
Zigbee2MQTT:info  2020-11-04 09:26:04: MQTT publish: topic 'zigbee2mqtt/Aqara PIR LivingRoom', payload '{"battery":97,"elapsed":10008,"illuminance":79,"illuminance_lux":79,"last_seen":"2020-11-04T09:25:54+00:00","linkquality":141,"occupancy":false,"voltage":2995}'
Zigbee2MQTT:info  2020-11-04 09:26:09: MQTT publish: topic 'zigbee2mqtt/Aqara PIR LivingRoom', payload '{"battery":97,"elapsed":4617,"illuminance":80,"illuminance_lux":80,"last_seen":"2020-11-04T09:26:09+00:00","linkquality":144,"occupancy":false,"voltage":2995}'
Zigbee2MQTT:info  2020-11-04 09:26:09: MQTT publish: topic 'zigbee2mqtt/Aqara PIR LivingRoom', payload '{"battery":97,"elapsed":10,"illuminance":80,"illuminance_lux":80,"last_seen":"2020-11-04T09:26:09+00:00","linkquality":144,"occupancy":true,"voltage":2995}'
Zigbee2MQTT:info  2020-11-04 09:26:17: MQTT publish: topic 'zigbee2mqtt/Aqara PIR LivingRoom', payload '{"battery":97,"elapsed":8214,"illuminance":81,"illuminance_lux":81,"last_seen":"2020-11-04T09:26:17+00:00","linkquality":147,"occupancy":true,"voltage":2995}'
Zigbee2MQTT:info  2020-11-04 09:26:17: MQTT publish: topic 'zigbee2mqtt/Aqara PIR LivingRoom', payload '{"battery":97,"elapsed":6,"illuminance":81,"illuminance_lux":81,"last_seen":"2020-11-04T09:26:17+00:00","linkquality":147,"occupancy":true,"voltage":2995}'
Zigbee2MQTT:info  2020-11-04 09:26:25: MQTT publish: topic 'zigbee2mqtt/Aqara PIR LivingRoom', payload '{"battery":97,"elapsed":7788,"illuminance":80,"illuminance_lux":80,"last_seen":"2020-11-04T09:26:25+00:00","linkquality":144,"occupancy":true,"voltage":2995}'
Zigbee2MQTT:info  2020-11-04 09:26:25: MQTT publish: topic 'zigbee2mqtt/Aqara PIR LivingRoom', payload '{"battery":97,"elapsed":9,"illuminance":80,"illuminance_lux":80,"last_seen":"2020-11-04T09:26:25+00:00","linkquality":144,"occupancy":true,"voltage":2995}'
Zigbee2MQTT:info  2020-11-04 09:26:26: MQTT publish: topic 'zigbee2mqtt/Aqara PIR Kitchen', payload '{"battery":100,"elapsed":204223,"illuminance":140,"illuminance_lux":140,"last_seen":"2020-11-04T09:26:26+00:00","linkquality":96,"occupancy":false,"voltage":3005}'
Zigbee2MQTT:info  2020-11-04 09:26:26: MQTT publish: topic 'zigbee2mqtt/Aqara PIR Kitchen', payload '{"battery":100,"elapsed":11,"illuminance":140,"illuminance_lux":140,"last_seen":"2020-11-04T09:26:26+00:00","linkquality":96,"occupancy":true,"voltage":3005}'
Zigbee2MQTT:info  2020-11-04 09:26:32: MQTT publish: topic 'zigbee2mqtt/Aqara PIR Kitchen', payload '{"battery":100,"elapsed":5637,"illuminance":141,"illuminance_lux":141,"last_seen":"2020-11-04T09:26:32+00:00","linkquality":96,"occupancy":true,"voltage":3005}'
Zigbee2MQTT:info  2020-11-04 09:26:32: MQTT publish: topic 'zigbee2mqtt/Aqara PIR Kitchen', payload '{"battery":100,"elapsed":9,"illuminance":141,"illuminance_lux":141,"last_seen":"2020-11-04T09:26:32+00:00","linkquality":99,"occupancy":true,"voltage":3005}
```

`

Faszakasza, 2020-11-04:
This is definitely an issue in 1.16.0

Today I've decided to move my HA installation to a new NUC, and reinstalled everything from scratch. I had 3 bulbs and a couple of other devices working without problems for a while on 1.15.0.

On the new installation running 1.16.0 I was unable to get the bulbs working, this showed up in the log:

_info  2020-11-04 21:48:40: IKEA-TradfriBulb-01 (0x842e14fffe7a7f9a): Not supported (Router)

debug 2020-11-04 22:15:47: Received Zigbee message from 'IKEA-TradfriBulb-01', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":553788977,"imageType":16643,"manufacturerCode":4476}' from endpoint 1 with groupID null
warn  2020-11-04 22:15:47: Received message from unsupported device with Zigbee model 'undefined'
warn  2020-11-04 22:15:47: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html._

This is on a ConBee II.

Switching back to 1.15.0 without changing anything else - the bulb registers and works fine.
However bulbs which are already registered keep working if I switch to 1.16.0 afterwards. Looks like only the pairing process is failing. I have reproduced the issue multiple times.

Koenkk, 2020-11-05:
@Faszakasza can you provide the herdsman debug log when pairing with 1.15.0 and 1.16.0?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

closed 2020-12-13 by github-actions[bot]