Koenkk/zigbee2mqtt #4851: Philips Hue P45 Light Bulb - Model 9290024406

Fridolin34, 2020-11-02:
Philips Hue P45 Light Bulb
Warm white light  2700K, Dimmable

https://www.philips-hue.com/en-gb/p/hue-white-1-pack-luster-p45-e14/8719514266889#specifications

    {
        zigbeeModel: ['LWU001'],
        model: '9290024406',
        vendor: 'Philips',
        description: 'Philips Hue P45 Light Bulb',
        extend: hue.light_onoff_brightness,
        ota: ota.zigbeeOTA,
    },
	
..

I tried to contribute something, but until now, I failed - see e.g. https://github.com/danielwelch/hassio-zigbee2mqtt/issues/457
After unsuccessfully checking the list of devices supported in the dev version, I filed this request.

It seems that the way to support new devices changed and for me it is quite complex to get into the topic, but so far I didn't give up.

Thanks in advance,
Fridolin

Koenkk, 2020-11-03:
Added.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Fridolin34, 2020-11-04:
Thank you for adding the entry, but it still seems to be buggy.

After deinstalling and again installing the edge-addon yesterday , all I can get from the logs concerning this new device was the following line:

info  2020-11-04 06:52:47: MQTT publish: topic 'zigbee2mqtt/bridge/info', payload '{"commit":"828b4b3","config":{"advanced":{"adapter_concurrent":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_timeout":0,"availability_whitelist":[],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":11,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"homeassistant_discovery_topic":"homeassistant","homeassistant_legacy_triggers":true,"homeassistant_status_topic":"hass/status","last_seen":"disable","legacy_api":true,"log_directory":"/share/zigbee2mqtt/log/%TIMESTAMP%","log_file":"log.txt","log_level":"debug","log_output":["console","file"],"log_rotation":true,"log_syslog":{},"pan_id":6754,"report":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"ban":[],"blocklist":[],"device_options":{},"devices":{"0x00124b001f84d1aa":{"friendly_name":"0x00124b001f84d1aa"},"0x00158d000450bc35":{"friendly_name":"0x00158d000450bc35"},"0x00158d00045be636":{"friendly_name":"0x00158d00045be636"},"0x001788010940bf8d":{"friendly_name":"0x001788010940bf8d"},"0x001788010940c3d2":{"friendly_name":"0x001788010940c3d2"}},"experimental":{"new_api":true,"output":"json"},"external_converters":[],"frontend":{"port":8099},"groups":{},"homeassistant":true,"map_options":{"graphviz":{"colors":{"fill":{"coordinator":"#e04e5d","enddevice":"#fff8ce","router":"#4ea3e0"},"font":{"coordinator":"#ffffff","enddevice":"#000000","router":"#ffffff"},"line":{"active":"#009900","inactive":"#994444"}}}},"mqtt":{"base_topic":"zigbee2mqtt","include_device_information":false,"server":"mqtt://core-mosquitto","user":"zigbee"},"passlist":[],"permit_join":true,"queue":{},"serial":{"disable_led":false,"port":"/dev/ttyACM0"},"whitelist":[]},"coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"debug","network":{"channel":11,"extended_pan_id":"0xdddddddddddddddd","pan_id":6754},"permit_join":true,"version":"1.16.0-dev"}'

Here the new device id is listed as: "0x001788010940bf8d":{"friendly_name":"0x001788010940bf8d"}

This morning I completely reinstalled the edge-version again and now not even this line is displayed anymore. The light bulb is currently placed directly in front of my NAS, so they should be able to communicate.

When I export the edge-container from my Synology docker host and extract the content, I can see the updated devices-file and the new entry for 'LWU001', but somehow this has no effect.

Koenkk, 2020-11-04:
Can you provide the log when trying to control the bulb?

Fridolin34, 2020-11-04:
Unfortunately I'm not getting to the point that I could see the device somewhere, so I cannot try to control it.
There is no announcement of new devices, there is no check of firmware updates anymore, ...nothing. This did happen, when I had manually created a 'devices.js' in \share\zigbee2mqtt. But I reverted all of this, because it did not lead to a satisfying result.

I've added the complete log to this post. If you need other details, please just let me know.

Thanks

[log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5489611/log.txt)

Koenkk, 2020-11-05:
The device is not under `info  2020-11-04 18:30:20: Currently 3 devices are joined:` so it's not paired, follow https://www.zigbee2mqtt.io/devices/3306431P7.html#pairing to pair

Fridolin34, 2020-11-05:
Thanks for the hint, but somehow I'm confused.

I tried to use Touchlink to reset the lamp, hoping that something happens, but unfortunately, it did not work. In my installation, the first page 'Home' of the zigbee2mqtt addon is empty, no map is shown and also scanning for Touchlink devices brings up nothing. This is not normal, but because I'm getting my sensor values, I did not worry until now.

Furthermore, somehow the bulb must have been connected before. Otherwise I would not have seen the bulb scanning for new firmware updates in the zigbee2mqtt log. What could have happened?

Thanks,
Fridolin

Fridolin34, 2020-11-05:
Sorry, please ignore the previous comment. I just did exactly the same steps again, brought bulb and stick as close together as can be, switched to the Touchlink page of the addon and nearly instantly it was shown. Also the map is working now. No clue, why it is still not working in the non-edge version, but that's another step. 
I now need to figure out, how to adapt the brightness, but at least the device has been added successfully. So this thread might be closed and I'm happily looking forward to see this in an upcoming extension of the release version.

Thanks a lot.

closed 2020-11-05 by Koenkk