Koenkk/zigbee2mqtt #4852: Gledopto PRO products support

regnerus, 2020-11-02:
Zigbee2MQTT has great support for the Gledopto products. About a month ago they released a new PRO series of products with mesh and Zigbee 3.0 support. https://www.reddit.com/r/Gledopto/comments/it4bed/what_is_pro_gledopto_new_product_series/

I am planning on buying some of these new products and wanted to create this issue to discuss the support for them. What I am also especially interested in if someone has already bought a RGB+CCT PRO controller and knows if it is possible to control both the RGB and CCT separately with Zigbee2MQTT?

regnerus, 2020-11-16:
Added the Pro controllers in a merge request (Koenkk/zigbee-herdsman-converters#768). Switching the light on/off seems much faster however one of the other promises by Gledopto was much darker min. brightess. This does not seem to be the case in my tests. 

[image]

Nebr88, 2020-11-16:
Hi @regnerus . Could you please tell if GL-D-003P was included

Nebr88, 2020-11-16:
I own one and can help you with logs or anything else

regnerus, 2020-11-16:
@Nebr88 I did not add GL-D-003P yet! I don't own any downlights so couldn't test it... It *should* be as easy as copying the existing entry for the GL-D-003 light and changing the ZigbeeModel to GL-D-003P. https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

You should try it on your own devices.js and then create a pull request in Koenkk/zigbee-herdsman-converters if it works. :)

Nebr88, 2020-11-16:
@regnerus I've tried but seems that I did something wrong.
I believe that the code should be like this 

`    {
        zigbeeModel: ['GL-D-003P'],
        model: 'GL-D-003P',
        vendor: 'Gledopto',
        description: 'LED RGB + CCT downlight PRO',
        extend: gledopto.light_onoff_brightness_colortemp_colorxy,
    },`

regnerus, 2020-11-17:
@Nebr88 I added it to my devices.js, replace your devices.js file in zigbee-herdsman-converters with this one and it should work! https://gist.github.com/regnerus/02b5269597dfd6eafc7327d548886440

Nebr88, 2020-11-17:
@regnerus Thanks!

regnerus, 2020-11-18:
@Koenkk to continue the discussion from Koenkk/zigbee-herdsman-converters#768, I contacted Gledopto and they claim that the lights can be set to lower brightness and made a video showing this in with the Hue controller https://youtu.be/gxgk1zDfJT0. 

Their comment:

> In many of the tests we've done, the pro version has lower brightness. If you have different brightness, it may be because there are some problems with the instructions sent by your gateway ( Zigbee2MQTT, This gateway is a DIY gateway. If it is not use the standard ZigBee protocol , there may be some instruction problems ), You can try to use other gateways, such as Philips hue, smartThings, Amazon echo plus, Tuya zigbee gateway, etc.

Their comment about Zigbee2MQTT not using the "standard" ZigBee protocol is nonsense of course however could there be a difference between the way the Philips Hue Hub sets the brightness vs Zigbee2MQTT?

closed 2020-12-26 by github-actions[bot]

regnerus, 2020-12-30:
Issue still hasn't been solved however I will reopen once I get more information.