Koenkk/zigbee2mqtt #4853: CC2538+CC2592 router support

jhhbe, 2020-11-02:
I have a CC2538+CC2592 as a coordinator which works great, however with a new CC2538+CC2592 stick flashed as a router I have the impression my remote still uses the coordinator. Also the coordinator shows up in the Zigbee2Mqtt map but not in the devices overview, but it works great.

The router connects to the network and shows as unsupported. Not having too much knowledge sitting in the way I've added it to devices.js as shown below - hoping that with an empty fromZigbee / toZigbee it would simply forward all messages. (again I have the impression it's not doing it but cannot 200% prove it). 

In the Zigbee2Mqtt map it shows my Ikea router (E1746) with solid lines to the coordinator and the CC2538 router as a separate network with a dotted line to the Ikea remote button (E1524/E1810) and no connection to the coordinator. 

Below entry removes the unsupported message but I fear it does not add much support:

    const devices = [
    // tried to add "something" in devices.js
    {
      zigbeeModel: ['cc2538.router.v1'],
      model: 'cc2538.router.v1',
      description: 'signal repeater',
      vendor: 'jethome',
      fromZigbee: [],
      toZigbee: [],
    },

## Information about the device + link
[https://github.com/jethome-ru/zigbee-firmware/blob/master/ti/router/cc2538_cc2592/README.md](url)

## data/database.db entry of the device

`{"id":5,"type":"Router","ieeeAddr":"0x00124b0006104ce8","nwkAddr":25423,"manufId":0,"manufName":"jethome","powerSource":"Mains (single phase)","modelId":"cc2538.router.v1","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,3],"outClusterList":[0],"clusters":{"genBasic":{"attributes":{"manufacturerName":"jethome","powerSource":1,"zclVersion":1,"hwVersion":1,"dateCode":"20200522        "}}},"binds":[],"meta":{}}},"hwVersion":1,"dateCode":"20200522        ","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1603443955100}`

I've looked at other routers like lumi.router and Xbee but I'm a bit lost. Thanks for reading this far as I realize there have been messages about CC2538 earlier.

closed 2020-12-10 by github-actions[bot]

vascojdb, 2021-01-20:
I also have an issue with these CC2538+CC2592 working as a router. I have 4 PCBs with these modules and zigbee2mqtt says they are not supported as a router. In anyone working on this?

hrosak88, 2021-03-10:
vascohb: The same problem. Cc2538+cc2592 as router with new jethome firmware not working in zigbee2mqtt - unsupported device.

jhhbe, 2021-03-10:
Meanwhile I think it is supported:

Friendly name: 0x00124b0006104ce8
Device type: Router
Zigbee model: cc2538.router.v1
Description: MODKAM stick СС2538 router
Support status: Supported

IEEE address: 0x00124b0006104ce8
Network address: 0x634F
Firmware build date: 20200522
Vendor: Custom devices (DiY)
Model: CC2538.ROUTER
Power source
Interview completed: Yes

hrosak88, 2021-03-10:
Jhhbe: yop, that is right. But is it supported with firmware 20200522. Now with new firmware 20210115 shows unsupported device. Model Identifier - cc2538.router.v2.

shardik86, 2021-03-17:
Same issue here. I just flashed firmware 20210310 (model is cc2538.router.v2) and get unsupported device.
Is it good to "downgrade" to firmware 20200522, or is it better to wait for support of newer one?

jhhbe, 2021-03-17:
I think it will be supported in the end - meanwhile I suppose you can copy and paste the v1 into a v2 section in devices.js - there is a section in the documentation on adding new devices that explains how to do that.