Koenkk/zigbee2mqtt #4855: Ikea Tradfre E1524 button Received message from unsupported device with Zigbee model 'undefined'

djansen1987, 2020-11-02:
#Bug Report
## What happened
After moving houses i wanted to gave my zigbee network a fresh reset.
All devices can be linked except the E1524 device

## What did you expect to happen
after pressing the pair button 4 times to start pairing to z2mqtt

## How to reproduce it (minimal and precise)
enabled allow parring
reset button pressing the pairbutton 4 times following:
https://www.zigbee2mqtt.io/devices/E1524_E1810.html

## Debug Info
Zigbee2MQTT version: 1.16.0
Adapter hardware: CC2530
Adapter firmware version: unknown

Zigbee2MQTT:info  2020-11-02 23:03:25: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x000b57fffeecdcb2"},"type":"pairing"}'
Zigbee2MQTT:debug 2020-11-02 23:03:26: Received Zigbee message from '0x000b57fffeecdcb2', type 'commandTradfriArrowSingle', cluster 'genScenes', data '{"value":257,"value2":13}' from endpoint 1 with groupID 0
Zigbee2MQTT:warn  2020-11-02 23:03:26: Received message from unsupported device with Zigbee model 'undefined'
Zigbee2MQTT:warn  2020-11-02 23:03:26: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:debug 2020-11-02 23:03:26: Received Zigbee message from '0x000b57fffeecdcb2', type 'commandToggle', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 0

Koenkk, 2020-11-03:
Seems interview failed, try to repair close to the coordinator.

djansen1987, 2020-12-04:
HI Koenkk,

That seemed to have been the problem, i moved closer to the gateway and paired again and this then worked.
Strange part was that i was pairing multiple devices and this was the only one from that distance that didn't work. (probably battery related)

Thanks for your response!

closed 2020-12-04 by djansen1987