Koenkk/zigbee2mqtt #4856: ELKO Smart RS16 Super TR RF PH - Thermostat

Gaute2000, 2020-11-02:
**ELKO Smart RS16 Super TR RF PH - Thermostat**

**Technical specifications:** 
_Voltage: 230 V/50-60 Hz.
Max load: 2300W
Electronic short circuit fuse
Temperature sensor: Airsensor, floorsensor or regulator (%)
Choose between different suppliers floorsensors
Simple and easy to use saving program.
Antifreeze function
Big display - adjustable brightness
Input for external control_

**devices.js**
> {
>         zigbeeModel: ['Super TR'],
>         model: 'Super TR RF',
>         vendor: 'ELKO',
>         description: 'In-Wall Thermostat for floor heating',
>         supports: 'Temperature sensor: Airsensor, floorsensor or regulator, Saving program, Child-lock',
>         fromZigbee: [],
>         toZigbee: [],
>     },

Tried to copy fromZigbee and toZigbee from other thermostat, but where not able to get it work.
Followed: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html 

**Information about the device**
[Datasheet](https://proff.elko.no/getfile.php/13124139/produkter/doc/EKO05677/Datasheet4523433.pdf)
[Device product page](https://proff.elko.no/elko-smart-super-termostat/elko-smart-rs16-super-tr-rf-ph-article2621-2970.html)

**Database**
[database.zip](https://github.com/Koenkk/zigbee2mqtt/files/5478439/database.zip)

Gaute2000, 2020-11-03:
Please let me know, if its some logfiles or data I can provide to assist to make this device supported.
For information Im running bare-metal of zigbee2mqtt on a Pi4 with a Electrolama zig-a-zig-ah! (zzh!) stick with firmware CC26X2R1_20200925 installed.

Gaute2000, 2020-12-07:
Still active and a open issue

sluquet, 2021-01-10:
Would love to have this added :-)

henrikrox, 2021-01-12:
Would also love this to be added

henrikrox, 2021-01-19:
@Gaute2000 I tried your code in devices.js, but i get alot of 

ReferenceError: bind is not defined at Object.configure, it sees the device, but "something" is messed up. @Koenkk Will this device get added? Or is it something i can do.

Koenkk, 2021-01-20:
Changing the content of `configure` to below should fix this:

```js
const endpoint = device.getEndpoint(1);
await reporting.bind(endpoint, coordinatorEndpoint, ['hvacThermostat', 'genBasic', 'genIdentify']);
await reporting.thermostatTemperature(endpoint);
await reporting.thermostatSystemMode(endpoint);
await reporting.thermostatOccupiedHeatingSetpoint(endpoint);
await reporting.thermostatSystemMode(endpoint);
```

closed 2021-02-27 by github-actions[bot]

sluquet, 2021-03-07:
Did any of you get this to work? No luck here.