Koenkk/zigbee2mqtt #4858: Wall switch Lonsonho X703A is recognized as Zemismart ZM-L03E-Z

szeryf85, 2020-11-03:
Hello,
I have four wall switches Lonsonho X703A (with neutral), and all of them are recognized as Zemismart ZM-L03E-Z.
The switches work fine, but although they are recognized as a router, no other device wants to connect to them. Could the problem be in the wrong identification of the switch?

Regards,
Lukas

Koenkk, 2020-11-03:
These are actually the same device (both are whitelabel from TuYa). How Zigbee2MQTT recognises them doesn't affect the functionality, also Zigbee2MQTT control the routers a device use. Try to repair the device close to the desired router helps sometimes.

klami85, 2020-11-05:
Same here, but all works fine (including router functionality).
Only photo is incorrect.

closed 2020-12-13 by github-actions[bot]