Koenkk/zigbee2mqtt #4859: debounce_ignore not working for occupancy in Aqara PIR

McGiverGim, 2020-11-03:
## What happened
I'm not too sure if I have understood the `debounce` feature correctly. I have several Aqara PIR RTCGQ11LM (https://www.zigbee2mqtt.io/devices/RTCGQ11LM.html) that send the illuminance and the occupancy.
I have configured it to group the messages except for the occupancy one, that I don't want any delay. 
I did it with something like this:
```yaml
'0x00158d0004147c40':
  friendly_name: presencia_pasillo
  debounce: 5
  debounce_ignore:
    - occupancy
```
I have used a high `debounce` value in this test to be sure of the problem.
The result is that I don't receive the `occupancy: true` until 5 seconds later than it is detected.
 
## What did you expect to happen
Receive the `occupancy: true` message almost instantly.

## How to reproduce it (minimal and precise)
Configure as in my explanation and test it.

## Debug info
Zigbee2MQTT version: 1.16.0
Adapter hardware: CC2531
Adapter firmware version: zStack12.20190608

Koenkk, 2020-11-03:
That is not what `debounce_ignore` does, 2 examples to explain, with `debounce: 5`.

**Without debounce_ignore**
00:00 - occupancy: true received
00:01 - occupancy: false received
00:06 - occupancy false published

**With debounce_ignore: occupancy**
00:00 - occupancy: true received
00:01 - occupancy: false received
00:01 - occupancy: true published
00:06 - occupancy false published

McGiverGim, 2020-11-03:
Ok understood. It was not clear on the documentation.
I want it published without any delay, so I will not use the debounce option at all.
Thanks for your explanation, I close this issue.

closed 2020-11-03 by McGiverGim

dankarization, 2023-11-11:
> That is not what `debounce_ignore` does, 2 examples to explain, with `debounce: 5`.
> 
> **Without debounce_ignore** 00:00 - occupancy: true received 00:01 - occupancy: false received 00:06 - occupancy false published
> 
> **With debounce_ignore: occupancy** 00:00 - occupancy: true received 00:01 - occupancy: false received 00:01 - occupancy: true published 00:06 - occupancy false published
> 
> I think in your case a debounce: 1 without debounce_ignore should be fine.

I don't get this logic. Why in the 2nd case occupancy: true was published 1 second after it was recieved, and false was 5 seconds after it was recieved?

Knellius, 2024-01-02:
> I don't get this logic. Why in the 2nd case occupancy: true was published 1 second after it was recieved, and false was 5 seconds after it was recieved?
If I got it right, with debounce_ignore very first message will be published without delay, and next message with unique value of ignored attribute will be published after time, which is set in debounce option.