Koenkk/zigbee2mqtt #4864: [device] Hue Motion and binding to groups?

sjorge, 2020-11-03:
While messing with my Hue Motion I notice it had the following output clusters:
[image]

My remote have the exact same output cluster, binding genOnOff and genLevelCtrl to a group or bulb has no effect though.

I wonder why the cluster is there if it can't be used? Is there maybe a Hue specific property we need to write to enable it? It would be nice to have the sensor talk to a group/device directly via binding like we can do with the ikea motion ones.

MattWestb, 2020-11-03:
I have 2 of them but is using as environment sensors (light measuring primary).
But if they is working like the HUE dimmer switch they need being unbonded from the coordinator for start sending commands to one groupe. I don't knowing only thinking it can being worth testing.

sjorge, 2020-11-03:
@MattWestb the coordinator isn't bound to ep1 though, I also tried to unbind but that failed.
Binding to a group works, but nothing happens ( checked the traffic )

So I'm thinking there might be something we need to configure to make it do stuff.

MattWestb, 2020-12-04:
Have you looking if deCONz is handling it ? They have one HUE fane that is normally digging very deep to finding out how its working (also with LIDL christmas tree lights) :-))

sjorge, 2020-12-04:
I did a google search but nothing popped up, aside from people having issues with lux and temp readings.

MattWestb, 2020-12-04:
They do binding but more things before that and i dont know if there is doing it for the on/off cluster 0x0006 so its working the the ZNCP is offline.
https://github.com/dresden-elektronik/deconz-rest-plugin/search?q=hue+dimmer+switch
Look in the bindings.cpp for SML00.

sjorge, 2020-12-04:
Looks to do the same as z2m is doing, it just binds the msOccupancySensing cluster.

closed 2020-12-04 by sjorge

sjorge, 2020-12-04:
[image]
I did end up sniffing the traffic with the default setup, the genOnOff/genLevelCtrl bound to the coordinator and to a group... the result is the same... only sends out msOccupancy and nothing else. So looks like although the cluster is there, it's not used at all on the device side. Perhaps in a future update but who knows.

chris-kuhr, 2023-12-01:
I connt find a way to bind the motion sensor to a light group.
What is the state here? 
Why is the issue closed?

sjorge, 2023-12-01:
The device does not seem to support it, so there is nothing we can do to make it work.