Koenkk/zigbee2mqtt #4866: can zigbee2mqtt support chips from silicon labs

cyang10, 2020-11-03:
## Is your feature request related to a problem? Please describe
I have one product which is using silicon labs and very interested to add the wide ranges of devices that current zigbee2mqtt supports. Is there a plan to support other chips?

## Describe the solution you'd like
to support chips from silicon labs

## Describe alternatives you've considered
don't know whether there is an anternative.

## Additional context
No

Koenkk, 2020-11-04:
What do you mean?
- Are you trying to pair silicon labs based router/end devices to Zigbee2MQTT using the Conbee II or Texas Instrument adapter?
- Or are you trying to use a silicon labs coordinator with Zigbee2MQTT

cyang10, 2020-11-04:
I mean the 2nd case, i am working on a product with Silicon labs coordinator, is it possible to make zigbee2mqtt to work with that product. Thanks.

Koenkk, 2020-11-05:
Yes it's possible and can be done by implementing the following interface: https://github.com/Koenkk/zigbee-herdsman/blob/master/src/adapter/adapter.ts

As reference adapter code for zigate, zstack and deconz can be found here: https://github.com/Koenkk/zigbee-herdsman/tree/master/src/adapter

Very recently the zigate adapter has been added, you can find the merge request here: https://github.com/Koenkk/zigbee-herdsman/pull/241/files (good reference to see what you have to change)

cyang10, 2020-11-05:
Great, Thanks a lot! Will learn and let you know the updates.

cyang10, 2020-11-06:
Koen, 
Thanks a lot for your great suggestions.
I just realized that the core part of the effort is to let silicon labs chips to support those legacy or even proprietary protocols, which seems to be covered by the firmware you provided in github. Would you like to share some insights how to support such wide range of devices at firmware level? Thanks a lot!

MattWestb, 2020-11-06:
@cyang10 The ERF32 platform can being updated to the latest version of EZSP (ver 8) and we having more devs that can building the firmware that working. With the ember ones its little more tricky and i dont knowing if its possible getting all devices on the new protocol (EZSP ver 8) or must running on ver 6/7. 
EZSP ver 8 is NOT backward compatible with older versions (4-7), EZSP ver 7 is backward compatibility with older versions of the protocol (ver 4-6).
If implanting only EZSP ver 8 its being mutch easter and most NCPs is being supported.
ZHA / Zigpy have doing one commit with nearly 20000 lines commitment to bellows (how many hours spent of the devs is not know) for supporting all sirlabs EZSP protocol from ver 4 to the latest 8. I saying its great and nice done but is for one new project little overkill and EZSP ver 8 is enough or if being very generous with time also ver 6/7 is realistic but on the border to overkill.
Tasmota have only implantes EZSP ver 8 and and is targeting all EFR32 and leaving the ember platform unsupported.

Then starting something new its normally better throwing all old things and starting without the old historical things in the bag that is making things more complex and more problem in the dev process.

cyang10, 2020-11-10:
Matt,
Thanks a lot for your insights. It seems that it is a technical doable efforts but need more considerations.

MattWestb, 2020-11-10:
If going for only protocol ver 8 then take one look on Tasmota implanting of Sonoff Zigbee Bridge.
Stephan have doing all from scratch and looking on other systems but in the end its works nice for small system and pretty easy for the user to managen.
The problem its that the security its being setted pretty low opposite to ZHA that have trying having its high as possible for the V8 (install conde is enabled for Zigbee 3 but HA have the GUI not implanted) and lower for the older protocol implantation.
The problem is that ZHA is python and Tasmota is written in C and Z2M is coded with much caffeine  ;-)

closed 2020-12-19 by github-actions[bot]

Hedda, 2021-03-02:
FYI, kirovilya started working on a EmberZNet/EZSP adapter and is asking for collaboration help in Koenkk/zigbee-herdsman#317

cyang10, 2021-03-03:
@Hedda  Thanks a lot for your information, it really helps!