Koenkk/zigbee2mqtt #4867: Support for Node v15

xaque208, 2020-11-04:
## Is your feature request related to a problem? Please describe

Running with node v15 seems to prevent `zigbee2mqtt` from starting.

## Describe the solution you'd like

I would like to be able to run this project using the newer node version.  Downgrading node on my operatingsystem isn't a preferred solution due to the rolling-release nature archlinux. 

## Additional context

```
Nov 03 21:51:47 rp41 systemd[1]: Started zigbee2mqtt.
Nov 03 21:51:48 rp41 npm[11048]: > zigbee2mqtt@1.15.0 start /opt/zigbee2mqtt
Nov 03 21:51:48 rp41 npm[11048]: > node index.js
Nov 03 21:51:48 rp41 npm[11059]:                 Zigbee2MQTT requires node version ^10 || ^12 || ^14, you are running v15.0.1!
Nov 03 21:51:49 rp41 npm[11059]: Zigbee2MQTT:info  2020-11-03 21:51:49: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-11-03.21-51-49' filename: log.txt
```

Rotzbua, 2020-11-04:
Doesn't look like error message, more like warning..
Maybe consider the docker image as alternative.

Koenkk, 2020-11-04:
I did some small test and Zigbee2MQTT runs fine on node 15, marked it as supported now.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-11-04 by Koenkk

xaque208, 2020-11-04:
Excellent, thank you.  It was a dark morning with the daemon not running :).  But also, I'll check out the docker image since it looks to run on the raspberry pi.  I must have just overlooked that when I was setting up the PoC for my house.

xaque208, 2020-11-07:
Docker works great.