Koenkk/zigbee2mqtt #4871: Failed to call 'Receive' 'onZigbeeEvent' TypeError: Cannot read property 'type' of undefined at /zigbee2mqtt-1.16.0

iamkarlson, 2020-11-04:
## What happened
After upgrade to the 1.16.0 switches stopped working. Error I'm getting from the home assistant addon log:

```
Zigbee2MQTT:debug 2020-11-04 10:50:03: Received Zigbee message from '0x680ae2fffe7abf10', type 'commandUpOpen', cluster 'closuresWindowCovering', data '{}' from endpoint 1 with groupID 0
Zigbee2MQTT:error 2020-11-04 10:50:03: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at /zigbee2mqtt-1.16.0/lib/extension/receive.js:120:42
    at Array.filter (<anonymous>)
    at Receive.onZigbeeEvent (/zigbee2mqtt-1.16.0/lib/extension/receive.js:119:65)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.16.0/lib/controller.js:378:44)
    at Controller.onZigbeeEvent (/zigbee2mqtt-1.16.0/lib/controller.js:261:14)
    at Zigbee.emit (events.js:315:20)
    at Controller.<anonymous> (/zigbee2mqtt-1.16.0/lib/zigbee.js:68:52)
    at Controller.emit (events.js:315:20)
    at Controller.<anonymous> (/zigbee2mqtt-1.16.0/node_modules/zigbee-herdsman/dist/controller/controller.js:545:22)
    at Generator.next (<anonymous>))
Zigbee2MQTT:debug 2020-11-04 10:51:29: Received Zigbee message from '0x000d6ffffe459b43', type 'commandOnWithTimedOff', cluster 'genOnOff', data '{"ctrlbits":0,"offwaittime":0,"ontime":1800}' from endpoint 1 with groupID 0
Zigbee2MQTT:error 2020-11-04 10:51:29: Failed to call 'Receive' 'onZigbeeEvent' (TypeError: Cannot read property 'type' of undefined
    at /zigbee2mqtt-1.16.0/lib/extension/receive.js:120:42
    at Array.filter (<anonymous>)
    at Receive.onZigbeeEvent (/zigbee2mqtt-1.16.0/lib/extension/receive.js:119:65)
    at Controller.callExtensionMethod (/zigbee2mqtt-1.16.0/lib/controller.js:378:44)
    at Controller.onZigbeeEvent (/zigbee2mqtt-1.16.0/lib/controller.js:261:14)
    at Zigbee.emit (events.js:315:20)
    at Controller.<anonymous> (/zigbee2mqtt-1.16.0/lib/zigbee.js:68:52)
    at Controller.emit (events.js:315:20)
    at Controller.<anonymous> (/zigbee2mqtt-1.16.0/node_modules/zigbee-herdsman/dist/controller/controller.js:545:22)
    at Generator.next (<anonymous>))
```
## What did you expect to happen
Switches work as usual.

## How to reproduce it (minimal and precise)
Get switch paired to z2m, click on any button. 

## Debug info
Zigbee2MQTT version: 1.16.0
Adapter hardware: Conbee II
Adapter firmware version: Conbee II default

Koenkk, 2020-11-04:
Did you ever modify `devices.js`? (e.g. in order to add support for new devices?)

iamkarlson, 2020-11-04:
damn, yes, thanks, sorry. removed devices.js and problem solved. thanks

closed 2020-11-04 by iamkarlson