Koenkk/zigbee2mqtt #4872: TuYa TS0013 error on coordinator reboot.

u17194907425, 2020-11-04:
## What happened

First of all, this is a minor thing since it seems that no functionality is blocked. Nevertheless.

After each reboot of Zigbee2mqtt plug in for Home Assistant I observe the following error in log:
```
Zigbee2MQTT:error 2020-11-04 14:41:39: Failed to configure 'hall_wall_switch_batch_wc', attempt 1 (Error: ConfigureReporting 0x680ae2fffe71af76/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 28619 - 1 - 3 - 6 - 7 after 10000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.16.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))
```

It is related to my wall switch [TS0013 by TuYa](https://www.zigbee2mqtt.io/devices/TS0013.html). However if I try toggling the switches in Home Assistant's interface, the device reacts as it should. E.g. I toggle a switch at 14:41:43 (4 seconds after receiving the 1st error and I see this:
```
Zigbee2MQTT:info  2020-11-04 14:41:43: MQTT publish: topic 'zigbee2mqtt/hall_wall_switch_batch_wc', payload '{"linkquality":63,"state_center":"OFF","state_left":"ON","state_right":"OFF"}'
Zigbee2MQTT:info  2020-11-04 14:41:43: Configuring 'hall_wall_switch_batch_wc'
Zigbee2MQTT:info  2020-11-04 14:41:43: MQTT publish: topic 'zigbee2mqtt/hall_wall_switch_batch_wc', payload '{"linkquality":63,"state_center":"OFF","state_left":"ON","state_right":"OFF"}'

...

Zigbee2MQTT:error 2020-11-04 14:42:07: Failed to configure 'hall_wall_switch_batch_wc', attempt 2 (Error: ConfigureReporting 0x680ae2fffe71af76/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 28619 - 1 - 5 - 6 - 7 after 10000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.16.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))
```

- I press ONCE, but there seems to be 2 messages.
- There is another attempt to configure the device but it fails again (like immediately after reboot)

Then I toggle the switch off.
```
Zigbee2MQTT:info  2020-11-04 14:45:54: MQTT publish: topic 'zigbee2mqtt/hall_wall_switch_batch_wc', payload '{"linkquality":66,"state_center":"OFF","state_left":"OFF","state_right":"OFF"}'
Zigbee2MQTT:info  2020-11-04 14:45:54: Configuring 'hall_wall_switch_batch_wc'
Zigbee2MQTT:info  2020-11-04 14:45:54: MQTT publish: topic 'zigbee2mqtt/hall_wall_switch_batch_wc', payload '{"linkquality":63,"state_center":"OFF","state_left":"OFF","state_right":"OFF"}'

...

Zigbee2MQTT:error 2020-11-04 14:46:18: Failed to configure 'hall_wall_switch_batch_wc', attempt 3 (Error: ConfigureReporting 0x680ae2fffe71af76/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 28619 - 1 - 7 - 6 - 7 after 10000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.16.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))
```

Then if I toggle the switches no more configuration attempts happen but what remains is that double-messaging.
```
Zigbee2MQTT:info  2020-11-04 14:47:16: MQTT publish: topic 'zigbee2mqtt/hall_wall_switch_batch_wc', payload '{"linkquality":63,"state_center":"OFF","state_left":"ON","state_right":"OFF"}'
Zigbee2MQTT:info  2020-11-04 14:47:16: MQTT publish: topic 'zigbee2mqtt/hall_wall_switch_batch_wc', payload '{"linkquality":63,"state_center":"OFF","state_left":"ON","state_right":"OFF"}'

Zigbee2MQTT:info  2020-11-04 14:47:30: MQTT publish: topic 'zigbee2mqtt/hall_wall_switch_batch_wc', payload '{"linkquality":63,"state_center":"ON","state_left":"ON","state_right":"OFF"}'
Zigbee2MQTT:info  2020-11-04 14:47:30: MQTT publish: topic 'zigbee2mqtt/hall_wall_switch_batch_wc', payload '{"linkquality":63,"state_center":"ON","state_left":"ON","state_right":"OFF"}'

Zigbee2MQTT:info  2020-11-04 14:47:39: MQTT publish: topic 'zigbee2mqtt/hall_wall_switch_batch_wc', payload '{"linkquality":63,"state_center":"ON","state_left":"ON","state_right":"ON"}'
Zigbee2MQTT:info  2020-11-04 14:47:39: MQTT publish: topic 'zigbee2mqtt/hall_wall_switch_batch_wc', payload '{"linkquality":66,"state_center":"ON","state_left":"ON","state_right":"ON"}'
```

If I press the physical buttons on the wall switch itself then there is no messaging duplication happens.

## What did you expect to happen

1.  No error messaging in the log.
2.  No duplicated messages.

## How to reproduce it (minimal and precise)

1. Have the TS0013 paired with coordinator
1. Reboot the coordinator.
1. Watch logs
1. Try toggling switches from Home Assistant.

## Debug info
Zigbee2MQTT version: 1.16.0
Adapter hardware:  CC26X2R1
Adapter firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20200417,"transportrev":2},"type":"zStack3x0"}'

u17194907425, 2020-11-04:
A speculation about duplicated messaging. May it happen because the wall switch itself broadcasts its state anytime it has changed? So when I toggle a switch in HA the following might happen

1. the switch is toggled
1. a message is being broadcast to initiate a state change inside the wall switch
1. that state is changed
1. the switch then announces its new state with an exact same message. Like it does when I press the physical button on the wall switch itself.

Koenkk, 2020-11-04:
Can you provide the debug log of this?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

klami85, 2020-11-06:
Same problem here. Debug log for changing state to on and then to off:

```
Zigbee2MQTT:debug 2020-11-06 09:46:28: Received MQTT message on 'zigbee2mqtt/Swiatlo lazienka/set' with data '{"state_center":"ON"}'

Zigbee2MQTT:debug 2020-11-06 09:46:28: Publishing 'set' 'state' to 'Swiatlo lazienka'

Zigbee2MQTT:debug 2020-11-06 09:46:29: Received Zigbee message from 'Swiatlo lazienka', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 2 with groupID 0

Zigbee2MQTT:info  2020-11-06 09:46:29: MQTT publish: topic 'zigbee2mqtt/Swiatlo lazienka', payload '{"linkquality":47,"state":"OFF","state_center":"ON","state_left":"OFF","state_right":"OFF"}'

Zigbee2MQTT:info  2020-11-06 09:46:29: Configuring 'Swiatlo lazienka'

Zigbee2MQTT:info  2020-11-06 09:46:29: MQTT publish: topic 'zigbee2mqtt/Swiatlo lazienka', payload '{"linkquality":47,"state":"OFF","state_center":"ON","state_left":"OFF","state_right":"OFF"}'

Zigbee2MQTT:debug 2020-11-06 09:46:32: Received MQTT message on 'zigbee2mqtt/Swiatlo lazienka/set' with data '{"state_center":"OFF"}'

Zigbee2MQTT:debug 2020-11-06 09:46:32: Publishing 'set' 'state' to 'Swiatlo lazienka'

Zigbee2MQTT:debug 2020-11-06 09:46:51: Received MQTT message on 'zigbee2mqtt/Swiatlo lazienka/set' with data '{"state_center":"OFF"}'

Zigbee2MQTT:debug 2020-11-06 09:46:51: Publishing 'set' 'state' to 'Swiatlo lazienka'

Zigbee2MQTT:error 2020-11-06 09:46:52: Failed to configure 'Swiatlo lazienka', attempt 2 (Error: ConfigureReporting 0xec1bbdfffeb63a35/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 62584 - 1 - 9 - 6 - 7 after 10000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.16.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))

Zigbee2MQTT:debug 2020-11-06 09:46:52: Received Zigbee message from 'Swiatlo lazienka', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 2 with groupID 0

Zigbee2MQTT:info  2020-11-06 09:46:52: MQTT publish: topic 'zigbee2mqtt/Swiatlo lazienka', payload '{"linkquality":47,"state":"OFF","state_center":"OFF","state_left":"OFF","state_right":"OFF"}'

Zigbee2MQTT:info  2020-11-06 09:46:52: Configuring 'Swiatlo lazienka'

Zigbee2MQTT:info  2020-11-06 09:46:52: MQTT publish: topic 'zigbee2mqtt/Swiatlo lazienka', payload '{"linkquality":47,"state":"OFF","state_center":"OFF","state_left":"OFF","state_right":"OFF"}'

Zigbee2MQTT:debug 2020-11-06 09:46:52: Received Zigbee message from 'Swiatlo lazienka', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 2 with groupID 0

Zigbee2MQTT:info  2020-11-06 09:46:52: MQTT publish: topic 'zigbee2mqtt/Swiatlo lazienka', payload '{"linkquality":47,"state":"OFF","state_center":"OFF","state_left":"OFF","state_right":"OFF"}'

Zigbee2MQTT:info  2020-11-06 09:46:52: MQTT publish: topic 'zigbee2mqtt/Swiatlo lazienka', payload '{"linkquality":47,"state":"OFF","state_center":"OFF","state_left":"OFF","state_right":"OFF"}'
```

Koenkk, 2020-11-06:
@klami85 how many gangs does your switch have? (3 ?)

klami85, 2020-11-06:
Yes 3 gangs.
I think it is incorrectly identified. 
I will open another issue about that.

u17194907425, 2020-11-07:
Hi @Koenkk 
I was able to capture log only now. Here it is attached.

key points:
- `error 2020-11-07 16:05:19: Failed to configure 'hall_wall_switch_batch_wc'` - the 1st error. after reboot. then there are no errors for a while and...
- `error 2020-11-07 16:14:01: Failed to configure 'hall_wall_switch_batch_wc', attempt 2` - the 2nd one. Happens after I try to switch.

The device is 3 gang. No neutral line. Model is stated as X713A and part number as X713A-0011

[z2m-log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5504777/z2m-log.txt)

Koenkk, 2020-11-07:
Issue should be fixed in latest dev.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

u17194907425, 2020-11-08:
@Koenkk Thank you! It looks like the approach is just to ignore the error since it does not seem to do any damage. I'll wait for the next release and validate absence of errors.
Cheers!

closed 2020-12-17 by github-actions[bot]