Koenkk/zigbee2mqtt #4873: Error: Interview failed because can not get node descriptor, konke motion sensor

rcpoison, 2020-11-04:
Receiving 'Error: Interview failed because can not get node descriptor'
```
2020-11-04T12:07:43.883Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - tcDeviceInd - {"nwkaddr":49875,"extaddr":"0x14b457fffe46a85d","parentaddr":0}
2020-11-04T12:07:43.883Z zigbee-herdsman:controller:log Device '0x14b457fffe46a85d' joined
2020-11-04T12:07:43.883Z zigbee-herdsman:controller:log Device '0x14b457fffe46a85d' accepted by handler
2020-11-04T12:07:43.883Z zigbee-herdsman:controller:log New device '0x14b457fffe46a85d' joined
2020-11-04T12:07:43.883Z zigbee-herdsman:controller:log Creating device '0x14b457fffe46a85d'
2020-11-04T12:07:43.886Z zigbee-herdsman:controller:log Interview '0x14b457fffe46a85d' start
2020-11-04T12:07:43.886Z zigbee-herdsman:controller:device:log Interview - start device '0x14b457fffe46a85d'
2020-11-04T12:08:06.917Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x14b457fffe46a85d', attempt 1
2020-11-04T12:08:29.948Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x14b457fffe46a85d', attempt 2
2020-11-04T12:08:52.978Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x14b457fffe46a85d', attempt 3
2020-11-04T12:09:16.009Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x14b457fffe46a85d', attempt 4
2020-11-04T12:09:39.041Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x14b457fffe46a85d', attempt 5
2020-11-04T12:10:02.073Z zigbee-herdsman:controller:device:log Interview - node descriptor request failed for '0x14b457fffe46a85d', attempt 6
2020-11-04T12:10:02.074Z zigbee-herdsman:controller:device:log Interview - failed for device '0x14b457fffe46a85d' with error 'Error: Interview failed because can not get node descriptor ('0x14b457fffe46a85d')
2020-11-04T12:10:02.075Z zigbee-herdsman:controller:error Interview failed for '0x14b457fffe46a85d with error 'Error: Interview failed because can not get node descriptor ('0x14b457fffe46a85d')'
```

System:
- Linux/x86_64
- CC2531

Zigbee2MQTT versions tested:
- 1.16.0
- 1.15.0
- 1.14.2
- 1.14.4 (commit #0cd46078)

Sensors:
2x Konke motion sensor, new (0x14b457fffe46a84f, 0x14b457fffe46a85d)

Steps tried according to https://www.zigbee2mqtt.io/information/FAQ.html#interview-fails
- reboot :P
- directly next to stick
- directly plugged in without extension cable
- replaced a battery on one of the sensors
- tried resetting+re-pairing quite a few times
- reflashing the stick, multiple times

Log of pairing attempts of both devices:
[log-konke.txt](https://github.com/Koenkk/zigbee2mqtt/files/5272395/log-konke.txt)

EDIT: Just found my second CC2531 and tried with it, same result.

_Originally posted by @rcpoison in https://github.com/Koenkk/zigbee2mqtt/issues/4366#issuecomment-698012476_

rcpoison, 2020-11-04:
Here#s the log captured with 1.16.0:
[log-konke-1.16.0.txt](https://github.com/Koenkk/zigbee2mqtt/files/5487707/log-konke-1.16.0.txt)

Also have an existing konke motion sensor paired, no idea if this is a new revision.

Koenkk, 2020-11-04:
What Zigbee channel are you running on?

rcpoison, 2020-11-04:
channel: 15

Koenkk, 2020-11-04:
Since you have a spare CC2531, can you sniff the traffic when pairing? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

rcpoison, 2020-11-04:
will try :)

closed 2020-12-12 by github-actions[bot]

sieukrem, 2021-11-21:
> will try :)

I have similar issue with ikea led power supply. Did you find a solution?

bblicke1, 2025-02-21:
I'm having the same issue with an Aqara Temperature Sensor