Koenkk/zigbee2mqtt #4878: IKEA light randomly turns on

Wesleyl89, 2020-11-05:
## What happened

I have an [Ikea light](https://www.zigbee2mqtt.io/devices/LED1733G7.html) that is directly[ binded to a E1743](https://www.zigbee2mqtt.io/devices/E1743.html#binding) switch through zigbee2mqtt. Everything works fine except that sometimes it randomly turns on for no reason.

From the logs it seems that the group that the light is in was turned ON for some reason, which will obviously turn the light ON.

## What did you expect to happen

Not turn on randomly

## How to reproduce it (minimal and precise)

Can't really reproduce it and seems to be pretty random.

## Debug info

I renamed the base_topic to homeassistant (the group isn't exposed in Home assistant btw)
```
info  2020-11-05 02:16:28: Successfully setup reporting for '0x842e14fffe633d92' - 1 - genOnOff
info  2020-11-05 02:16:28: Successfully setup reporting for '0x842e14fffe633d92' - 1 - genLevelCtrl
info  2020-11-05 02:16:29: Successfully setup reporting for '0x842e14fffe633d92' - 1 - lightingColorCtrl
info  2020-11-05 02:16:29: MQTT publish: topic 'homeassistant/bridge/event', payload '{"data":{"friendly_name":"nightlight_wesley","ieee_address":"0x842e14fffe633d92"},"type":"device_announce"}'
info  2020-11-05 02:16:29: MQTT publish: topic 'homeassistant/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"nightlight_wesley"},"type":"device_announced"}'
info  2020-11-05 02:16:29: MQTT publish: topic 'homeassistant/nightlight_wesley', payload '{"brightness":24,"color":{"x":0.4599,"y":0.4106},"color_temp":454,"linkquality":108,"state":"OFF","update":{"state":"available"},"update_available":true}'
info  2020-11-05 02:16:30: MQTT publish: topic 'homeassistant/nightlight_wesley_group', payload '{"brightness":24,"color":{"x":0.4599,"y":0.4106},"color_temp":454,"state":"ON"}'
info  2020-11-05 02:16:30: MQTT publish: topic 'homeassistant/nightlight_wesley', payload '{"brightness":24,"color":{"x":0.4599,"y":0.4106},"color_temp":454,"linkquality":105,"state":"ON","update":{"state":"available"},"update_available":true}'
info  2020-11-05 02:16:31: MQTT publish: topic 'homeassistant/nightlight_wesley', payload '{"brightness":24,"color":{"x":0.4599,"y":0.4106},"color_temp":454,"linkquality":105,"state":"ON","update":{"state":"available"},"update_available":true}'
info  2020-11-05 02:16:31: MQTT publish: topic 'homeassistant/nightlight_wesley', payload '{"brightness":24,"color":{"x":0.4599,"y":0.4106},"color_temp":454,"linkquality":108,"state":"ON","update":{"state":"available"},"update_available":true}'
```

Zigbee2MQTT version: 1.16.0
Adapter hardware: CC2652 (Slaesh)
Adapter firmware version:

Koenkk, 2020-11-05:
Please provide the debug logging when this happens.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

Wesleyl89, 2020-11-05:
I turned on the debug log. Now I'll have to wait untill it happens again.

lolorc, 2020-11-07:
I had this issue this week at 5am, the bulb is in my bedroom :-)
here are the logs, the device is led1733g7_02

[led1733g7_02.txt](https://github.com/Koenkk/zigbee2mqtt/files/5505359/led1733g7_02.txt)

Koenkk, 2020-11-08:
@lolorc I don't see any commands send from Zigbee2MQTT but maybe it happens because of pinging (disable availability feature) or reporting being setup (disable report feature)

lolorc, 2020-11-08:
me neither, I also thought of the pinging.
I've updated the firmware of the bulb, will do with the other 1733G7 bulb I'm using, will see if it happens again.
This bulb was also grouped / bound to a E1743 switch, but the switch's battery have been depleted and I haven't replaced them for quite some time.
Don't know if it's related, will see if that happens again with pings on and the updated firmware. (will disable pings if it happens again)

It looks that it happens to @Wesleyl89 more often tho.

Wesleyl89, 2020-11-12:
The light turned on randomly again, but this time I had debugging on:
```
debug 2020-11-12 11:38:37: Received Zigbee message from 'nightlight_wesley', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":1,"fileVersion":537011747,"imageType":16900,"manufacturerCode":4476}' from endpoint 1 with groupID 0
debug 2020-11-12 11:38:37: Check if update available for '0x842e14fffe633d92' (TRADFRI bulb E14 WS opal 600lm)
debug 2020-11-12 11:38:37: Device 'nightlight_wesley' announced itself
debug 2020-11-12 11:38:37: Setup reporting for '0x842e14fffe633d92' - 1 - genOnOff
info  2020-11-12 11:38:37: Successfully setup reporting for '0x842e14fffe633d92' - 1 - genOnOff
debug 2020-11-12 11:38:37: Setup reporting for '0x842e14fffe633d92' - 1 - genLevelCtrl
debug 2020-11-12 11:38:37: Received MQTT message on 'homeassistant/bridge/logging' with data '{"level":"info","message":"Successfully setup reporting for '0x842e14fffe633d92' - 1 - genOnOff"}'
debug 2020-11-12 11:38:37: Is new image available for '0x842e14fffe633d92', current '{"fieldControl":1,"manufacturerCode":4476,"imageType":16900,"fileVersion":537011747}', latest meta '{"fileVersion":537040419,"url":"http://fw.ota.homesmart.ikea.net/global/GW1.0/01.11.051/bin/10038562-2.1-TRADFRI-sy5882-bulb-ws-2.0.
029.ota.ota.signed"}'
debug 2020-11-12 11:38:37: Update available for '0x842e14fffe633d92': YES
info  2020-11-12 11:38:37: MQTT publish: topic 'homeassistant/nightlight_wesley', payload '{"brightness":21,"color":{"x":0.4599,"y":0.4106},"color_temp":454,"linkquality":69,"state":"OFF","update":{"state":"available"},"update_available":true}'
info  2020-11-12 11:38:37: Update available for 'nightlight_wesley'
info  2020-11-12 11:38:37: MQTT publish: topic 'homeassistant/bridge/log', payload '{"message":"Update available for 'nightlight_wesley'","meta":{"device":"nightlight_wesley","status":"available"},"type":"ota_update"}'
debug 2020-11-12 11:38:37: Received MQTT message on 'homeassistant/bridge/logging' with data '{"level":"info","message":"MQTT publish: topic 'homeassistant/nightlight_wesley', payload '{\"brightness\":21,\"color\":{\"x\":0.4599,\"y\":0.4106},\"color_temp\":454,\"linkquality\":69,\"state\":\"OFF\",\"update\":{\"state
\":\"available\"},\"update_available\":true}'"}'
debug 2020-11-12 11:38:37: Received MQTT message on 'homeassistant/nightlight_wesley' with data '{"brightness":21,"color":{"x":0.4599,"y":0.4106},"color_temp":454,"linkquality":69,"state":"OFF","update":{"state":"available"},"update_available":true}'
debug 2020-11-12 11:38:37: Received MQTT message on 'homeassistant/bridge/logging' with data '{"level":"info","message":"Update available for 'nightlight_wesley'"}'
debug 2020-11-12 11:38:37: Received MQTT message on 'homeassistant/bridge/logging' with data '{"level":"info","message":"MQTT publish: topic 'homeassistant/bridge/log', payload '{\"message\":\"Update available for 'nightlight_wesley'\",\"meta\":{\"device\":\"nightlight_wesley\",\"status\":\"available\"},\"type\":\"o
ta_update\"}'"}'
debug 2020-11-12 11:38:37: Received MQTT message on 'homeassistant/bridge/log' with data '{"message":"Update available for 'nightlight_wesley'","meta":{"device":"nightlight_wesley","status":"available"},"type":"ota_update"}'
info  2020-11-12 11:38:37: Successfully setup reporting for '0x842e14fffe633d92' - 1 - genLevelCtrl
debug 2020-11-12 11:38:37: Setup reporting for '0x842e14fffe633d92' - 1 - lightingColorCtrl
debug 2020-11-12 11:38:37: Received MQTT message on 'homeassistant/bridge/logging' with data '{"level":"info","message":"Successfully setup reporting for '0x842e14fffe633d92' - 1 - genLevelCtrl"}'
info  2020-11-12 11:38:37: Successfully setup reporting for '0x842e14fffe633d92' - 1 - lightingColorCtrl
info  2020-11-12 11:38:37: MQTT publish: topic 'homeassistant/bridge/event', payload '{"data":{"friendly_name":"nightlight_wesley","ieee_address":"0x842e14fffe633d92"},"type":"device_announce"}'
info  2020-11-12 11:38:37: MQTT publish: topic 'homeassistant/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"nightlight_wesley"},"type":"device_announced"}'
debug 2020-11-12 11:38:37: Received MQTT message on 'homeassistant/bridge/logging' with data '{"level":"info","message":"Successfully setup reporting for '0x842e14fffe633d92' - 1 - lightingColorCtrl"}'
debug 2020-11-12 11:38:37: Received MQTT message on 'homeassistant/bridge/logging' with data '{"level":"info","message":"MQTT publish: topic 'homeassistant/bridge/event', payload '{\"data\":{\"friendly_name\":\"nightlight_wesley\",\"ieee_address\":\"0x842e14fffe633d92\"},\"type\":\"device_announce\"}'"}'
debug 2020-11-12 11:38:37: Received MQTT message on 'homeassistant/bridge/event' with data '{"data":{"friendly_name":"nightlight_wesley","ieee_address":"0x842e14fffe633d92"},"type":"device_announce"}'
debug 2020-11-12 11:38:37: Received MQTT message on 'homeassistant/bridge/logging' with data '{"level":"info","message":"MQTT publish: topic 'homeassistant/bridge/log', payload '{\"message\":\"announce\",\"meta\":{\"friendly_name\":\"nightlight_wesley\"},\"type\":\"device_announced\"}'"}'
debug 2020-11-12 11:38:37: Received MQTT message on 'homeassistant/bridge/log' with data '{"message":"announce","meta":{"friendly_name":"nightlight_wesley"},"type":"device_announced"}'    
[... cut ...]
debug 2020-11-12 11:38:40: Received MQTT message on 'homeassistant/bridge/logging' with data '{"level":"info","message":"MQTT publish: topic 'homeassistant/nightlight_wesley', payload '{\"brightness\":21,\"color\":{\"x\":0.4599,\"y\":0.4106},\"color_temp\":454,\"linkquality\":72,\"state\":\"ON\",\"update\":{\"state\":\"available\"},\"update_available\":true}'"}'
debug 2020-11-12 11:38:40: Received MQTT message on 'homeassistant/nightlight_wesley' with data '{"brightness":21,"color":{"x":0.4599,"y":0.4106},"color_temp":454,"linkquality":72,"state":"ON","update":{"state":"available"},"update_available":true}'
```

lolorc, 2020-11-16:
@Wesleyl89 it looks like the same issue I have encountered, for some reason your bulb restarted and re-joinded the network. Since it's been rebooted it restarted with the light on, a bit like when the bulb is power on.

On your log we don't see if you have enabled the "availability" feature, can't see if a ping triggered the bulb reset, see https://www.zigbee2mqtt.io/information/availability.html

concerning your firmware update issue, please have a look at https://github.com/Koenkk/zigbee2mqtt/issues/2921 (last comments)

Wesleyl89, 2020-11-19:
> @Wesleyl89 it looks like the same issue I have encountered, for some reason your bulb restarted and re-joinded the network. Since it's been rebooted it restarted with the light on, a bit like when [...]

Ok, I did not have this enabled no. Will this feature fix this problem or is it just to check if it triggers something?

In the meantime I had another one at night, but with another bulb:
```
debug 2020-11-19 01:26:25: Device 'fronthall' announced itself
debug 2020-11-19 01:26:25: Setup reporting for '0x842e14fffe43987e' - 1 - genOnOff
info  2020-11-19 01:26:26: Successfully setup reporting for '0x842e14fffe43987e' - 1 - genOnOff
debug 2020-11-19 01:26:26: Setup reporting for '0x842e14fffe43987e' - 1 - genLevelCtrl
debug 2020-11-19 01:26:26: Received MQTT message on 'homeassistant/bridge/logging' with data '{"level":"info","message":"Successfully setup reporting for '0x842e14fffe43987e' - 1 - genOnOff"}'
info  2020-11-19 01:26:26: Successfully setup reporting for '0x842e14fffe43987e' - 1 - genLevelCtrl
info  2020-11-19 01:26:26: MQTT publish: topic 'homeassistant/bridge/event', payload '{"data":{"friendly_name":"fronthall","ieee_address":"0x842e14fffe43987e"},"type":"device_announce"}'
info  2020-11-19 01:26:26: MQTT publish: topic 'homeassistant/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"fronthall"},"type":"device_announced"}'
debug 2020-11-19 01:26:26: Received MQTT message on 'homeassistant/bridge/logging' with data '{"level":"info","message":"Successfully setup reporting for '0x842e14fffe43987e' - 1 - genLevelCtrl"}'
debug 2020-11-19 01:26:26: Received MQTT message on 'homeassistant/bridge/logging' with data '{"level":"info","message":"MQTT publish: topic 'homeassistant/bridge/event', payload '{\"data\":{\"friendly_name\":\"fronthall\",\"ieee_address\":\"0x842e14fffe43987e\"},\"type\":\"device_announce\"}'"}'
debug 2020-11-19 01:26:26: Received MQTT message on 'homeassistant/bridge/event' with data '{"data":{"friendly_name":"fronthall","ieee_address":"0x842e14fffe43987e"},"type":"device_announce"}'
debug 2020-11-19 01:26:26: Received MQTT message on 'homeassistant/bridge/logging' with data '{"level":"info","message":"MQTT publish: topic 'homeassistant/bridge/log', payload '{\"message\":\"announce\",\"meta\":{\"friendly_name\":\"fronthall\"},\"type\":\"device_announced\"}'"}'
debug 2020-11-19 01:26:26: Received MQTT message on 'homeassistant/bridge/log' with data '{"message":"announce","meta":{"friendly_name":"fronthall"},"type":"device_announced"}'
debug 2020-11-19 01:26:27: Received Zigbee message from 'fronthall', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2020-11-19 01:26:27: MQTT publish: topic 'homeassistant/fronthall', payload '{"brightness":89,"linkquality":144,"state":"ON","update":{"state":"available"},"update_available":true}'
debug 2020-11-19 01:26:27: Received MQTT message on 'homeassistant/bridge/logging' with data '{"level":"info","message":"MQTT publish: topic 'homeassistant/fronthall', payload '{\"brightness\":89,\"linkquality\":144,\"state\":\"ON\",\"update\":{\"state\":\"available\"},\"update_available\":true}'"}'
debug 2020-11-19 01:26:27: Received MQTT message on 'homeassistant/fronthall' with data '{"brightness":89,"linkquality":144,"state":"ON","update":{"state":"available"},"update_available":true}'
debug 2020-11-19 01:26:28: Received Zigbee message from 'fronthall', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":89}' from endpoint 1 with groupID 0
info  2020-11-19 01:26:28: MQTT publish: topic 'homeassistant/fronthall', payload '{"brightness":89,"linkquality":144,"state":"ON","update":{"state":"available"},"update_available":true}'
debug 2020-11-19 01:26:28: Received MQTT message on 'homeassistant/bridge/logging' with data '{"level":"info","message":"MQTT publish: topic 'homeassistant/fronthall', payload '{\"brightness\":89,\"linkquality\":144,\"state\":\"ON\",\"update\":{\"state\":\"available\"},\"update_available\":true}'"}'
debug 2020-11-19 01:26:28: Received MQTT message on 'homeassistant/fronthall' with data '{"brightness":89,"linkquality":144,"state":"ON","update":{"state":"available"},"update_available":true}'
debug 2020-11-19 01:26:30: Received Zigbee message from 'wasmachine', type 'attributeReport', cluster 'seMetering', data '{"currentSummDelivered":[0,443]}' from endpoint 1 with groupID 0
info  2020-11-19 01:26:30: MQTT publish: topic 'homeassistant/wasmachine', payload '{"current":0,"energy":4.43,"linkquality":78,"power":0,"state":"ON","voltage":220}'
debug 2020-11-19 01:26:30: Received MQTT message on 'homeassistant/bridge/logging' with data '{"level":"info","message":"MQTT publish: topic 'homeassistant/wasmachine', payload '{\"current\":0,\"energy\":4.43,\"linkquality\":78,\"power\":0,\"state\":\"ON\",\"voltage\":220}'"}'
debug 2020-11-19 01:26:30: Received MQTT message on 'homeassistant/wasmachine' with data '{"current":0,"energy":4.43,"linkquality":78,"power":0,"state":"ON","voltage":220}'
debug 2020-11-19 01:26:33: Received Zigbee message from 'wasmachine', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":222}' from endpoint 1 with groupID 0
info  2020-11-19 01:26:33: MQTT publish: topic 'homeassistant/wasmachine', payload '{"current":0,"energy":4.43,"linkquality":78,"power":0,"state":"ON","voltage":222}'
[... cut ...]
info  2020-11-19 01:26:40: MQTT publish: topic 'homeassistant/dressoir_1', payload '{"brightness":254,"color":{"x":0.476993298233858,"y":0.413675289942435},"color_temp":400,"linkquality":144,"state":"OFF","update":{"state":"available"},"update_available":true}'
debug 2020-11-19 01:26:40: Received MQTT message on 'homeassistant/bridge/logging' with data '{"level":"info","message":"MQTT publish: topic 'homeassistant/dressoir_1', payload '{\"brightness\":254,\"color\":{\"x\":0.476993298233858,\"y\":0.413675289942435},\"color_temp\":400,\"linkquality\":144,\"state\":\"OFF\",\"update\":{\"state\":\"available\"},\"update_available\":true}'"}'
```

lolorc, 2020-11-20:
no the availability check feature won't fix your issue since I had it enabled when it happened to me.

Since the issue happens quite often for you, you might want to upgrade one of your bulb with the test/beta firmware from Ikea, see the other link I gave in my previous comment. Since they are test/beta firmwares, that might brick your bulb, I can't be held responsible ;-)

chrstnwhlrt, 2020-11-30:
@Wesleyl89 I had the same behaviour of a ikea tradfri E27. Tried everything  but it looked like the bulb re-announced itself without any outer influences. I suspected the power supply but tested it using another bulb. That worked fine until now so I think the problematic bulb is simple done. Was a quite old one after all - I think first generation.

sjorge, 2020-11-30:
I have one GU10 that had the same behavior, it had a hummm too. It happened after the ota checkin too, I think the bulb might indeed crash and reset.

closed 2021-01-07 by github-actions[bot]