Koenkk/zigbee2mqtt #4879: tuya TS0601_thermostat power indicator gone

kozlik07, 2020-11-05:
Hello, there is a problem with tuya TS0601_thermostat

after updating to 1.16
the battery indicator disappears,% flashes in its place

## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

Koenkk, 2020-11-05:
Can you make a screenshot of this?

kozlik07, 2020-11-05:
[image]

kozlik07, 2020-11-06:
[image]

Koenkk, 2020-11-06:
What do you see under the "State" tab when clicking on the device?

kozlik07, 2020-11-06:
[image]

Koenkk, 2020-11-06:
Seems its caused because a battery report is missing, can you ask in this thread if others also experience this: #3821? (I don't have this device/knowledge about it)

boesing, 2020-11-06:
As far as I can tell (and I only have these valves for like 2-3 weeks), battery reporting never worked so far.

insipiens, 2020-11-07:
We're working on it, we believe the DP 366 reports low battery - which hasn't been configured yet in Z2M...still testing.

If you want to help I posted the clause needed to be added to report it, just need to make sure it solves it for all/most models of the valve.

groenmarsmannetje, 2020-12-14:
After 2 weeks of playing with my new thermostat, my thermostat reports low battery now in the "battery_low" boolean field, mostly when the motor is running to open/close valve. So if useful I can help to do some test with this.

As a coincidence I also saw a screen shot how battery is displayed in Tuya app. And it seems that they display the battery voltage on there... so probably that value comes from the device. And should be quite straightforward to transform that to a (rounded) percentage.

[image]

artkrz, 2021-01-03:
@groenmarsmannetje 

> After 2 weeks of playing with my new thermostat, my thermostat reports low battery now in the "battery_low" boolean field, mostly when the motor is running to open/close valve. So if useful I can help to do some test with this.

Do You mean that it reports low battery only when valve is moving the motor? Because that's what I think is happening, which gives a lot of state changes for this and basically makes it very unreliable since You can only judge by amount of state changes in the history?

groenmarsmannetje, 2021-01-04:
> Do You mean that it reports low battery only when valve is moving the motor? Because that's what I think is happening, which gives a lot of state changes for this and basically makes it very unreliable since You can only judge by amount of state changes in the history?

The battery low sensor starts flashing on and off when motor is moving, which is first signal that batteries are getting empty. This is no surprise because the motor needs the most power. After a day or two the sensor was constantly on. And shortly after that communication with device was lost and I needed to replace batteries. The battery percentage was never available.I had this experience now with two different Moes TRV devices.

artkrz, 2021-01-04:
> After a day or two the sensor was constantly on. And shortly after that communication with device was lost and I needed to replace batteries. 

Did state of `battery_low` was set permanently to `true` and cleared after you inserted new batteries when this happened?

Morphy99, 2021-01-13:
> Seems its caused because a battery report is missing, can you ask in this thread if others also experience this: #3821? (I don't have this device/knowledge about it)

[Apparently ](https://github.com/Koenkk/zigbee2mqtt/issues/3821#issuecomment-708114140) this device doesn't expose battery level just battery low. Probably needs removing.

Koenkk, 2021-01-13:
@Morphy99 thanks, removed it

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

groenmarsmannetje, 2021-01-13:
> > After a day or two the sensor was constantly on. And shortly after that communication with device was lost and I needed to replace batteries.
> 
> @groenmarsmannetje Was state of `battery_low` set permanently to `true` and cleared after you inserted new batteries when this happened?

Yes it was.

salopette, 2021-01-13:
Perhaps you can also correct the 5th point?
It's valve_detection

`<strong title="{&quot;access&quot;:7,&quot;name&quot;:&quot;state&quot;,&quot;property&quot;:&quot;valve_detection&quot;,&quot;type&quot;:&quot;binary&quot;,&quot;value_off&quot;:&quot;OFF&quot;,&quot;value_on&quot;:&quot;ON&quot;,&quot;value_toggle&quot;:&quot;TOGGLE&quot;}">state</strong>`

Koenkk, 2021-01-13:
@salopette what is showed in the state tab?

salopette, 2021-01-13:
`igbee2MQTT

WC-EG-TRV
About
Exposes
Bind
Reporting
State
Clusters
{
    "auto_lock": "MANUAL",
    "away_mode": "OFF",
    "away_preset_days": 1,
    "away_preset_temperature": 15,
    "battery_low": false,
    "boost_time": 0,
    "child_lock": "UNLOCKED",
    "comfort_temperature": 21,
    "current_heating_setpoint": 21.5,
    "eco_temperature": 15,
    "force": "normal",
    "holidays": [
        {
            "hour": 6,
            "minute": 0,
            "temperature": 20
        },
        {
            "hour": 8,
            "minute": 0,
            "temperature": 15
        },
        {
            "hour": 11,
            "minute": 30,
            "temperature": 15
        },
        {
            "hour": 12,
            "minute": 30,
            "temperature": 15
        },
        {
            "hour": 17,
            "minute": 30,
            "temperature": 20
        },
        {
            "hour": 22,
            "minute": 0,
            "temperature": 15
        }
    ],
    "last_seen": "2021-01-13T20:12:08.161Z",
    "linkquality": 135,
    "local_temperature": 21.5,
    "local_temperature_calibration": 0,
    "max_temperature": 25,
    "min_temperature": 15,
    "position": 0,
    "preset": "manual",
    "system_mode": "auto",
    "update": {
        "state": "idle"
    },
    "update_available": false,
    "week": "7",
    "window_detection": "OFF",
    "window_detection_params": {
        "minutes": 227,
        "temperature": 122
    },
    "workdays": [
        {
            "hour": 6,
            "minute": 0,
            "temperature": 20
        },
        {
            "hour": 8,
            "minute": 0,
            "temperature": 21
        },
        {
            "hour": 11,
            "minute": 30,
            "temperature": 19
        },
        {
            "hour": 12,
            "minute": 30,
            "temperature": 18
        },
        {
            "hour": 17,
            "minute": 30,
            "temperature": 20
        },
        {
            "hour": 22,
            "minute": 0,
            "temperature": 15
        }
    ]
}`

Koenkk, 2021-01-13:
There is no state for `valve_detection` yet, try turning it on/off once.

salopette, 2021-01-13:
`Debug Received MQTT message on 'zigbee2mqtt/WC-EG-TRV/set' with data '{"valve_detection":"OFF"}'
Debug Publishing 'set' 'valve_detection' to 'WC-EG-TRV'
Debug Received MQTT message on 'zigbee2mqtt/WC-EG-TRV/set' with data '{"valve_detection":"ON"}'
Debug Publishing 'set' 'valve_detection' to 'WC-EG-TRV'
Debug Received MQTT message on 'zigbee2mqtt/WC-EG-TRV/get' with data '{"valve_detection":""}'
Error No converter available for 'get' 'valve_detection' ()`

Koenkk, 2021-01-15:
@salopette can you provide the herdsman debug log when doing the `'zigbee2mqtt/WC-EG-TRV/set' with data '{"valve_detection":"OFF"}' ` and the same when setting the `max_temperature`?

groenmarsmannetje, 2021-01-15:
To me it is not clear what valve_detection should actually do.

salopette, 2021-01-15:
`Debug Received MQTT message on 'zigbee2mqtt/Flure-TRV/set' with data '{"valve_detection":"ON"}'
Debug Received MQTT message on 'zigbee2mqtt/Flure-TRV/set' with data '{"valve_detection":"OFF"}'
Debug Received MQTT message on 'zigbee2mqtt/Flure-TRV/get' with data '{"valve_detection":""}'
Debug Received MQTT message on 'zigbee2mqtt/Flure-TRV/set' with data '{"max_temperature":28}'
Debug Received Zigbee message from 'Flure-TRV', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,28],"type":"Buffer"},"datatype":2,"dp":103,"fn":0,"status":97,"transid":219}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Flure-TRV', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,28],"type":"Buffer"},"datatype":2,"dp":103,"fn":0,"status":97,"transid":220}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Flure-TRV', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,230],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":97,"transid":221}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Flure-TRV', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":107,"fn":0,"status":97,"transid":222}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Flure-TRV', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,17],"type":"Buffer"},"datatype":2,"dp":108,"fn":0,"status":97,"transid":223}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Flure-TRV', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[6,0,20,8,0,15,11,30,15,12,30,15,17,30,20,150,0,15],"type":"Buffer"},"datatype":0,"dp":112,"fn":0,"status":97,"transid":224}' from endpoint 1 with groupID 0
Debug Received MQTT message on 'zigbee2mqtt/Flure-TRV/set' with data '{"max_temperature":29}'
Debug Received MQTT message on 'zigbee2mqtt/Flure-TRV/set' with data '{"min_temperature":14}'
Debug Received Zigbee message from 'Flure-TRV', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[6,0,20,8,0,15,11,30,15,12,30,15,17,30,20,22,0,15],"type":"Buffer"},"datatype":0,"dp":113,"fn":0,"status":97,"transid":225}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Flure-TRV', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,15],"type":"Buffer"},"datatype":2,"dp":114,"fn":0,"status":97,"transid":226}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Flure-TRV', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,29],"type":"Buffer"},"datatype":2,"dp":103,"fn":0,"status":97,"transid":228}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Flure-TRV', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,230],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":97,"transid":229}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Flure-TRV', type 'commandSetDataResponse', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,14],"type":"Buffer"},"datatype":2,"dp":102,"fn":0,"status":97,"transid":230}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Flure-TRV', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,230],"type":"Buffer"},"datatype":2,"dp":2,"fn":0,"status":97,"transid":231}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Flure-TRV', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,22],"type":"Buffer"},"datatype":2,"dp":107,"fn":0,"status":97,"transid":232}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Flure-TRV', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,17],"type":"Buffer"},"datatype":2,"dp":108,"fn":0,"status":97,"transid":233}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Flure-TRV', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[6,0,20,8,0,15,11,30,15,12,30,15,17,30,20,150,0,15],"type":"Buffer"},"datatype":0,"dp":112,"fn":0,"status":97,"transid":234}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Flure-TRV', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[6,0,20,8,0,15,11,30,15,12,30,15,17,30,20,22,0,15],"type":"Buffer"},"datatype":0,"dp":113,"fn":0,"status":97,"transid":235}' from endpoint 1 with groupID 0
Debug Received Zigbee message from 'Flure-TRV', type 'commandGetData', cluster 'manuSpecificTuya', data '{"data":{"data":[0,0,0,15],"type":"Buffer"},"datatype":2,"dp":114,"fn":0,"status":97,"transid":236}' from endpoint 1 with groupID 0`

Koenkk, 2021-01-16:
@salopette does setting the valve detection actually work? (I have no clue what this should do)

salopette, 2021-01-16:
I haven't found out anything yet. I don't know what this function is for.
It's not necessary for me, maybe it can be done away with completely?

groenmarsmannetje, 2021-01-16:
When the device initially is started it runs a sequence to determine the minimum and maximum position of the valve. That could be considered as some kind of “valve_detection” but if this is also intention of this switch is not clear to me. I have not seen anything happening when I tried the switch.

k-zakhariy, 2021-01-19:
i'm still having errors like this one, even after switched to edge version:
```zigbee-herdsman-converters:tuyaThermostat: NOT RECOGNIZED DP #115 with data {"status":132,"transid":214,"dp":115,"datatype":1,"fn":0,"data":{"type":"Buffer","data":[0]}}
```

frogale, 2021-02-01:
So there is no way how to get battery state from Tuya? Uff..

closed 2021-03-11 by github-actions[bot]

CWempe, 2021-05-01:
> I haven't found out anything yet. I don't know what this function is for.
> It's not necessary for me, maybe it can be done away with completely?

I got some information about this: https://github.com/Koenkk/zigbee2mqtt/issues/7270#issuecomment-830576992

It is a "valve fault detection" to indicate if the vault would stay closed (radiator not heating) even when the TRV is "open".