Koenkk/zigbee2mqtt #4884: Version 1.16.0 overall slow network response

Imaginous, 2020-11-06:
## What happened
When pressing a zigbee button or sending in MQTT command to a lamp takes about 1.5 seconds. When sending 2 commands directly after one another to turn on 2 lights. They will turn on about 1.5 - 2.0 seconds after each other.

This started since version 1.16.0. Copied back my 1.15.0 version and everything runs smoothly again.

## What did you expect to happen
Thate the commands would be almost instantanious.

## How to reproduce it (minimal and precise)
Upgrade to version 1.16.0 -> zigbee network becomes slow.
Restore to version 1.15.0 -> zigbee network is fast again.

No devices have been added or changed between the upgrade.

## Debug info
Zigbee2MQTT version: 1.16.0
Adapter hardware: Conbee II
Adapter firmware version: 0x26660700

Koenkk, 2020-11-06:
@chrishae has anything changed in the Conbee implementation that could cause this?

Imaginous, 2020-11-07:
@Koenkk is there a possibility for you to build a 1.16.1 version with the old 1.15.0 Deconz code in it?

I can then try if that runs smoothly and maybe pinpoint the problem a bit better.

2x4ever, 2020-11-07:
I can confirm that Zigbee network became much slower in 1.6.X version.
I've used conbeeII adapter and light switch responses are slowed down some time up to 5-10 seconds, while in 1.5 it was around 10s of milliseconds.
I thought it was due to the fact that I had a lot of devices (around 30 actually) and conbeeII adapter works with Baudrate of 38400.
Then i switched the network to ZZH, repaired all devices, but nothing changed.
I noticed that log files became flooded by repeated 'attrubuteReports' from the same device. It looks like it is just multiplied several times. And the more network is running the more reports are multiplied.
https://pastebin.com/Rurtm4LK
Same is for a dimmer action.
This is for example dimmer ON action response:
 https://pastebin.com/RkXjYBEP

One ON action leads to network flood with attributeReports.
Being repeated several times (2-3 switch presses) slows down reaction to 5-10 seconds.

Koenkk, 2020-11-07:
@Imaginous if running bare metal, you can downgrade the zigbee-herdsman library (which contain the driver code) to the 1.15.0 version when executing in `/opt/zigbee2mqtt` `npm install zigbee-herdsman@0.13.11`

@2x4ever does downgrading to 1.15.0 fix your issue?

srnoth, 2020-11-08:
Can confirm the same slowdown on 1.6.X with a conbee II adapter. Looking at the MQTT feed, there would be anywhere from 0.5 to several seconds delay between pressing a button, and the corresponding MQTT message showing up.

Return to 1.5.X and the response time immediately returned to normal (MQTT message appears almost immediately as a button is pressed).

Did not have a chance to do any further troubleshooting but I'd be happy to run any diagnostics that would help narrow this down.

Imaginous, 2020-11-08:
@Koenkk  I have done a couple of test. I run bare metal, so I have downgraded herdsman to 0.13.11 in my 1.16.0 installation.

As I can conclude, the difference must be somewhere in the herdsman code. Since the problem disappears after downgrading the herdsman code.

**Note:** When the button is pressed I send 2 individual MQTT commands to the 2 gang dimmer module to turn on each lamp.

BTW: is there a json command to set both states at the same time?

Here are the results of different installations.

## Hybrid version: 1.16.0 with downgraded herdsman 0.13.11, reacts normal

### Single press on, on the Ikea button:
`Nov 08 12:00:21 domoticapi1 npm[24022]: Zigbee2MQTT:info  2020-11-08 12:00:21: MQTT publish: topic 'zigbee2mqtt/Alarm_Knop_IKEA', payload '{"action":"on","battery":74,"click":"on","linkquality":5,"update":{"state":"idle"},"update_available":false}'
Nov 08 12:00:22 domoticapi1 npm[24022]: Zigbee2MQTT:info  2020-11-08 12:00:22: MQTT publish: topic 'zigbee2mqtt/Slaapkamer1_Verlichting', payload '{"brightness":0,"brightness_l1":254,"brightness_l2":254,"linkquality":14,"state":"OFF","state_l1":"ON","state_l2":"OFF"}'
Nov 08 12:00:22 domoticapi1 npm[24022]: Zigbee2MQTT:info  2020-11-08 12:00:22: MQTT publish: topic 'zigbee2mqtt/Slaapkamer1_Verlichting', payload '{"brightness":0,"brightness_l1":254,"brightness_l2":254,"linkquality":14,"state":"OFF","state_l1":"ON","state_l2":"ON"}'
Nov 08 12:00:22 domoticapi1 npm[24022]: Zigbee2MQTT:info  2020-11-08 12:00:22: MQTT publish: topic 'zigbee2mqtt/Slaapkamer1_Verlichting', payload '{"brightness":0,"brightness_l1":254,"brightness_l2":254,"linkquality":14,"state":"OFF","state_l1":"ON","state_l2":"ON"}'
Nov 08 12:00:22 domoticapi1 npm[24022]: Zigbee2MQTT:info  2020-11-08 12:00:22: MQTT publish: topic 'zigbee2mqtt/Slaapkamer1_Verlichting', payload '{"brightness":0,"brightness_l1":254,"brightness_l2":254,"linkquality":14,"state":"OFF","state_l1":"ON","state_l2":"ON"}'
Nov 08 12:00:25 domoticapi1 npm[24022]: Zigbee2MQTT:info  2020-11-08 12:00:25: MQTT publish: topic 'zigbee2mqtt/Slaapkamer1_Verlichting', payload '{"brightness":0,"brightness_l1":254,"brightness_l2":254,"linkquality":14,"state":"OFF","state_l1":"ON","state_l2":"ON"}'
Nov 08 12:00:25 domoticapi1 npm[24022]: Zigbee2MQTT:info  2020-11-08 12:00:25: MQTT publish: topic 'zigbee2mqtt/Slaapkamer1_Verlichting', payload '{"brightness":0,"brightness_l1":254,"brightness_l2":254,"linkquality":14,"state":"OFF","state_l1":"ON","state_l2":"ON"}'`

### Single press off on the Ikea button:

`Nov 08 12:01:11 domoticapi1 npm[24022]: Zigbee2MQTT:info  2020-11-08 12:01:11: MQTT publish: topic 'zigbee2mqtt/Alarm_Knop_IKEA', payload '{"action":"off","battery":74,"click":"off","linkquality":5,"update":{"state":"idle"},"update_available":false}'
Nov 08 12:01:11 domoticapi1 npm[24022]: Zigbee2MQTT:info  2020-11-08 12:01:11: MQTT publish: topic 'zigbee2mqtt/Slaapkamer1_Verlichting', payload '{"brightness":0,"brightness_l1":254,"brightness_l2":254,"linkquality":14,"state":"OFF","state_l1":"OFF","state_l2":"ON"}'
Nov 08 12:01:11 domoticapi1 npm[24022]: Zigbee2MQTT:info  2020-11-08 12:01:11: MQTT publish: topic 'zigbee2mqtt/Slaapkamer1_Verlichting', payload '{"brightness":0,"brightness_l1":254,"brightness_l2":254,"linkquality":14,"state":"OFF","state_l1":"OFF","state_l2":"ON"}'
Nov 08 12:01:12 domoticapi1 npm[24022]: Zigbee2MQTT:info  2020-11-08 12:01:12: MQTT publish: topic 'zigbee2mqtt/Slaapkamer1_Verlichting', payload '{"brightness":0,"brightness_l1":254,"brightness_l2":254,"linkquality":14,"state":"OFF","state_l1":"OFF","state_l2":"OFF"}'`

## Version: 1.16.1, reacts slow

### Single press on, on the Ikea button:

`Nov 08 12:52:40 domoticapi1 npm[31538]: Zigbee2MQTT:info  2020-11-08 12:52:40: MQTT publish: topic 'zigbee2mqtt/Alarm_Knop_IKEA', payload '{"action":"on","battery":87,"click":"on","linkquality":200,"update":{"state":"idle"},"update_available":false}'
Nov 08 12:52:41 domoticapi1 npm[31538]: Zigbee2MQTT:info  2020-11-08 12:52:41: MQTT publish: topic 'zigbee2mqtt/Slaapkamer1_Verlichting', payload '{"brightness":0,"brightness_l1":254,"brightness_l2":254,"linkquality":200,"state":"OFF","state_l1":"ON","state_l2":"OFF"}'
Nov 08 12:52:41 domoticapi1 npm[31538]: Zigbee2MQTT:info  2020-11-08 12:52:41: MQTT publish: topic 'zigbee2mqtt/Slaapkamer1_Verlichting', payload '{"brightness":0,"brightness_l1":254,"brightness_l2":254,"linkquality":198,"state":"OFF","state_l1":"ON","state_l2":"ON"}'
Nov 08 12:52:42 domoticapi1 npm[31538]: Zigbee2MQTT:info  2020-11-08 12:52:42: MQTT publish: topic 'zigbee2mqtt/Slaapkamer1_Verlichting', payload '{"brightness":0,"brightness_l1":254,"brightness_l2":254,"linkquality":200,"state":"OFF","state_l1":"ON","state_l2":"ON"}'
Nov 08 12:52:42 domoticapi1 npm[31538]: Zigbee2MQTT:info  2020-11-08 12:52:42: MQTT publish: topic 'zigbee2mqtt/Slaapkamer1_Verlichting', payload '{"brightness":0,"brightness_l1":254,"brightness_l2":254,"linkquality":200,"state":"OFF","state_l1":"ON","state_l2":"ON"}'
Nov 08 12:52:45 domoticapi1 npm[31538]: Zigbee2MQTT:info  2020-11-08 12:52:45: MQTT publish: topic 'zigbee2mqtt/Slaapkamer1_Verlichting', payload '{"brightness":0,"brightness_l1":254,"brightness_l2":254,"linkquality":200,"state":"OFF","state_l1":"ON","state_l2":"ON"}'
Nov 08 12:52:45 domoticapi1 npm[31538]: Zigbee2MQTT:info  2020-11-08 12:52:45: MQTT publish: topic 'zigbee2mqtt/Slaapkamer1_Verlichting', payload '{"brightness":0,"brightness_l1":254,"brightness_l2":254,"linkquality":200,"state":"OFF","stat [... cut ...]
Nov 08 12:41:36 domoticapi1 npm[30463]: Zigbee2MQTT:info  2020-11-08 12:41:36: MQTT publish: topic 'zigbee2mqtt/Slaapkamer1_Verlichting', payload '{"brightness":0,"brightness_l1":254,"brightness_l2":254,"linkquality":199,"state":"OFF","state_l1":"OFF","state_l2":"ON"}'
Nov 08 12:41:36 domoticapi1 npm[30463]: Zigbee2MQTT:info  2020-11-08 12:41:36: MQTT publish: topic 'zigbee2mqtt/Slaapkamer1_Verlichting', payload '{"brightness":0,"brightness_l1":254,"brightness_l2":254,"linkquality":199,"state":"OFF","state_l1":"OFF","state_l2":"OFF"}''

Koenkk, 2020-11-08:
@chrishae there seems to be some regression in performance for the Conbee II in herdsman 0.13.11 - 0.13.26, can you investigate this?

Imaginous, 2020-11-08:
@Koenkk I have found another strange behavior, I don't think it is directly related. 
When I send 2 MQTT commands to turn on each light separately they turn on directly after one another with version 1.15.0. 

However when I send:
`{
  "state_l1": "on", "state_l2": "on"
}`

So turn both lights on in one command, there is the same 1-1.5 second delay between turning on each lamp as in version 1.16.0, but in this case it also happens in version 1.15.0.

Don't know if it is related, but I maybe it is.

ChrisHae, 2020-11-10:
I have slowed down data transmit a bit to improve stability when sending many commands at once (Some devices did not respond when sending many requests to fast). I could tweak the times to make it a bit faster again. Delay should be around 1sec. 3 or more seconds would indicate a very busy network I think. 

You could also use group commands when controlling many lights at once. This should be much faster.

@Imaginous How do I use that command:` { "state_l1": "on", "state_l2": "on" }`
to control two lights with one command. I would like to test this. How is the exact syntax and topic?
in z2mqtt doc I only found this: `zigbee2mqtt/[FRIENDLY_NAME]/set`

Imaginous, 2020-11-10:
@ChrisHae would it be possible to make the delay a parameter in the settings of zigbee2mqtt?

As a programmer I think delays are best avoided. 
I don't know if there is a way to check if a command has been send? 

If you can point me to the source code, I can have a look at it. 

I send commands directly after one another and never experienced any problems. On the other hand the zigbee protocol should take care of resends. 

The command is send to a single zigbee device with 2 lamps.
I send it to: zigbee2mqtt/[FRIENDLY_NAME]/set
And a json payload of: `{ "state_l1": "on", "state_l2": "on" }`

Imaginous, 2020-11-10:
@Koenkk Tonight I have tested V1.16.1 with a Slaesh stick.
Then everything works perfectly normal. So it is clearly the timing adjustment made by @ChrisHae.

I don't want to "bash" anyone, I'm just wanting to help and improve the great work you guys do!

ChrisHae, 2020-11-11:
There were Issues of user who sent many requests (about 10 -20) nearly at the same time (https://github.com/Koenkk/zigbee2mqtt/issues/4215) . Some devices did not respond or the messages did not arrived there. I think it is best to use group commands then but never the less I think there should be some time between the messages so that the devices and the coordinator can better handle it.

So did not hard coded a delay between the sending of the messages but instead the coordinator waited until a data request got confirmed before sending the next one. This usualy took about 800ms.

Now I have added a parameter that defines the delay between two requests. I have halved the time and it still works pretty stable. You could test it if the speed is acceptable now. It is in the newest herdsman commit.

It would be also possible to let the user change this parameter in the configuration if @koenkk would let me add this parameter to the configuration.yaml. Let's call it packet_spacing or send_delay or so.

srnoth, 2020-11-11:
From an end user perspective 800ms is a massive delay. E.g. I did not get full buy-in from the wife until I changed "multiple_press_timeout" on the Philips Hue remotes from 250ms to 0.

250ms meant the difference between walking into a room, turning on a light and it almost immediately turning on, vs. pressing and consciously waiting for the light to turn on.

I agree that a user configurable value would make more sense.

2x4ever, 2020-11-11:
> @Imaginous if running bare metal, you can downgrade the zigbee-herdsman library (which contain the driver code) to the 1.15.0 version when executing in `/opt/zigbee2mqtt` `npm install zigbee-herdsman@0.13.11`
> 
> @2x4ever does downgrading to 1.15.0 fix your issue?

Sorry for the long answer. Was looking for a way to downgrade the hass.io add-on version.
Switched to bare metal version, downgraded to 1.15.0 and problem disappeared.
So I don't think it is only conbeeii problem, because I switched to zzh stick.
I think that reduced conbee transmit rate is only making it worse (triggers it earler).

ChrisHae, 2020-11-11:
@2x4ever I'm speaking about a delay from about 400ms to 1s. Some User reported several seconds. This should be another issue here. Also pressing a button on a zigbee remote should not be affected by this. Remotes are bound and communicate directly to the lights.

srnoth, 2020-11-11:
@ChrisHae not all use cases allow remotes to be bound directly to lights. Previous to this change, using a remote to trigger automations in HASS was still pretty snappy, especially with "multiple_press_timeout" set to 0ms.

Imaginous, 2020-11-11:
@ChrisHae My Ikea button does not communicate directly to the lights. The button press is caught by my software and then the 2 commands are send to the lights. I would say that the delay with the 1.16.0 version might be 1 second. I did not time it.

How can I clone / add only the latest Herdsman version into the existing version?
Do I just copy the git files to the /opt/zigbee2mqtt folder and re-run npm ci?

ChrisHae, 2020-11-12:
@Imaginous I can't say if this also works. I cloned the zigbee-herdsman repository to my raspberry and added the path to /opt/zigbee2mqtt/lib/zigbee.js
`const ZigbeeHerdsman = require('/home/pi/zigbee-herdsman');`

Or you wait until Koenkk merged this into the newest version of zigbee2mqtt

Koenkk, 2020-11-12:
@ChrisHae I don't see any new commits to https://github.com/Koenkk/zigbee-herdsman/commits/master ?

ChrisHae, 2020-11-12:
@Koenkk Oh sorry forgot to push. Now it's online.

Koenkk, 2020-11-12:
It's in the latest dev now.

@ChrisHae next time you make a change to zigbee-herdsman, you can trigger a new release by going to https://github.com/Koenkk/zigbee-herdsman/actions?query=workflow%3A%22Create+new+release%22 , press "Run workflow" -> press "Run workflow" (you can leave Release type patch as is). This will automatically create a new release and make a PR to Zigbee2MQTT to update zigbee-herdsman (which I will get a notification for).

juslex, 2020-11-15:
I recently left the zzh and returned to conbee 2, however, as I was used to the delay of the zzh, when I returned to conbee 2, I noticed some delay of this in relation to the zzh. What's the reason for that? (Ps. I'm running the herdsman version to 0.13.11 with the conbee.

Imaginous, 2020-11-15:
@Koenkk, @ChrisHae 

Today I installed the latest dev version of 1.16.1. If I understood correctly this should contain the latest modifications of Chris.

`git checkout HEAD -- npm-shrinkwrap.json`
`git fetch`
`git checkout dev # Change 'dev' to 'master' to switch back to the release version`
`git pull`
`npm ci`

This version has still got the slow response with the Conbee II stick. Maybe I did something wrong, please let me know if so.

juslex, 2020-11-15:
After my previous post, To fix the home assistant bug, I updated yesterday the dev version, which removed the herdsman version to 0.13.11. And I noticed that the slowness continues. I went back to the herdsman version to 0.13.11.

srnoth, 2020-11-16:
As long as there are still delays added (even if they are halved) there will be a noticeable slowness. Even the adjustable 250ms delay on the Hue Dimmer Remote was noticeable. 

@ChrisHae @Koenkk do you think we will be able to get a user-customizable option to remove the delays added with this fix? It seems a pity to negatively impact usability for use cases that benefit from response time over reliability, just to fix the use case where someone wants to send commands to many zigbee devices simultaneously. Is there a reason they can't use group messaging in that scenario?

ChrisHae, 2020-11-17:
@Imaginous The version 1.16.1 is released 16 days ago, so I think it does not contain the new code.
@juslex how bis is the delay of that you talk? about 1 second, or severall seconds?

I think it would be nice if I could add a parameter to the configuration.yaml. Where the user could chosse a delay between 10 to 1000ms. it could default to 200 if not specified. @Koenkk Is it possible for me to do that? in my driver.ts of deconz adapter there is already a variable for it.

Koenkk, 2020-11-18:
@ChrisHae  done, the delay is now passed (https://github.com/Koenkk/zigbee-herdsman/blob/e1eaaf0c20418463ab531cc929bf12dbbd8e66f4/src/adapter/deconz/adapter/deconzAdapter.ts#L52). You can specify by adding to `configuration.yaml`:

```yaml
advanced:
  adapter_delay: 200
```

Note users: this will not work until @ChrisHae starts using the delay in the implementation.

juslex, 2020-11-18:
> @Imaginous The version 1.16.1 is released 16 days ago, so I think it does not contain the new code.
> @juslex how bis is the delay of that you talk? about 1 second, or severall seconds?
>
> I think it [...]

Dear @ChrisHae 

I was using zzh instead of conbee II, however, I had several problems with it after some time of use. So I decided to go back to the conbee II and started to notice a longer delay - around 800 ms on most cases and sometimes 1s - when compared to zzh when executing this automation the delay is around 200ms.

Automation virtual three-away using zemismart HGZB-42 and HGZB-43 switches:

- alias: three away
  description: ''
  trigger:
  - entity_id: &id001
    - switch.room_switch_center
    - switch.room_switch_bottom
    platform: state
    to:
    - 'on'
    - 'off'
  action:
  - service: homeassistant.turn_{{trigger.to_state.state}}
    data:
      entity_id: *id001
  mode: single

ChrisHae, 2020-11-20:
@koenkk thanks, I will implement this as soon as possible. Let's see if this helps to get the desired behavior for most users.

ChrisHae, 2020-11-20:
Ok, the parameter is implemented. add
```
advanced:
  adapter_delay: 200
```
to your configuration.yaml
it can take values from 50 to 1200 (It is the delay in ms). If you don't add the parameter the default value is 300ms.

Note that you need the latest dev-branch of zigbee2mqtt. I triggered a new release for zigbee-herdsman so I think it will be soon in z2mqtt-dev.

Koenkk, 2020-11-20:
@ChrisHae merged so it is in dev now, (for Home Assistant edge wait 1 hour)

Imaginous, 2020-11-20:
@ChrisHae is there a specific reason you chose 50 as minimum and not 0 or 1?

iAcki, 2020-11-20:
Hi @Koenkk && @ChrisHae,

I've pulled the dev branch and now it's really snappy again compared to the behavior before. Thank you very much for fixing it so fast. :)

bkupidura, 2020-11-20:
So i also switched to `latest-dev` container, set `adapter_delay: 50` under advanced - but network is still slower than on 1.15(?). This is especially visible when i turn on HA group with multiple (3) zigbee bulbs.

Previously, all bulbs were turned on in same time (or at least my eye was not able to catch difference), and currently i can easy see that bulbs are turned on in sequence.

Is this possible that `adapter_delay` is setting just one of multiple (used to send commands to nodes) delays?

Imaginous, 2020-11-21:
As @bkupidura mentioned, I also pulled the latest dev version. When adding adapter_delay: 50 to the configuration.yaml and restarting zigbee2mqtt it is a bit faster, but still way slower then it was before.

@ChrisHae , what did you change? I would like to review the changes, as I don't understand what is causing the delay. 
If you just changed the delay time then it would make sense that if you put back the old value (or let us put in the value via the new property) it is as fast as it was before. If you have added extra delay code, it might be an ID to encase those in an IF statement. If the configuration property = 0, we ignore the new delays.

Also it seems logical to me that the delay value is only restricted by it's type. Let's say a positive 16bits integer.

As said before, I just want to help. I love the work you have done... it's just that this is a regression for some people, so if we can make it as snappy as it was again via a parameter both sides would be happy.

ChrisHae, 2020-11-24:
Thx for your feedback @all 
@Imaginous @bkupidura wait for new z2mqtt release and then try a delay value of 0 in the configuration.yaml
It is a speedup for nearly every internal queues and should be as fast as before.

Imaginous, 2020-12-02:
@Koenkk, @ChrisHae 

I have test 1.16.2 with adapter_delay: 0 in the advanced section of the config. It is still slow. 
I have a feeling that the adapter_delay setting is not working at all.

juslex, 2020-12-02:
Same situation here

bkupidura, 2020-12-02:
Have same feelings, this is why 12d ago i was asking if this delay is controlling all delays used to send commands.

srnoth, 2020-12-03:
Was this change included in the new 1.16.2.X releases? Didn't see anything mentioned in the release notes.

Koenkk, 2020-12-03:
I've added some extra debug logging to check if it is working correctly. Please provide me the herdsman debug logging when starting z2m.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

bkupidura, 2020-12-04:
@Koenkk 
```
> zigbee2mqtt@1.16.2-dev start /app
> node index.js

ESC[32mZigbee2MQTT:info ESC[39m 2020-12-04 22:53:55: Logging to console only'
ESC[32mZigbee2MQTT:info ESC[39m 2020-12-04 22:53:55: Starting Zigbee2MQTT version 1.16.2-dev (commit #47d26c8)
ESC[32mZigbee2MQTT:info ESC[39m 2020-12-04 22:53:55: Starting zigbee-herdsman (0.13.39)
2020-12-04T21:53:55.335Z zigbee-herdsman:deconz:driver Set delay to 25
CREATED DECONZ ADAPTER
2020-12-04T21:53:55.337Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[<Key>],"panID":6754,"extendedPanID":[<ePanId>],"channelList":[25]},"serialPort":{"baudRate":38400,"path":"/dev/ttyACM0","adapter":"deconz"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"concurrent":null,"delay":25}}'
2020-12-04T21:53:55.339Z zigbee-herdsman:deconz:driver Opening with /dev/ttyACM0
2020-12-04T21:53:55.342Z zigbee-herdsman:deconz:driver Serialport opened
2020-12-04T21:53:55.342Z zigbee-herdsman:controller:log Started with result 'resumed'
2020-12-04T21:53:55.342Z zigbee-herdsman:controller:log Injected database: true, adapter: true
2020-12-04T21:53:55.344Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2020-12-04T21:53:55.344Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2020-12-04T21:53:55.344Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2020-12-04T21:53:55.344Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2020-12-04T21:53:55.344Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2020-12-04T21:53:55.344Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2020-12-04T21:53:55.437Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 1 paramId: 1
2020-12-04T21:53:55.440Z zigbee-herdsman:deconz:driver query aps data indication
```

Do you need anything more?

I dont see any difference between delay 0 and 25.

Koenkk, 2020-12-06:
@bkupidura I don't understand why 0 is not working yet, what is logged when you set it to 1?

bkupidura, 2020-12-06:
Start log with delay=1:
```
Zigbee2MQTT:info  2020-12-06 19:36:05: Logging to console only'                                                                                                                                                                      Zigbee2MQTT:info  2020-12-06 19:36:05: Starting Zigbee2MQTT version 1.16.2-dev (commit #47d26c8)                                                                                                                                     Zigbee2MQTT:info  2020-12-06 19:36:05: Starting zigbee-herdsman (0.13.39)                                                                                                                                                            2020-12-06T18:36:05.795Z zigbee-herdsman:deconz:driver Set delay to 1                                                                                                                                                                CREATED DECONZ ADAPTER                                                                                                                                                                                                               2020-12-06T18:36:05.796Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[Key],"panID":6754,"extendedPanID":[extendedPanID],"channelList":[25]},"serialPort":{"baudRate":38400,"path":"/dev/ttyACM0","adapter":"deconz"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"concurrent":null,"delay":1}}'                                                                                                                                                             2020-12-06T18:36:05.798Z zigbee-herdsman:deconz:driver Opening with /dev/ttyACM0                                                                                                                                                     2020-12-06T18:36:05.801Z zigbee-herdsman:deconz:driver Serialport opened                                                                                                                                                             2020-12-06T18:36:05.802Z zigbee-herdsman:controller:log Started with result 'resumed'                                                                                                                                                2020-12-06T18:36:05.802Z zigbee-herdsman:controller:log Injected database: true, adapter: true                                                                                                                                       2020-12-06T18:36:05.803Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010                                                                                                                                           2020-12-06T18:36:05.803Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2020-12-06T18:36:05.803Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010                                                                                                                                           2020-12-06T18:36:05.804Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2020-12-06T18:36:05.804Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010                                                                                                                                           2020-12-06T18:36:05.804Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2020-12-06T18:36:05.897Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 1 paramId: 1                                                                                                                   2020-12-06T18:36:05.901Z zigbee-herdsman:deconz:driver query aps data indication                                                                                                                                                     2020-12-06T18:36:05.903Z zigbee-herdsman:deconz:frameParser MAC: 0x00212effff054b79                                                                                                                                                  2020-12-06T18:36:05.997Z zigbee-herdsman:deconz:driver send read parameter request from queue. seqNr: 3 paramId: 7                                                                                                                   2020-12-06T18:36:06.000Z zigbee-herdsman:deconz:frameParser NWKADDR: 0                                                                                                                                                               Zigbee2MQTT:info  2020-12-06 19:36:06: zigbee-herdsman started
```

Logs from turning on group with delay=1:
```
2020-12-06T18:39:22.602Z zigbee-herdsman:controller:endpoint Command 0xec1bbdfffe023c7b/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
2020-12-06T18:39:22.602Z zigbee-herdsman:deconz:adapter zclFrame.payload:
[... cut ...] ndpoint":1,"linkquality":130,"groupID":null}'
2020-12-06T18:39:23.526Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-12-06T18:39:23.526Z zigbee-herdsman:deconz:adapter response received Zigbee2MQTT:info
2020-12-06 19:39:23: MQTT publish: topic 'z2ml0/light_kitchen_center_3', payload '{"brightness":254,"last_seen":1607279963526,"linkquality":130,"state":"ON","update":{"state":"idle"},"update_available":false}'
```

Koenkk, 2020-12-06:
@bkupidura looks good, so with 1 you still have delays?

bkupidura, 2020-12-06:
Yes, delays are still here - its especially visible with this group, as those are bulbs in single "lamp". But based on log timestamps:
2020-12-06T18:39:23.025Z Light1 turned on
2020-12-06T18:39:23.127Z Light2 turned on 
2020-12-06T18:39:23.526Z Light3 turned on

Its ~500ms delay (not 1ms as in config) - and turning on 3 lights in 0.5s is definitely received by human eye as "in sequence" ;)

Koenkk, 2020-12-07:
@chrishae can you investigate this further. It looks like the delay option is set correctly but somehow is not honored everywhere.

ChrisHae, 2020-12-08:
I see the delay option does not have the desired effect. So I will do some rework and change the behaviour of the delay setting of 0 to the old behaviour of  version 1.15.0. I think it will be done tomorrow or some other day this week.

ChrisHae, 2020-12-09:
Ok, adapter_delay: 0 in configuration.yaml is now the same behaviour as it was in version 1.15.0 and before. Therefore the delays should be gone. (available with new release)
for explanation:

> adapter_delay: 0 = minimal/no delay
> adapter_delay: 1 .. 50 = small delay (may improve devices not function as expected)
> adapter_delay: 51 .. 1200 = delay up to 1,2 seconds (use this if you experience some devices not responding or other errors) 

please note: delay of 0 does not mean literally "no delay".

Koenkk, 2020-12-09:
@ChrisHae fixed, Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Rene-Sackers, 2020-12-09:
@Koenkk is this available yet? I can't find any relevant commits or pull requests and I see no `dev` branch. I reinstalled edge by following the instructions, including reloading the add-ons page, I configured the device_delay to several different values, but no effect.

Koenkk, 2020-12-10:
@Rene-Sackers can you provide the herdsman debug log of starting Zigbee2MQTT?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Rene-Sackers, 2020-12-10:
> @Rene-Sackers can you provide the herdsman debug log of starting Zigbee2MQTT?
> 
> To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

I'm running it as an addon on HASS, the link you have says `Set "zigbee_herdsman_debug": true in the configuration.`, but I don't know what `the configuration` is. This is JSON, but `the configuration` that I know is the yaml file you get under the addon in the HASS interface. What configuration file am I supposed to change?

juslex, 2020-12-10:
> I'm running it as an addon on HASS, the link you have says `Set "zigbee_herdsman_debug": true in the configuration.`, but I don't know what `the configuration` is. This is JSON, but `the [...]

Do you user zigbee2mqtt assistant?

If yes, mark here in status tab:

[image]

Rene-Sackers, 2020-12-10:
I managed to enable it by sending the MQTT message, but how do I read the herdsman logs? https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging states `Important: this is not logged to the log files and is only available on the STDOUT.`. I can attach to the zigbee2mqtt docker container, but that only gives me the same log lines that I do see in the log files. How do I get the STDOUT of herdsman?

DenisBY, 2020-12-11:
> I managed to enable it by sending the MQTT message, but how do I read the herdsman logs? https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging states `Important: this is [...]

Try to use `docker logs <container_id>`

Rene-Sackers, 2020-12-11:
I'm having a hard time getting herdsman debugging enabled. Are any of you on the Discord and maybe able to help me in real time?
Specifically, I don't know what `the configuration` is in this instruction.
```
Home Assistant OS/Supervised addon #
Set "zigbee_herdsman_debug": true in the configuration.
```

Rene-Sackers, 2020-12-11:
Managed to enable it. So apparently you do just add it to the yaml configuration file in yaml format.
Logs:
```
Zigbee2MQTT:info  2020-12-11 14:26:59: MQTT publish: topic 'zigbee2mqtt/dinner_table_lights_x', payload '{"brightness":13,"color":{"x":0.509709837429556,"y":0.415043632474748},"color_temp":463,"state":"ON"}'
Zigbee2MQTT:info  2020-12-11 14:26:59: MQTT publish: topic 'zigbee2mqtt/dinner_table_right', payload '{"brightness":13,"color":{"x":0.509709837429556,"y":0.415043632474748},"color_temp":463,"linkquality":188,"state":"ON"}'
2020-12-11T14:26:59.853Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101110
2020-12-11T14:26:59.853Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2020-12-11T14:26:59.875Z zigbee-herdsman:deconz:driver delay sending of APS Request
2020-12-11T14:26:59.875Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 52
2020-12-11T14:26:59.876Z zigbee-herdsman:deconz:driver query aps data confirm
2020-12-11T14:26:59.878Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 52 srcAddr: 0xabd9 destAddr: 0x0 profile id: 0x104 cluster id: 0x6
2020-12-11T14:26:59.878Z zigbee-herdsman:deconz:frameParser response payload: 8,11,11,1,0
2020-12-11T14:26:59.878Z zigbee-herdsman:deconz:adapter resolve data request with transSeq Nr.: 11
2020-12-11T14:26:59.878Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":11,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":1,"statusCode":0}},"address":43993,"endpoint":1,"linkquality":188,"groupID":null}'
2020-12-11T14:26:59.878Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-12-11T14:26:59.878Z zigbee-herdsman:deconz:adapter response received
Zigbee2MQTT:info  2020-12-11 14:26:59: MQTT publish: topic 'zigbee2mqtt/dinner_table_center', payload '{"brightness":13,"color":{"x":0.509709837429556,"y":0.415043632474748},"color_temp":463,"linkquality":188,"state":"ON"}'
2020-12-11T14:26:59.975Z zigbee-herdsman:deconz:driver delay sending of APS Request
2020-12-11T14:26:59.975Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 53
2020-12-11T14:26:59.977Z zigbee-herdsman:deconz:driver query aps data confirm
2020-12-11T14:26:59.977Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xabd9 request id: 11 confirm status: 0
2020-12-11T14:26:59.977Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-12-11T14:26:59.978Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 11
2020-12-11T14:27:00.075Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xbe63 EP:1 SeqNr. 49 request id: 12
2020-12-11T14:27:00.076Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 54
2020-12-11T14:27:00.078Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 12 status: 0
2020-12-11T14:27:00.078Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-12-11T14:27:00.078Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - seqNr.: 54 status: 5
2020-12-11T14:27:00.086Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2020-12-11T14:27:00.086Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
2020-12-11T14:27:00.155Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101110
2020-12-11T14:27:00.155Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 1 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2020-12-11T14:27:00.177Z zigbee-herdsman:deconz:driver query aps data confirm
2020-12-11T14:27:00.277Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 55
2020-12-11T14:27:00.277Z zigbee-herdsman:deconz:driver query aps data indication
2020-12-11T14:27:00.279Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xbe63 request id: 12 confirm status: 0
2020-12-11T14:27:00.279Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 1 configChanged: 0 apsRequestFreeSlots: 1
2020-12-11T14:27:00.279Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send with transSeq Nr.: 12
2020-12-11T14:27:00.279Z zigbee-herdsman:deconz:driver query aps data indication
2020-12-11T14:27:00.377Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 56
2020-12-11T14:27:00.379Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 56 srcAddr: 0xbe63 destAddr: 0x0 profile id: 0x104 cluster id: 0x6
2020-12-11T14:27:00.379Z zigbee-herdsman:deconz:frameParser response payload: 8,12,11,1,0
2020-12-11T14:27:00.379Z zigbee-herdsman:deconz:adapter resolve data request with transSeq Nr.: 12
2020-12-11T14:27:00.379Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":12,"manufacturerCode":null,"commandIdentifier [... cut ...]
2020-12-11T14:27:00.480Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr.: 57 status: 5
2020-12-11T14:27:05.193Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 58
2020-12-11T14:27:05.196Z zigbee-herdsman:deconz:frameParser device state: 10100010
2020-12-11T14:27:05.198Z zigbee-herdsman:deconz:driver networkstate: 2 apsDataConfirm: 0 apsDataIndication: 0 configChanged: 0 apsRequestFreeSlots: 1
```
I have dinner_table_(left/right/center) hooked up to a single group. They turn on one after another with about a 0.5s delay between each.

Koenkk, 2020-12-11:
@Rene-Sackers now provide the same lgo when starting zigbee2mqtt

Koenkk, 2020-12-11:
@Rene-Sackers where do you specify this? It should be specified in the configuration shown on the addon page.

[60 more replies]

closed 2021-04-02 by github-actions[bot]