Koenkk/zigbee2mqtt #4886: ZG-DS112L - Lonsonho L+N switches are incorectly identyfied as TS0011, TS0012, TS0013

klami85, 2020-11-06:
## What happened
I have 16 Lonsonho Tuya Zigbee Smart switches with Live+Neutral wires. All variants: 1 Gang, 2 Gangs, 3 Gangs.
Interrior/exterrior photos of 1 Gang version here: https://abload.de/img/switchxmjr2.jpg
2 and 3 gang versions looks the same - just more components soldered.
They are identyfied as TS0011, TS0012, TS0013 (no neutral wire, end device).
Switches work fine except: 
- missing router functionality 
- some errors already mentioned on issue #4872

form database.db:
```
{"id":8,"type":"EndDevice","ieeeAddr":"0xec1bbdfffedf63b5","nwkAddr":9423,"manufId":4098,"manufName":"_TYZB01_qeqvmvti","powerSource":"Battery","modelId":"TS0011","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,4,5,6],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65506":19,"modelId":"TS0011","manufacturerName":"_TYZB01_qeqvmvti","powerSource":3,"appVersion":65,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1604673189878}
{"id":12,"type":"EndDevice","ieeeAddr":"0xec1bbdfffe8a19eb","nwkAddr":1804,"manufId":4098,"manufName":"_TYZB01_vzrytttn","powerSource":"Battery","modelId":"TS0012","epList":[1,2],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,4,5,6],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65506":19,"modelId":"TS0012","manufacturerName":"_TYZB01_vzrytttn","powerSource":3,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":"","appVersion":65}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0014d4887b","endpointID":1}],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0014d4887b","endpointID":1}],"meta":{}}},"appVersion":255,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{"configured":1},"lastSeen":1604673109704}
{"id":18,"type":"EndDevice","ieeeAddr":"0xec1bbdfffeb7cef8","nwkAddr":13324,"manufId":4098,"manufName":"_TYZB01_mqel1whf","powerSource":"Battery","modelId":"TS0013","epList":[1,2,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":256,"inClusterList":[0,4,5,6],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65506":19,"modelId":"TS0013","manufacturerName":"_TYZB01_mqel1whf","powerSource":3,"zclVersion":3,"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b0014d4887b","endpointID":1}],"meta":{}},"2":{"profId":260,"epId":2,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":256,"inClusterList":[4,5,6],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"onOff":0,"onTime":0,"offWaitTime":0}}},"binds":[],"meta":{}}},"appVersion":65,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1604682384538}
```

## What did you expect to happen
They should be identified with powerSource: Mains (single phase), and they should be routers.

## How to reproduce it (minimal and precise)
Just pair them with coordinator and they are visible as TS0011, TS0012, TS0013.

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC2531 
Adapter firmware version: 
{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}

Koenkk, 2020-11-07:
Zigbee2MQTT cannot add (or take away) the router functionality, what you see in the database is just how the device announces itself. Since these are whitelabel devices these are identified as the generic TS0011, TS0012, TS0013 devices so this is expected.

klami85, 2020-11-07:
I am trying to add (fix) support for this device but I am new to this.  
3 Gang switch with L+N wires connected to it identyfied itself as TS0013, 2 Gang (L+N) as TS0012, 1 Gang (L+N) as TS0011,
[TS0013](https://www.zigbee2mqtt.io/devices/TS0013.html) is an end device with no N wire and also looks totally different so it is clearly a different device than mine.
I assume that different devices share the same identifier like in issue #2778

So if router device is marked in database as an end device router functionality no longer work?
On map there is no connections to this device.
Or this device is not a router despite have L+N wires?

For now I do that for device.js

```{
        fingerprint: [{modelID: 'TS0013', manufacturerName: '_TYZB01_mqel1whf'}],
        model: 'Lonsonho 3 Gang zigbee smart switch',
        vendor: 'Lonsonho',
        description: '3 gang switch module with neutral wire',
        extend: generic.switch,
        exposes: [e.switch().withEndpoint('left'), e.switch().withEndpoint('center'), e.switch().withEndpoint('right')],
        endpoint: (device) => {
            return {'left': 1, 'center': 2, 'right': 3};
        },
        meta: {configureKey: 1, multiEndpoint: true},
        configure: async (device, coordinatorEndpoint) => {
            await bind(device.getEndpoint(1), coordinatorEndpoint, ['genOnOff']);
            await bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
            await bind(device.getEndpoint(2), coordinatorEndpoint, ['genOnOff']);
        },
    },

Koenkk, 2020-11-08:
These changes are only cosmetic but will not change the device from being an end device to router. The firmware of this device says its an end device, not a router.

Note that pictures can be off for these white label devices (since they are whitelabel)

klami85, 2020-11-10:
After checking out tuya developer platform all becomes clear to me.
Someone (lonsonho) ordered TYZS3 module from TUYA with firmware for end device switch and put it in to L+N Switch (maybe they got too many...)
Seems like lonsonho makes another devices the same way. That's why we get duplicate IDs for some devices.
Before checking, I thought that if module is universal for tuya family devices the firmware is also universal and I can flip some values on the fly.
SDK for used SOC is available so maybe someone, someday make it better...
https://www.silabs.com/wireless/zigbee/efr32mg13-series-1-socs/device.efr32mg13p732f512gm48

closed 2020-12-19 by github-actions[bot]

adizag, 2023-02-15:
@klami85  Did you have success with fixing it? did you flash a new FW?