Koenkk/zigbee2mqtt #4887: Orvibo smart sticker stopped working after migrating to Conbee2

Keeze, 2020-11-07:
Hello,

Today I migrated to a Conbee2 Adapter and I added a network_key. All devices are repaired and work. Only with one device there is a problem. It is the Orvibo Smartsticker CR11S8UZ. After repair the device appeared in Z2M.
Logs say:

`{"dateCode":"2016-05-20      ","description":"Smart sticker switch","friendly_name":"Orvibo01","hardwareVersion":0,"ieeeAddr":"0x00124b0013dc11e0","lastSeen":1604707206669,"manufacturerID":0,"manufacturerName":"欧瑞博","model":"CR11S8UZ","modelID":"3c4e4fc81ed442efaf69353effcdfc5f","networkAddress":47631,"powerSource":"Battery","type":"EndDevice","vendor":"ORVIBO"}],"type":"devices"}
`

But no response when I click a button (not in the log, mqtt or home-assistant). I removed the device (had to use force-remove) and paired it again. It reappeared as expected. Still no luck.

Does anyone have a clue?

Z2M 1.6.1 running in Docker.

Thank,
Kees

alexkutsan, 2020-12-14:
Have the same issue this this device

closed 2021-01-22 by github-actions[bot]

Keeze, 2021-07-16:
Issue still exists, has anyone any clue on this?