Koenkk/zigbee2mqtt #4889: Last update => Port '8485' is already in use by something else on the host

ABRT29, 2020-11-07:
Hello,

I have just corrected the update of zigbee2mqtt but cannot restart it displays the following error :
**Port '8485' is already in use by something else on the host.**

I didn't add anything more (add-on, plugin or other) to my system.

Thank you!

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC2531
Home Assistant version: 0.117.5

sjorge, 2020-11-07:
You can try changing the port the frontend runs on to avoid the error.

klumpler, 2020-11-12:
Same situation here. No other Home Assistant config change, just add-on update to 1.16.1 and suddenly unable to start it. Port '8485' is already in use by something else on the host.

klumpler, 2020-11-12:
One more info - after update to HassOS 4.16 and full reboot, the problem dissapeared. Everything is running now... Strange.

closed 2020-12-21 by github-actions[bot]

FaBRiK74, 2021-02-12:
Today morning the Z2M has stopped to work. When I wanted to restart, I got the same error that port 8485 is used by something else. But I didn't change anything in the morning.
What is the solution?

Lesani, 2023-01-29:
and today, after updating to 2023.1.7 I get this message as well, I did not change anything apart from letting Homeassistant Core update and reboot

ToMs77130, 2023-02-09:
Same for me, I've checked port with netstat but it doesn't seems to be used, despite the message.
I've changed port from 8485 to 8486 and z2m started again.

kwint, 2023-02-14:
Same issue here, zigbee2mqtt seems to stop working randomly., and when I start it again I get this error

Gandizzle91, 2023-02-17:
Same here.
After i changed the Port i dont get the Message anymore. But it still dont start without any error

wojciechczyz, 2023-05-03:
Same problem with latest version upgrade

squromiv, 2023-05-14:
After updating to Home Assistant Core 2023.5 got the same issue.

satwiz-md, 2023-05-18:
me too I'm not a beta tester!!!!

DanielNagy, 2023-05-26:
Also had this issue with trying to start z2m and nut server add ins. Seems a ha issue if you have exposed port configured in respective add in. 
To make zigbee2mqtt start, remove the 8485 configured exposed port in the addin config, and it will start.

albertombl, 2024-12-11:
Same situation today. Z2m not working because the port 8485 is already been used. Is there any final solution to the problem or just remove 8485 from addon configuration ?

BardaktakiAdam, 2025-02-07:
Same situation. I have updated the zigbee2mqtt, and it starts to Port '8485' is already in use by something else on the host

FaBRiK74, 2025-03-16:
> same her, what ist the solution?

As above, just change the port to e.g. 8486