Koenkk/zigbee2mqtt #4890: Template File for unsupported devices

chrisrock1984, 2020-11-07:
I have two devices which are not supported at the moment and I am running zigbee2mqtt under Homeassistant.

I know that there are tutorials for adding new unsupported devices.
As I do not want to mess things up within the docker image of zigbee2mqtt, would it be possible to add a file for an unsupported device, which is editable in Homeassistant or accessible via SMB?
So for each new unsupported device there is a file for it.
Those files can then be uploaded and later integrated into the next version.
Also a database can be created for new devices to sharing and testing.

What do you think of this idea?

Br
Chris

Koenkk, 2020-11-07:
You can use the external converters: https://www.zigbee2mqtt.io/information/configuration.html#external-converters-configuration

closed 2020-11-07 by Koenkk

chrisrock1984, 2020-11-07:
Thx and sorry for the unnecessary issue....