Koenkk/zigbee2mqtt #4891: Creating UDEV rule not working

Anjerlaan, 2020-11-07:
Hi,

I have created a UDEV rule for the ZZH adapter while following this page:
https://www.zigbee2mqtt.io/information/FAQ.html#verify-that-the-user-you-run-zigbee2mqtt-as-has-write-access-to-the-port
But the rule is not applied when `ATTRS{serial}=="3f980000.usb"` is added to the rule.
Leaving the serial out does create a rule.

## What happened
I have performed the following steps:

```
pi@Domotica-Pi:~ $ test -w /dev/ttyUSB1 && echo success || echo failure
success
pi@Domotica-Pi:~ $ udevadm info -a -n /dev/ttyUSB1 | grep 'serial'
    SUBSYSTEMS=="usb-serial"
    ATTRS{serial}=="3f980000.usb"
```

Created a rules file: `sudo nano /etc/udev/rules.d/99-usb-serial.rules` and added the following rule:

`SUBSYSTEM=="tty", ATTRS{idProduct}=="7523", ATTRS{idVendor}=="1a86", ATTRS{serial}=="3f980000.usb", SYMLINK="ttyUSB_ZZH", OWNER="pi"`

But when I checked with: `ls -l /dev/ttyUSB*` nothing happened:
```
pi@Domotica-Pi:~ $ ls -l /dev/ttyUSB*
crw-rw---- 1 root dialout 188, 0 nov  7 13:37 /dev/ttyUSB0
crw-rw---- 1 pi   dialout 188, 1 nov  7 13:45 /dev/ttyUSB1
```

So I start testing, and when I removed the `ATTRS{serial}=="3f980000.usb"`, the rule was applied

```
ls -l /dev/ttyUSB*
crw-rw---- 1 root dialout 188, 0 nov  7 13:37 /dev/ttyUSB0
crw-rw---- 1 pi   dialout 188, 1 nov  7 13:45 /dev/ttyUSB1
lrwxrwxrwx 1 root root         7 nov  7 13:28 /dev/ttyUSB_ZZH -> ttyUSB1
```

Why is this not working?

## Debug info
Zigbee2MQTT version: v.1.15.0 
Adapter hardware: CC26X2R1
Adapter firmware version: (zStack3x0 20200925)

Anjerlaan, 2020-11-08:
I have figured it out, I have been using the product and vendor ID's of the wrong node
I was looking at `../devices/platform/soc/3f980000.usb/usb1/1-1/1-1.2`.
But it It should be the vendor and product id of the `../devices/platform/soc/3f980000.usb/usb1` node where also the ATTRS{serial}=="3f980000.usb" is shown.
It is working now

Anjerlaan, 2020-11-08:
This issue can be closed now

Anjerlaan, 2020-11-09:
I thought I had it working but suddenly `ls -l /dev/serial/by-id` was not found anymore.

This is because ATTRS{serial}=="3f980000.usb" in the rule is causing the error.
When I perform a serial grep on both USB ports, the same serial is returned.

```
pi@Domotica-Pi:~ $ udevadm info -a -n /dev/ttyUSB0 | grep 'serial'
    SUBSYSTEMS=="usb-serial"
    ATTRS{serial}=="3f980000.usb"
pi@Domotica-Pi:~ $ udevadm info -a -n /dev/ttyUSB1 | grep 'serial'
    SUBSYSTEMS=="usb-serial"
    ATTRS{serial}=="DO2VYPCR"
    ATTRS{serial}=="3f980000.usb"
```
So these 2 rules are working
```
SUBSYSTEM=="tty", ATTRS{idProduct}=="7523", ATTRS{idVendor}=="1a86", SYMLINK+="ttyUSB_ZZH", OWNER="pi"
SUBSYSTEM=="tty", ATTRS{idProduct}=="6015", ATTRS{idVendor}=="0403", SYMLINK+="ttyUSB_RFX", OWNER="pi"
```

milkpirate, 2020-12-07:
Ok I dont have that exact same problem, but this issue could become udev problem collection.

So anyway, I have the following udev rule:
```
SUBSYSTEM=="tty", \
  ATTRS{idVendor}=="0451", \
  ATTRS{idProduct}=="16a8", \
  SYMLINK+="ttyCC2531", \
  GROUP="dialout", \
  MODE="0666"
```
which works nearly as expected as far as I can tell:
```
$ sudo udevadm control --reload-rules
$ sudo udevadm trigger
$ udevadm test $(udevadm info --query=path --name=/dev/ttyACM0)    # <-- the CC2531 device
...
DEVLINKS=/dev/ttyCC2531 /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00xxxxxxxxxxxxxxx-if00 
...
```
So obviously it creates a symlink with the expected name but the target is wrong:
```
$ ls /dev
...
666 crw-rw-rw- 1 root dialout 166,   0 Dec  7 21:53 ttyACM0
777 lrwxrwxrwx 1 root root           5 Dec  7 21:48 ttyCC2531 -> vcsu5
...
```
And even stranger, every time I do a `sudo udevadm trigger` the symlink changes to another (wrong) device. Just a selection:
```
ttyCC2531 -> mmcblk0p2
ttyCC2531 -> i2c-1
ttyCC2531 -> watchdog
...
```
I dont see any pattern, do you? And whats wrong anyway?! This is really anoying. Hope you can help me out with that.

milkpirate, 2021-01-12:
Ok, I think I found the problem:

replacing
```
- SUBSYSTEM=="tty", \
+ KERNEL=="ttyACM?", \
```
made it work. The subsystem directive seems to break everything :/

Anyway, hope this helps someone!

closed 2021-01-20 by github-actions[bot]

kinghill44, 2022-01-13:
This Code is working for me on Raspberry with Debian

`SUBSYSTEMS=="usb", KERNEL=="ttyUSB*", ATTRS{idProduct}=="7523", ATTRS{idVendor}=="1a86", SYMLINK="Zigbee01",`