Koenkk/zigbee2mqtt #4894: Can't add Ikea on/off switch to group, model E1603/E1702

brunoamaral, 2020-11-07:
## What happened

I see an error when trying to add a tradfri remote to a group or bind it to an outlet.

## What did you expect to happen

Device should have been added to the group or binded with the outlet

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: docker-compose 1.16.1
Adapter hardware: CC2652R1 
Adapter firmware version: CC26X2R1_20201026.hex

```
Zigbee2MQTT:debug 2020-11-07 14:29:42: Received MQTT message on 'zigbee2mqtt/bridge/request/group/members/add' with data '{"device":"0x14b457fffec6954c/1","group":"Living Room"}'
Zigbee2MQTT:info  2020-11-07 14:30:05: MQTT publish: topic 'zigbee2mqtt/bridge/response/group/members/add', payload '{"data":{"device":"0x14b457fffec6954c/1","group":"Living Room"},"error":"Failed to add from group (Command 0x14b457fffec6954c/1 genGroups.add({\"groupid\":1,\"groupname\":\"\"}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 38252 - 1 - 4 - 4 - 0 after 10000ms))","status":"error"}'
```

Koenkk, 2020-11-07:
Make sure to wakeup the remote right before exeucting the command (otherwise it will fail because it is sleeping). You can do this by clicking a button on it.

brunoamaral, 2020-11-07:
Thank you for that, I tried it but it still didn't work.

Looking through the logs I did find a timeout error that I don't quite understand, but where the group name is empty:

```json
error 2020-11-07 19:56:18: Failed to add from group (Command 0x14b457fffe384bde/242 genGroups.add({"groupid":2,"groupname":""}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 8727 - 242 - 131 - 4 - 0 after 10000ms))
```

Koenkk, 2020-11-08:
You should not add E1743 to the group, but only bind it to the group.

brunoamaral, 2020-11-08:
Thank you, that is working. I still can't seem to switch off the group with the remote but I guess it's due to my lack of knowledge right now.

closed 2020-11-08 by brunoamaral