Koenkk/zigbee2mqtt #4897: Issue to recognize device: HGZB-01A

kevincaradant, 2020-11-07:
Hi

I don"t why, but when I paired my HGZB-01A, I got this line:

`zigbee2mqtt    | Zigbee2MQTT:warn  2020-11-07 19:49:26: Device '0x086bd7fffe3f3938' with Zigbee model 'LXN59-1S7LX1.0' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html`

Not sure to understand why it says: model 'LXN59-1S7LX1.0'.

What I'm supposed to do ? Add manually the device ? 

Is there an issue on Z2M to recognize the good device ? 

Thank you in advance

Regards

Koenkk, 2020-11-08:
This device has been addd in Zigbee2MQTT 1.16.0 so just update to the latest version.

closed 2020-11-08 by Koenkk

kevincaradant, 2020-11-08:
Ah yes you're right it works well in 1.16.1.

Thank you

kevincaradant, 2020-12-14:
Hi

Just to know is it possible to fix the AC power issue and the icon (image) of the Switch which is wrong ?

I'm running with this one:
[image]

And I have this under Z2M:
[image]

kevincaradant, 2020-12-14:
Alright Thanks :)

Koenkk, 2020-12-15:
- Can you provide your zigbee2mqtt debug log?
- Can you provide the data/database.db entry of this device.

kevincaradant, 2020-12-15:
Hi 

My apologies, not sure if it was because I just did a restart instead of down / up with my docker or because I pulled it a little bit too soon yesterday from the docker hub but today it's working fine ;)

kevincaradant, 2020-12-22:
Hi

Finally I didn't notice but the picture is the generic picture of zigbee "Z" red but not really the device :)

Is it normal? 

[image]

Koenkk, 2020-12-23:
Yes, https://github.com/Koenkk/zigbee2mqtt/issues/4897#issuecomment-744638206

> Picture will be fixed on new z2m release.