Koenkk/zigbee2mqtt #4898: with Zigbee model 'GASSensor-EFR-3.0' is NOT supported

vpereiradn, 2020-11-07:
Estou com um problema nesse sensor de gas
Zigbee2MQTT:warn  2020-11-07 18:54:32: Device '0xbc33acfffe03f318' with Zigbee model 'GASSensor-EFR-3.0' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Zigbee2MQTT:info  2020-11-07 18:54:32: MQTT publish: topic 'zigbee2mqtt/bridge/devices', payload '[{"definition":null,"endpoints":{"1":{"bindings":[],"clusters":{"input":[],"output":[]}},"11":{"bindings":[],"clusters":{"input":["ssIasAce"],"output":["ssIasZone","ssIasWd"]}},"110":{"bindings":[],"clusters":{"input":[],"output":[]}},"12":{"bindings":[],"clusters":{"input":[],"output":[]}},"13":{"bindings":[],"clusters":{"input":["genOta"],"output":[]}},"2":{"bindings":[],"clusters":{"input":[],"output":[]}},"242":{"bindings":[],"clusters":{"input":[],"output":[]}},"3":{"bindings":[],"clusters":{"input":[],"output":[]}},"4":{"bindings":[],"clusters":{"input":[],"output":[]}},"47":{"bindings":[],"clusters":{"input":[],"output":[]}},"5":{"bindings":[],"clusters":{"input":[],"output":[]}},"6":{"bindings":[],"clusters":{"input":[],"output":[]}},"8":{"bindings":[],"clusters":{"input":[],"output":[]}}},"friendly_name":"Coordinator","ieee_address":"0x00124b0014da4394","interview_completed":true,"interviewing":false,"network_address":0,"supported":false,"type":"Coordinator"},{"date_code":"2019.4.10","definition":null,"endpoints":{"1":{"bindings":[],"clusters":{"input":["genBasic","genIdentify","ssIasZone","haDiagnostic"],"output":["genOta"]}},"242":{"bindings":[],"clusters":{"input":[],"output":["greenPower"]}}},"friendly_name":"0xbc33acfffe03f318","ieee_address":"0xbc33acfffe03f318","interview_completed":true,"interviewing":false,"network_address":36963,"power_source":"Mains (single phase)","supported":false,"type":"Router"}]'
Zigbee2MQTT:info  2020-11-07 18:54:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0xbc33acfffe03f318","supported":false},"type":"pairing"}'
Zigbee2MQTT:warn  2020-11-07 18:54:58: Received message from unsupported device with Zigbee model 'GASSensor-EFR-3.0'
Zigbee2MQTT:warn  2020-11-07 18:54:58: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:warn  2020-11-07 18:55:01: Received message from unsupported device with Zigbee model 'GASSensor-EFR-3.0'
Zigbee2MQTT:warn  2020-11-07 18:55:01: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:warn  2020-11-07 18:55:04: Received message from unsupported device with Zigbee model 'GASSensor-EFR-3.0'
Zigbee2MQTT:warn  2020-11-07 18:55:04: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

vpereiradn, 2020-11-12:
uso o raspberry pi 4
em cima do raspbian..

Koenkk, 2020-11-12:
English only.

closed 2020-11-12 by Koenkk

vpereiradn, 2020-11-13:
I have a problem with this HEIMAN sensor
is giving as not supported by zigbeemqtt2, any solution or update of it?

Zigbee 'GASSensor-EFR-3.0

vpereiradn, 2020-11-13:
I'm using raspberry pi 4 raspbian,
raspberry home assintant with hassio!!

Koenkk, 2020-11-13:
What model number is written on the device itself?

vpereiradn, 2020-11-13:
HEIMAN  model 'GASSensor-EFR-3.0
on the raspberry com CC2531 module

vpereiradn, 2020-11-13:
I tried this Zigbee2mqtt Edge too, without success

Koenkk, 2020-11-14:
I expect something like HS1CG-M or HS3CG to be written on the device (physically)

vpereiradn, 2020-11-20:
?

vpereiradn, 2020-11-20:
M414-5Ever1.3

Koenkk, 2020-11-21:
Can you provide a link to the product (where you bought it, aliexpress?)

vpereiradn, 2020-11-21:
yes I can yes, just that in the details of the seller is one thing and in the manual and in the status of the sensor this is what I sent you up there !!

https://pt.aliexpress.com/item/32840620167.html?spm=a2g0o.cart.0.0.72c43c00XygpsL&mp=1

Koenkk, 2020-11-21:
Good, details say it's a `HS1CG-E`, added it

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

vpereiradn, 2020-11-21:
Hass.io addon?

vpereiradn, 2020-11-22:
Hello good night !, still with the same error.
even installing the brand!
Edge ADDON em home assistant!
I'm almost giving up and using the native that comes with the home assistant, which is p Zigbee Home automation

Zigbee2MQTT: warn 2020-11-21 23:34:49: Received message from unsupported device with Zigbee model 'GASSensor-EFR-3.0' and manufacturer name 'HEIMAN'
Zigbee2MQTT: warn 2020-11-21 23:34:49: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.

Koenkk, 2020-11-22:
I forgot to update, sorry, changes will be in latest dev in a few hours from now.

vpereiradn, 2020-11-22:
Hello good Morning!!
would you have to uninstall and install the home assistant again?

Koenkk, 2020-11-22:
Yes (and make sure you are on the edge version)

vpereiradn, 2020-11-22:
Wow bro thank you very much working correctly HEIMAN GA sensor, you are the same guy!!!

Nossa mano muito obrigado funcionando certinho Sensor de GAs HEIMAN, voce e o cara mesmo!

vpereiradn, 2020-11-23:
I think that the system does not detect more in the home assistant than in a gas leak.  and the automations didn't work.

Battery low
false
Gas
false
Linkquality
28
Tamper

gas
Indicates whether the device detected gas
false
battery_low
Indicates if the battery of this device is almost empty
false
tamper
Indicates whether the device is tampered
false

vpereiradn, 2020-11-23:
hello good morning, device identified correctly in the ZIgbee, however it does not have the status of when the gas test button is activated, and the automations are not working, I don't know what it can be !!

Zigbee2MQTT:debug 2020-11-23 09:20:31: Received MQTT message on 'homeassistant/binary_sensor/0xbc33acfffe03f318/battery_low/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xbc33acfffe03f318"],"manufacturer":"HEIMAN","model":"Combustible gas sensor (HS1CG-E)","name":"Sensor de Gas de cozinha HEIMAN","sw_version":"Zigbee2MQTT 1.16.1-dev"},"device_class":"battery","json_attributes_topic":"zigbee2mqtt/Sensor de Gas de cozinha HEIMAN","name":"Sensor de Gas de cozinha HEIMAN battery_low","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/Sensor de Gas de cozinha HEIMAN","unique_id":"0xbc33acfffe03f318_battery_low_zigbee2mqtt","value_template":"{{ value_json.battery_low }}"}'
Zigbee2MQTT:debug 2020-11-23 09:20:31: Received MQTT message on 'homeassistant/binary_sensor/0xbc33acfffe03f318/tamper/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xbc33acfffe03f318"],"manufacturer":"HEIMAN","model":"Combustible gas sensor (HS1CG-E)","name":"Sensor de Gas de cozinha HEIMAN","sw_version":"Zigbee2MQTT 1.16.1-dev"},"json_attributes_topic":"zigbee2mqtt/Sensor de Gas de cozinha HEIMAN","name":"Sensor de Gas de cozinha HEIMAN tamper","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/Sensor de Gas de cozinha HEIMAN","unique_id":"0xbc33acfffe03f318_tamper_zigbee2mqtt","value_template":"{{ value_json.tamper }}"}'
Zigbee2MQTT:debug 2020-11-23 09:20:31: Received MQTT message on 'homeassistant/sensor/0xbc33acfffe03f318/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"}],"device":{"identifiers":["zigbee2mqtt_0xbc33acfffe03f318"],"manufacturer":"HEIMAN","model":"Combustible gas sensor (HS1CG-E)","name":"Sensor de Gas de cozinha HEIMAN","sw_version":"Zigbee2MQTT 1.16.1-dev"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/Sensor de Gas de cozinha HEIMAN","name":"Sensor de Gas de cozinha HEIMAN linkquality","state_topic":"zigbee2mqtt/Sensor de Gas de cozinha HEIMAN","unique_id":"0xbc33acfffe03f318_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'

vpereiradn, 2020-11-23:
Below when I pressed the sensor test button

Zigbee2MQTT:debug 2020-11-23 09:54:55: Received Zigbee message from 'Sensor de Gas de cozinha HEIMAN', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":34}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-11-23 09:54:55: MQTT publish: topic 'zigbee2mqtt/Sensor de Gas de cozinha HEIMAN', payload '{"battery_low":false,"gas":false,"linkquality":13,"tamper":false}'
Zigbee2MQTT:debug 2020-11-23 09:54:56: Received Zigbee message from 'Sensor de Gas de cozinha HEIMAN', type 'commandStatusChangeNotification', cluster 'ssIasZone', data '{"extendedstatus":0,"zonestatus":32}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-11-23 09:54:56: MQTT publish: topic 'zigbee2mqtt/Sensor de Gas de cozinha HEIMAN', payload '{"battery_low":false,"gas":false,"linkquality":21,"tamper":false}'

Koenkk, 2020-11-23:
Fixed

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

vpereiradn, 2020-11-23:
funcionouuu maravilhoso muito obrigado 
voce e o cara mesmo em

vpereiradn, 2020-11-24:
My thanks for your attention and commitment, you really are top, thanks bro.

vytautasgimbutas, 2020-12-05:
@Koenkk 
I have heiman smart motion sensor. On the back of the device it says `smart motion sensor`, in the manual `M414-12Ever1.2`

Aliexpress link: https://pt.aliexpress.com/item/4000174635724.html?spm=a2g0s.9042311.0.0.76604c4dE9WQq0

Should I create a separate issue or could you add identifier `PIRSensor-EF-3.0` to the heiman motion sensor?

```
Zigbee2MQTT:info  2020-12-05 17:02:36: Device '0xccccccfffed8613a' joined
Zigbee2MQTT:info  2020-12-05 17:02:36: Starting interview of '0xccccccfffed8613a'
Zigbee2MQTT:info  2020-12-05 17:02:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xccccccfffed8613a"},"type":"device_connected"}'
Zigbee2MQTT:info  2020-12-05 17:02:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xccccccfffed8613a"},"type":"pairing"}'
Zigbee2MQTT:info  2020-12-05 17:02:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xccccccfffed8613a"},"type":"device_announced"}'
Zigbee2MQTT:info  2020-12-05 17:02:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xccccccfffed8613a"},"type":"device_announced"}'
Zigbee2MQTT:info  2020-12-05 17:03:39: Successfully interviewed '0xccccccfffed8613a', device has successfully been paired
Zigbee2MQTT:warn  2020-12-05 17:03:39: Device '0xccccccfffed8613a' with Zigbee model 'PIRSensor-EF-3.0' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
Zigbee2MQTT:info  2020-12-05 17:03:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"friendly_name":"0xccccccfffed8613a","supported":false},"type":"pairing"}'
Zigbee2MQTT:warn  2020-12-05 17:03:39: Received message from unsupported device with Zigbee model 'PIRSensor-EF-3.0'
Zigbee2MQTT:warn  2020-12-05 17:03:39: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:warn  2020-12-05 17:03:39: Received message from unsupported device with Zigbee model 'PIRSensor-EF-3.0'
Zigbee2MQTT:warn  2020-12-05 17:03:39: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:warn  2020-12-05 17:03:41: Received message from unsupported device with Zigbee model 'PIRSensor-EF-3.0'
Zigbee2MQTT:warn  2020-12-05 17:03:41: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:warn  2020-12-05 17:03:41: Received message from unsupported device with Zigbee model 'PIRSensor-EF-3.0'
Zigbee2MQTT:warn  2020-12-05 17:03:41: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:warn  2020-12-05 17:03:42: Received message from unsupported device with Zigbee model 'PIRSensor-EF-3.0'
Zigbee2MQTT:warn  2020-12-05 17:03:42: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.
Zigbee2MQTT:warn  2020-12-05 17:03:42: Received message from unsupported device with Zigbee model 'PIRSensor-EF-3.0'
```

Koenkk, 2020-12-06:
@vytautasgimbutas added, thanks

vpereiradn, 2020-12-06:
hello good afternoon it's ok i bought this and saw that there are no devices for cc2531

https://a.aliexpress.com/_mramgGt

vpereiradn, 2020-12-22:
hello good afternoon man!
so about the one I gave you, it was not supported, I removed and installed it first of all, even so it is still not supported !!
would be the one below !!!
https://en.aliexpress.com/item/1005001499921065.html?spm=a2g0s.9042311.0.0.32b7b90aAXM7VJ

maverik80, 2021-02-19:
good morning I also have problems with this gas sensor, m414-5E v1.3, it couples but does not expose any entity, I tried to install the edje version but still the same, the product link is here 
€ 23,05  9%OFF | Sensore di perdite di gas Zigbee 3.0 rilevatore di gas naturale sensore di Gas Zigbee gpl funzionante con SmartThing,Ziptao,Conbee Zigbee Gateway
https://a.aliexpress.com/_uws5gJ

Avamander, 2023-05-06:
@Koenkk Just wondering, would it be out of the question to add a notice next to this device that it might emit a very loud high-pitched coil whine at all times? I think a lot of people use the list of supported devices as a list of devices they can buy and it might save people the trouble.

Koenkk, 2023-05-07:
@Avamander you can do this by going to the device page and clicking on "Help to make the docu better and edit this page on Github" (scroll down)