Koenkk/zigbee2mqtt #4899: MQTT Command to restart z2m

svh1985, 2020-11-07:
## Is your feature request related to a problem? Please describe
Quite regularly I have one or two devices which are unresponsive. Even though I have a powerful zigbee stick (zzh). So, what I do to solve; is restarting the z2m bridge manually. 

## Describe the solution you'd like
Because I’m using HA, I could of course automate this. All that is missing, is a command to restart the bridge via mqtt. This way I could setup an automation.

An additional benefit is that this also makes restarting z2m somewhat easier.

Koenkk, 2020-11-08:
Can you provide the herdsman debug log of the device being unresponsive?  (because that should not happen in the first place)

svh1985, 2020-11-10:
You're right, once it happens again I will upload the log here.

But I do think it's nice to have a way to restart the bridge via an MQTT-command.

closed 2020-12-19 by github-actions[bot]