Koenkk/zigbee2mqtt #4901: Unbiding from non-existing group fails

docbobo, 2020-11-08:
## What happened

I bought an Ikea IKEA E1524/E1810 yesterday, which is pre-bound to a default group. In my case it was 901, I suspect this is the one Ikea uses for all its devices. 

In order to control my devices, I needed to bind the to the remote. However, according to the Z2M documentation, I had to first unbind from the default group. But that did not work because group 901 did not exist on my system before.

The UI was showing group 901 as empty, but neither could I delete it over there nor could I unbind the device from the group. Instead, I had to first manually add it to the configuration.yaml.

## What did you expect to happen

Unbinding the group & deleting it should work independent of whether it has an entry in configuration.yaml or not.

## How to reproduce it (minimal and precise)

See above. I suspect creating a group, binding a device, then manually deleting the group from configuration.yaml will set up the system in a similar way. Didn't try to.

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC1352P-2
Adapter firmware version: 20201026

Koenkk, 2020-11-08:
Can you provide the debug log when unbinding from this group?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

pierrepinon, 2020-11-08:
I have the same issue

Group 901 is created on UI without devices in :

[image]

When i try to remove 901 group : 
`debug 2020-11-08 13:46:00: Received MQTT message on 'zigbee2mqtt/bridge/request/group/remove' with data '901'

error 2020-11-08 13:46:00: Request 'zigbee2mqtt/bridge/request/group/remove' failed with error: 'message.trim is not a function'

info  2020-11-08 13:46:00: MQTT publish: topic 'zigbee2mqtt/bridge/response/group/remove', payload '{"data":{},"error":"message.trim is not a function","status":"error"}'`

pedrolamas, 2020-11-10:
@Koenkk I'm experiencing this right now with an IKEA E1743 switch!

[image]

[image]

[image]

```
Zigbee2MQTT:debug 2020-11-10 18:41:02: Received MQTT message on 'zigbee2mqtt/bridge/request/group/remove' with data '901'
Zigbee2MQTT:error 2020-11-10 18:41:02: Request 'zigbee2mqtt/bridge/request/group/remove' failed with error: 'message.trim is not a function'
Zigbee2MQTT:info  2020-11-10 18:41:02: MQTT publish: topic 'zigbee2mqtt/bridge/response/group/remove', payload '{"data":{},"error":"message.trim is not a function","status":"error"}'
```

pedrolamas, 2020-11-11:
After better reading [this](https://www.zigbee2mqtt.io/devices/E1743.html#binding) on the docs, I managed to solve my problem!

I still seem to not being able to fully "unbind" from the 901 group, but at least the button is usable inside the group I wanted!

[image]

docbobo, 2020-11-11:
When trying to unbind the device from a group that does not exist in the configuration.yaml - something definitely happening  with the IKEA default of 901, I get

```
Zigbee2MQTT:info  2020-11-11 16:28:00: MQTT publish: topic 'zigbee2mqtt/bridge/response/device/unbind', payload '{"data":{"from":"hue_dimmer_04/1","to":"4"},"error":"Target device or group '4' does not exist","status":"error"}'
Zigbee2MQTT:error 2020-11-11 16:28:00: Target device or group '4' does not exist
```

(I did try to reproduce this by first creating the group, the removing it from the yaml).  

Trying to delete the group, results in 

```
Zigbee2MQTT:error 2020-11-11 16:30:14: Request 'zigbee2mqtt/bridge/request/group/remove' failed with error: 'message.trim is not a function'

```

Koenkk, 2020-11-11:
- I've fixed the unbinding of this group, it will now appear as `default_bind_group` instead of `901`.
- This is a special group and cannot be removed. @nurikk I think we should now show this group in the group overview.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-12-13 by Koenkk