Koenkk/zigbee2mqtt #4902: How to configure Power-on behavior for Philips Hue

u17194907425, 2020-11-08:
## What happened
I was trying to configure the Philips Hue bulb (this specific model https://www.zigbee2mqtt.io/devices/9290023349.html) to stay off in case of a power outage. I was able to make it work but the [instructions on the device page](https://www.zigbee2mqtt.io/devices/9290023349.html#power-on-behavior) were not easy to interpret.

I was only able to configure one of my bulbs when I have searched and read comments in #4298 (specifically [this one](https://github.com/Koenkk/zigbee2mqtt/issues/4298#issuecomment-688483709)).

## What did you expect to happen
My expectation was that I just update the devices.yaml in order to define the desired Power-on behavior. Like this

```
'0x0017880108fe7a41':
  friendly_name: 'philips_hue_testing_2'
  retain: true
  hue_power_on_behavior: 'off'
```

however after I reboot the coordinator there is no any mention of the `hue_power_on_behavior` in the log.

```
Zigbee2MQTT:debug 2020-11-08 10:59:41: Received MQTT message on 'homeassistant/light/0x0017880108fe7a41/light/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/philips_hue_testing_2/availability"}],"brightness":true,"brightness_scale":254,"color_temp":false,"command_topic":"zigbee2mqtt/philips_hue_testing_2/set","device":{"identifiers":["zigbee2mqtt_0x0017880108fe7a41"],"manufacturer":"Philips","model":"Hue white A67 bulb E26 with Bluetooth (1600 Lumen) (9290023349)","name":"philips_hue_testing_2","sw_version":"Zigbee2MQTT 1.16.1"},"effect":true,"effect_list":["blink","breathe","okay","channel_change","finish_effect","stop_effect"],"hs":false,"json_attributes_topic":"zigbee2mqtt/philips_hue_testing_2","name":"philips_hue_testing_2_light","schema":"json","state_topic":"zigbee2mqtt/philips_hue_testing_2","unique_id":"0x0017880108fe7a41_light_zigbee2mqtt","xy":false}'
Zigbee2MQTT:debug 2020-11-08 10:59:41: Received MQTT message on 'homeassistant/sensor/0x0017880108fe7a41/linkquality/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/philips_hue_testing_2/availability"}],"device":{"identifiers":["zigbee2mqtt_0x0017880108fe7a41"],"manufacturer":"Philips","model":"Hue white A67 bulb E26 with Bluetooth (1600 Lumen) (9290023349)","name":"philips_hue_testing_2","sw_version":"Zigbee2MQTT 1.16.1"},"icon":"mdi:signal","json_attributes_topic":"zigbee2mqtt/philips_hue_testing_2","name":"philips_hue_testing_2_linkquality","state_topic":"zigbee2mqtt/philips_hue_testing_2","unique_id":"0x0017880108fe7a41_linkquality_zigbee2mqtt","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}"}'
Zigbee2MQTT:debug 2020-11-08 10:59:41: Received MQTT message on 'homeassistant/binary_sensor/0x0017880108fe7a41/update_available/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/philips_hue_testing_2/availability"}],"device":{"identifiers":["zigbee2mqtt_0x0017880108fe7a41"],"manufacturer":"Philips","model":"Hue white A67 bulb E26 with Bluetooth (1600 Lumen) (9290023349)","name":"philips_hue_testing_2","sw_version":"Zigbee2MQTT 1.16.1"},"json_attributes_topic":"zigbee2mqtt/philips_hue_testing_2","name":"philips_hue_testing_2_update_available","payload_off":false,"payload_on":true,"state_topic":"zigbee2mqtt/philips_hue_testing_2","unique_id":"0x0017880108fe7a41_update_available_zigbee2mqtt","value_template":"{{ value_json.update_available}}"}'
Zigbee2MQTT:debug 2020-11-08 10:59:41: Received MQTT message on 'homeassistant/sensor/0x0017880108fe7a41/update_state/config' with data '{"availability":[{"topic":"zigbee2mqtt/bridge/state"},{"topic":"zigbee2mqtt/philips_hue_testing_2/availability"}],"device":{"identifiers":["zigbee2mqtt_0x0017880108fe7a41"],"manufacturer":"Philips","model":"Hue white A67 bulb E26 with Bluetooth (1600 Lumen) (9290023349)","name":"philips_hue_testing_2","sw_version":"Zigbee2MQTT 1.16.1"},"icon":"mdi:update","json_attributes_topic":"zigbee2mqtt/philips_hue_testing_2","name":"philips_hue_testing_2_update_state","state_topic":"zigbee2mqtt/philips_hue_testing_2","unique_id":"0x0017880108fe7a41_update_state_zigbee2mqtt","value_template":"{{ value_json['update']['state'] }}"}'
```

I understand that my expectation might be incorrect and sending a message manually to `zigbee2mqtt/<DEVICE>/set/hue_power_on_behavior` is the designed way to do it. If that's the case, then please confirm and maybe I'll try to help updating the pages in https://www.zigbee2mqtt.io/devices so that it's more clear how to actually configure the device.

Maybe even make a one page and remove duplication of information on each Hue device with power-on config?

## How to reproduce it (minimal and precise)
Add a Philips Hue bulb into your ZigBee network. Try to configure power-on behavior.

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC26X2R1
Adapter firmware version: `info  2020-11-08 10:59:35: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20200417,"transportrev":2},"type":"zStack3x0"}'
`

Koenkk, 2020-11-08:
You indeed need to send a MQTT command to `zigbee2mqtt/<DEVICE>/set/hue_power_on_behavior`.

u17194907425, 2020-11-08:
@Koenkk 
I'd like to make it more clear on the www.zigbee2mqtt.io website, what would be the best way to start? Just propose a change via https://github.com/Koenkk/zigbee2mqtt.io ?

Koenkk, 2020-11-09:
You can click the pencil icon here: https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/9290023349.md (just edit one file, I will make sure it will be added to all other Hue devices)

closed 2020-11-10 by Koenkk

letty23, 2024-06-03:
Zwar schon ein sehr alter Beitrag, aber kann mir jemand sagen, wo ich diese beschriebenen Set-MQTT Befehl reinschreiben muss?
Ich finde keine Konsole, wo ich diese Befehl absenden kann.

prometoys, 2024-08-15:
@letty23: Du musst es via mqtt schicken. Z.B. mit einem Tool wie MQTT-Explorer oder via NodeRed. Du könntest theoretisch auch im zigbee2mqtt-Webinterface die Entwickler-Konsole (dev-console) des Gerätes nehmen, aber das ist Quatsch, weil du es dann ja gleich mit dem Web-Interface die Sachen einstellen kannst. (Gerät auswählen, dann unter dem Reiter Details)