Koenkk/zigbee2mqtt #4903: Philips Hue - Firmware is newer than latest firmware online.

u17194907425, 2020-11-08:
## What happened
```
debug 2020-11-08 11:07:44: Received Zigbee message from 'philips_hue_testing_2', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":16780812,"imageType":276,"manufacturerCode":4107}' from endpoint 11 with groupID 0
debug 2020-11-08 11:07:44: Check if update available for '0x0017880108fe7a41' (LWA009)
debug 2020-11-08 11:07:44: Is new image available for '0x0017880108fe7a41', current '{"fieldControl":0,"manufacturerCode":4107,"imageType":276,"fileVersion":16780812}', latest meta '{"fileVersion":16780804,"fileSize":324910,"url":"http://fds.dc1.philips.com/firmware/ZGB_100B_0114/16780804/100B-0114-01000E04-ConfLightBLE-Lamps-EFR32MG21.zigbee"}'
debug 2020-11-08 11:07:44: Update available for '0x0017880108fe7a41': NO
warn  2020-11-08 11:07:44: Firmware on '0x0017880108fe7a41' is newer than latest firmware online.
```

This message is reported for the Philips Hue bulb i have (this specific model https://www.zigbee2mqtt.io/devices/9290023349.html). Does not block any functionality.

I just wonder if Philips has changed the functionality for the OTA and zigbee2mqtt is possibly looking in an outdated place. Maybe I can help somehow to debug this?

## What did you expect to happen
No _Firmware is newer than latest firmware online._ messages

## How to reproduce it (minimal and precise)

1. Get Philips Hue bulb 9290023349
1. Pair it with Philips Hue Bluetooth application
1. Check for updates and install. As of now it is 1.65.11_hB798F2BF
1. Have this bulb connected to your ZigBee network
1. Observe log

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC26X2R1
Adapter firmware version: `info 2020-11-08 10:59:35: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20200417,"transportrev":2},"type":"zStack3x0"}'`

Koenkk, 2020-11-08:
Philips does officially not publish firmware files for its devices. Some people managed to reverse engineer some URLs, however since every firmware has its unique URL this can be come outdated. The URLs are listed here: https://github.com/Koenkk/zigbee-OTA/blob/54fa9052379ded2ab8c77bb1c5a07e285172a5e5/index.json#L63

closed 2020-11-08 by Koenkk

TheJulianJES, 2020-11-10:
Here are the links for the new firmwares:

https://storage.googleapis.com/firmware-hue/100b-111/16782852/ad7789b8-8770-44af-b8c0-cd001f5e0eee/100B-0111-01001604-ConfLight-ModuLum-EFR32MG13.zigbee (version 22.4)
https://storage.googleapis.com/firmware-hue/100b-112/16782596/6eeaf8f7-e2cc-4881-b58d-c985c409df16/100B-0112-01001504-ConfLightBLE-Lamps-EFR32MG13.zigbee (version 21.4)
https://storage.googleapis.com/firmware-hue/100b-114/16780812/be63f1c3-b395-44b1-8879-84c9d252f0fd/100B-0114-01000E0C-ConfLightBLE-Lamps-EFR32MG21.zigbee (version 14.12)
https://storage.googleapis.com/firmware-hue/100b-117/16780544/192a44f1-e3bf-46b9-aec8-499e273e1b8e/100B-0117-01000D00-ConfLightBLE-ModuLum-EFR32MG21.zigbee (version 1.76.10) (this seems to be new: manufacture_id=4107, image_type=279 -- there isn't a prior version for this LED Strip on the repo)

They were released on 2020-11-04 and 2020-11-06
The official changelog (https://www.philips-hue.com/de-de/support/veroffentlichungshinweis/lamps-luminaires) is not updated, but the changes are mostly related to better Bluetooth connectivity.

@mkaay (who already added every other Hue firmware image to the zigbee-OTA repo) uses another way of grabbing these images. It might be cleaner, if he grabs them all again (so the download location is more similar to the current images).

mkaay, 2020-11-10:
I just checked the Bridge API, no new releases yet. Looks like this update is only available via the Bluetooth App.

Koenkk, 2020-11-10:
@TheJulianJES thanks, added the new firmwars.

u17194907425, 2020-11-11:
I wish I can easily sniff traffic from my android app like I do it with Charles on my PC... I could probably grab the links from the Bluetooth App

TheJulianJES, 2020-11-11:
> I wish I can easily sniff traffic from my android app like I do it with Charles on my PC... I could probably grab the links from the Bluetooth App

Koenkk added the updates from the Bluetooth app (links I posted) to the zigbee-ota repo.

mkaay, 2020-11-15:
I just tried looking up the newly found ImageID 270 with the bridge API and i got a result with the version 16780288 which is 01000C00 in hex.
The URL that got returned was http://fds.dc1.philips.com/firmware/ZGB_100B_0117/16780288/100B-0117-01000C00-ConfLightBLE-ModuLum-EFR32MG21.zigbee and you can see a pattern in this.
I tried changing the version (hex and dec) to the ones @TheJulianJES  found and it looks like you can get arbitrary firmware via this scheme, as long as you know what version is released (and what the file name is).
ex: http://fds.dc1.philips.com/firmware/ZGB_100B_0117/16780544/100B-0117-01000D00-ConfLightBLE-ModuLum-EFR32MG21.zigbee
Interesting stuff so far... I havent tried to increment the version beyond the newest value.
It may be possible to get other versions, but if the Hue Bridge doesn't make these versions available, they are probably not relevant to the general Zigbee ecosystem.
I will keep checking the Bridge API for new versions periodically, or even implement some kind of automation in the future though :)

Koenkk, 2020-11-15:
@mkaay thanks for investigating, in case you want to add new firmwares to the zigbee-OTA repo, you can follow https://github.com/Koenkk/zigbee-OTA#adding-new-and-updating-existing-ota-files