Koenkk/zigbee2mqtt #4904: Smart9 Zigbee Smoke Detector

chrisrock1984, 2020-11-08:
## Information about the device + link
https://de.aliexpress.com/item/1005001383340876.html?spm=a2g0x.12057483.0.0.2c501270dFrSFn

## data/database.db entry of the device
{"id":77,"type":"EndDevice","ieeeAddr":"0x842e14fffef506a3","nwkAddr":26407,"manufId":0,"manufName":"_TYST11_t5p1vj8r","powerSource":"Battery","modelId":"5p1vj8r\u0000","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":0,"inClusterList":[0,3],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{"modelId":"5p1vj8r\u0000","manufacturerName":"_TYST11_t5p1vj8r","powerSource":3,"zclVersion":3,"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":"20180727"}}},"binds":[],"meta":{}}},"appVersion":73,"stackVersion":0,"hwVersion":1,"dateCode":"20180727","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1604826030793}

chrisrock1984, 2020-12-09:
Not so important device i guess or do you need more data?

Koenkk, 2020-12-09:
@chrisrock1984 to add support for a new device follow: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

closed 2020-12-17 by github-actions[bot]