Koenkk/zigbee2mqtt #4907: zigbee2mqtt startet nicht (zigbee-herdsman)

caleus2k, 2020-11-08:
Hey Leute

Ich habe hier ein Problem und hoffe das ihr mir helfen könnt. Ich habe mir einen ZigBee Stick CC2531 gekauft, dieser ist mit Default-Firmware von Koenkk vom 08.06.2019 installiert.
Wenn ich den anweisungen folge von

`https://www.zigbee2mqtt.io/`

läuft alles soweit durch aber wenn ich dann versuche es zu starten kommt folgendes

`
root@raspberrypi-4:/opt/zigbee2mqtt# npm start

> zigbee2mqtt@1.15.0 start /opt/zigbee2mqtt
> node index.js

Zigbee2MQTT:info  2020-11-08 15:28:32: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-11-08.15-28-32' filename: log.txt
Zigbee2MQTT:info  2020-11-08 15:28:32: Starting Zigbee2MQTT version 1.15.0 (commit #ed8b4e5)
Zigbee2MQTT:info  2020-11-08 15:28:32: Starting zigbee-herdsman...
Zigbee2MQTT:error 2020-11-08 15:28:33: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-11-08 15:28:33: Failed to start zigbee
Zigbee2MQTT:error 2020-11-08 15:28:33: Exiting...
Zigbee2MQTT:error 2020-11-08 15:28:33: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:136:32)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:27:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:23:12)
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:134:49)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.15.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.15.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-11-08T14_28_33_067Z-debug.log
`

ich weiß nicht wirklich was ich machen soll.

MfG Caleus
Beitrag editieren

Koenkk, 2020-11-09:
English only

closed 2020-11-09 by Koenkk

skullbook, 2020-12-26:
Hey,

i Know its not everyone's strength to write and tell problems in englisch.
Perhaps i could help, hab similar Problems with Zigbee2MQTT.

I saw late, that one Package hat an issue while installing.
Direktly after Installing, there was a hint line for that issue with e request zu fix it.
therefor i used the command 'npm audit fix'.

For German discription follow [https://openhabforum.de/viewtopic.php?p=27447#p27447](url)

Skullbook