Koenkk/zigbee2mqtt #4908: Cannot pair any devices anymore

frag-duino, 2020-11-08:
## What happened
I have about 22 Xiaomi Devices (Aqara Switches, Aqara Window-Sensors, Lumi Actors, PIR sensors, etc.). Now I have bought another WXKG11LM (Single Switch), but I cannot pair it. I tried everything (long press, short press, new battery, another USB Port for the Module, get close to the CC2531), but there is nothing in the Zigbee2Mqtt log. So I tried zigbee-herdsman debugging, but there is no message either. 
Then I tried to pair another window-contact, but also nothing happens. Everything else is woking perfectly: I have 2 Zigbee routers (lumi) and all device communicate perfectly.
Has anybody an idea, what I can do to fix it?

## What did you expect to happen
I expect the device to be paired.

## How to reproduce it (minimal and precise)
Start Zigbee2Mqtt, Set Permit-Join to true (through MQTT command). Long push the little button on the button.

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC2531
Adapter firmware version:{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}
Herdsmen debug: https://pastebin.com/e46ZdSM0
Configuration.yaml: https://pastebin.com/595qVTGh

frag-duino, 2020-11-09:
Funny story: today my Pi crashed (he never does that), and I had to reboot (Power off, power on).
Now I can pair again! (WTF :))
Anyway. Thanks for your attention

closed 2020-11-09 by frag-duino

19leva89, 2020-11-10:
permit_join: true

frag-duino, 2020-11-10:
> 
> 
> permit_join: true

As I said, I've activated permit_join through MQTT, not through the config.

19leva89, 2020-11-10:
Попробуйте физически вытащить и вставить обратно CC2531