Koenkk/zigbee2mqtt #4911: Moes BHT-002 spamming queue

Courty40, 2020-11-09:
## What happened
When in user the Moes BHT-002 thermostat doesn't send one message, it seems to send 30 in one go, all repeats of the last but with now message serial numbers

## What did you expect to happen
Single message every X seconds or on change

## How to reproduce it (minimal and precise)
Add Moes BHT-002 to ZIGBEE2MQTT, monitor the /# queue and change the temp. queue gets flooded with a chain of messages

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC26X2R1
Adapter firmware version: Latest (did same on CCCC2531 as well)

1300371, 2020-11-09:
Hello, you can use the `debounce` parameter like 
```
'0xfffffffffffffffffffffffffff':
  friendly_name: thermostat
  debounce: 1
```

See https://www.zigbee2mqtt.io/information/configuration.html#all-devices

Courty40, 2020-11-10:
debounce: 1   defo helps a lot but concerned that it sends so much to the queue. reading up on it, it looks like a few other devices do this too. poor hardware/firmware design by the looks of it.
Thanks for the nudge in the right direction..

closed 2020-11-10 by Courty40

1300371, 2020-11-10:
Yes. It's a pity

Falconpage, 2020-11-13:
Thanks for this tip, debounce: 1 solve my logs spam issue