Koenkk/zigbee2mqtt #4912: After each restart of HA connection to MQTT/zigbee2mqtt is lost and all devices are unavailable

skumka, 2020-11-09:
## Problem
After each restart of HA connection to MQTT is lost and all devices are unavailable. You need to stop and start zigbee2mqtt to get it to work again.

## Description

Restart HA from within HA.
When UI starts you can see the status of devices and switches available/active for few seconds. After that time all become unavailable.
Problem reported to HomeAssistant but after they analyze the problem seems to be in zigbee2mqtt.
To fix the problem you need to stop and restart zigbee2mqtt service
With version 1.15 all worked fine

## Debug info
Zigbee2MQTT version: 1.16 and 1.16.1

marklagendijk, 2020-11-09:
I'm having the same issue.

Koenkk, 2020-11-09:
Can you provide the debug log when Zigbee2MQTT is started for the first time after a HA restart?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

dennyreiter, 2020-11-09:
I'm having this problem also, but, I just restarted HA and and it didn't happen.  I thought it was something I had done, because I have/had been using mqtt_statestream between two HA instances and a script to force sensor discovery, but Wednesday my main instance crashed and was filled with sensors like sensor.shower_temperature_2_2_2_2_2_2_2_2_2_2_2_2
So I'm thinking that even though this might be a Z2M issues, that something changed in HA MQTT to reveal it and my statestream skulduggery. I have debug logging set to turn on for the next restart if it happens again.

bobbyschuitemaker, 2020-11-10:
Same here. Every day i have to restart once or twice.
I will see if I can send a log.

fperezm, 2020-11-10:
I'm also having the same problem, I thought it was my thing. I put the log at debug level but nothing is seen, only that the devices stop updating and does not send anything through MQTT

BinaryBrain, 2020-11-11:
I have a similar problem. It seems that the new key `availability` in the payload for HA is not supported by HA (tested with stable 1.117.5 and dev on 1.118dev).

The issue seems to come from this major change: https://github.com/Koenkk/zigbee2mqtt/commit/b2049b449b58162fdead7f0b183b2587315097bd#diff-0eca7eb0935564159201825dcf3b6a0c41cf0d23c8a183a7a2c618a933af5ee7

marklagendijk, 2020-11-11:
I just downgraded zigbee2mqtt to 1.15.0 and everything works. If the issue re-appears even with 1.15.0 I'll report back.

skumka, 2020-11-11:
I reverted to the 1.15 and problem does not exist. So 100% it is related to 1.16 changes. I am sure for 90% chances that due to "ameliorations" for MQTT

Koenkk, 2020-11-11:
Please provide the logging as requested in https://github.com/Koenkk/zigbee2mqtt/issues/4912#issuecomment-724112855 , otherwise it is impossible for me to fix anything.

skumka, 2020-11-11:
@Koenkk Here it is link to recorded video: https://streamable.com/r2pqre

and log file:
[log_2020-11-11.20-37-57.log](https://github.com/Koenkk/zigbee2mqtt/files/5526418/log_2020-11-11.20-37-57.log)

IronButterfly, 2020-11-12:
The same issue with 1.16.1

jesperldk, 2020-11-12:
this is a problem for me as well, I'm on HA 0.117.2

Koenkk, 2020-11-12:
@skumka 

- Have you ever enabled the availabilty feature?
- If you use a MQTT client and subscribe to `zigbee2mqtt/Entrance Door/availability`, do you get anything?

dennyreiter, 2020-11-13:
I'm not sure that this is solely a Z2M problem.  I'm now seeing this with most of my Tasmota devices, also.

skumka, 2020-11-13:
@Koenkk I had it enabled few weeks back for 2 days, but I switched it off per your recommendation linked with other issue I have reported.
What is the reason you ask about it?
WRT you 2nd question. I do not have availability activated so I am not sure if this should be checked.

jesperldk, 2020-11-13:
Hmm, I see it only for zigbee2mqtt; I do not see it for any of my
Tasmotas nor for miflora-mqtt.

dennyreiter, 2020-11-13:
I noticed "mqtt: Exception in availability_message_received" in my logs
and Google led me to this issue, which seems to still be open?

https://github.com/home-assistant/core/issues/40166

I did recently have config changes (CC2531 died and I replaced it with a CC2652R.) Wish it would happen again so I could grab the logs 😁

Koenkk, 2020-11-13:
@skumka can you check: If you use a MQTT client and subscribe to zigbee2mqtt/Entrance Door/availability, do you get anything?

I believe this issue happens after having turned on the availability feature once (while having it off now). There is still a retained `offline` message in `zigbee2mqtt/Entrance Door/availability` which will HA receive on startup. But I need your help to confirm :)

jesperldk, 2020-11-13:
I have never had the availability feature turn on.

skumka, 2020-11-13:
@Koenkk I have checked: 
[image]

Should I remove it?

jesperldk, 2020-11-13:
Well, I don't think I ever had availability enabled. Even after reading the doc I am not quite sure how to enable it if I wanted to ;-)

However, when I look in the homeassistant config topics, they contain "availability_topic":"zigbee2mqtt/bridge/state", but there is no value for the "zigbee2mqtt/bridge/state" topic. Also, the topic name seems strange. The homeassistant config topic for the same device also contains "state_topic":"zigbee2mqtt/motion2", and that is a much more meaningful topic name, and indeed this topic contains the state as expected.

This seems incorrect? Could HA have changed its reaction on this broken reference?

Koenkk, 2020-11-13:
@skumka issue should be fixed

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

@jesperldk looks the root cause for you is different, to be sure, can you first check with the latest dev and if that doesn't fix provide me the debug log when Zigbee2MQTT starts?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

skumka, 2020-11-13:
@Koenkk ok... but I would prefer to merge the code changes in my current stream rather than switch to dev.
I hope this will work for you too.

Koenkk, 2020-11-13:
@skumka you mean master? These will be included in the next release on 1 December. If you don't want to switch to dev you can clear all retained MQTT messages (https://community.openhab.org/t/clearing-mqtt-retained-messages/58221)

skumka, 2020-11-13:
Just for test I can make a merge on my environment (master) and let you know if it works.

skumka, 2020-11-13:
I know now that cleaning retained msgs will solve my issue, but for the fact you spend time for fixing it in code I may contribute by testing :-)

skumka, 2020-11-14:
@Koenkk Updated to dev and tested the fix. It works with the code changes. Closing the issue.

closed 2020-11-14 by skumka

jesperldk, 2020-11-14:
@Koenkk this fixes things for me as well, although the case seemed different. (well, pulling fixed it, I was on dev 1.15.0 #ed8b4e5 before)

I did notice some differences:
- Before zigbee2mqtt/bridge/state had the retain flag, now it does not
- Before the devices did not appear under zigbee2mqtt until the first message was received, now the appear right away

Checking the docs, I can see this behavior comes from cache_state: true and family. I newer have had them in my configuration.yaml, did the defaults perhaps change? I can see I ought to have had them on, but everything worked fine until it stopped working at every HA reboot since quite recently.

Anyway, thanks a lot, I'm a fan :-)

Koenkk, 2020-11-14:
@jesperldk bridge/state always has to retain flag set to true, but if you use an MQTT client this will only show as retain true when you initially connect with the server. `cache_state: true` was always the default and this did not change. Do I understand correctly that the problem is fixed for you?

jesperldk, 2020-11-14:
Yes, it is fixed. Before, starting a few weeks ago, when I restarted HA
alle z2m devices became unavailabe until I also restarted z2m. After
upgrade it works again. Weird.

jesperldk, 2020-11-14:
@Koenkk 

Damn, I still have the problem. Apparently only now and then. I am fairly certain that it came when I upgrade HA to some 0.117.x version. Probably some problem in HA? However, my tasmotas and miflowers does not have this problem.