Koenkk/zigbee2mqtt #4914: Can't set temperature tuya smart valve from google home

ale1800, 2020-11-09:
Hi everyone. Received, added and updated TS0601 tuya smart valve in z2m and everything works just fine through lovelace and alexa but not with google home. Through Google Home app I can only see the current temperature of the valve but can't set any mode or temperature. Also with voice I can't do anything. I have the official google home and alexa integrations thanks to Nabucasa. What can i do?
Thank you everyone!

## Debug info
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC2531
Adapter firmware version:

StimpyMGS, 2020-11-21:
Have you synced your devices in Google home? Say "hey Google sync devices" and try again

ale1800, 2020-11-22:
Sure, i can see trv from the app but i can't do anything

scoofz, 2020-12-11:
I have the exact same problem and it's a shame since I bought the google nest hub mainly to contol temperature but the same problem happens even when connected without using z2m and connecting with a zigbee hub and the tuya smart app... I also noticed there was no other mode than eco ...

ale1800, 2020-12-13:
I think a workaround can be creating a generic thermostat and linking It to the mqtt thermostat

ale1800, 2020-12-13:
Ok i found the solution. Now i'm able to set temperature from google nest hub. you have to create a fake generic thermostat and with this custom component https://github.com/daenny/climate_group you can create a "group" labeled as climate. Now you can use this new group on google nest hub to control the thermostat. I only solved for temperature settings. Not for modes, I think it's quite difficult.
[image]

closed 2021-01-20 by github-actions[bot]

tiefenthaler, 2024-12-14:
Same here, I can properly add the device to google home. But I cannot set temperature or configure it within the google home app. Only the current status is displayed