Koenkk/zigbee2mqtt #4915: EcoDim E27 filament bulb G125

Melissen71, 2020-11-09:
## Information about the device + link
..
EcoDim E27 G125 Zigbee bulb

## data/database.db entry of the device
..

The bulb can be paired with the cc2531 on a raspberry pi. But it doesn't show up in Domoticz?
I'm a newbee and don't know what to do, or what steps I've to do to support you with the right data

Using Putty on the Raspberry Pi. The lamp is present in the Configuration.yaml

Melissen71, 2020-11-10:
i got some info about this device in a text file....

Info looks similair to a Ikea bulb.... how can i add this to Domoticz. Or is there someone who can help me?
It's about the Paulmann bulb !!!

[zigbee na opstarten PAULMANN.txt](https://github.com/Koenkk/zigbee2mqtt/files/5519601/zigbee.na.opstarten.PAULMANN.txt)

Koenkk, 2020-11-10:
Looks that the device is already supported (`{"ieeeAddr":"0x00158d0001273925","type":"Router","networkAddress":2813,"model":"50064","vendor":"Paulmann","description":"SmartHome led spot","friendly_name":"0x00158d0001273925","manufacturerID":4137,"manufacturerName":"ZigBee/CCT","powerSource":"Mains (single phase)","modelID":"CCT Light","hardwareVersion":21,"softwareBuildID":"00.00.01","dateCode":"20200505","lastSeen":1605033647909}
`) you should create an issue here: https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin/issues to add domitcz support.

closed 2020-11-10 by Koenkk

Melissen71, 2020-11-11:
Created issue in other section