Koenkk/zigbee2mqtt #4916: Initial Group State when turned on

Anjerlaan, 2020-11-09:
## Is your feature request related to a problem? Please describe
Most people who have Bulbs or a Group of Bulbs controlled with a remote dimmer (smart or not smart), would like to have an initial state for brightness and/or color (the state which is used most) if turned ON.
In Z2m the state, when turning OFF, is remembered. But this is not always the intended state for the group when turned ON again.

I am using the **"domoticz-zigbee2mqtt-plugin"**.
This plug-in automatically discovers Z2M devices and groups. Depending on the Z2M group name, Dimmers Switches are created in Domoticz. The problem is that it is difficult (with a script) to set the slider to an initial level, because this function "dimTo(80).silent()" turns the group always ON to level 80....even when it is turned OFF or the slider is used.

The silent() function in this script language does not trigger follow-up events...which means I can turn off a device in a certain state. It looks like that Z2M doesn't have a function with this functionality is it?

## Describe the solution you'd like
As far as I know it is not possible to set an initial brightness\color state for Groups when turned ON?
Would this be possible?

sjorge, 2020-11-09:
It's a zigbee problem, if you change the brightness (level) of a bulb, it will turn to that level... setting it to 128 will turn it on and set to 50%.

Seen support was recently added though! So you could in theory make a scene that has the bulbs configured with the initial state and then have that state recalled to reset then.

Anjerlaan, 2020-11-10:
I have been testing with Scenes, this is a nice option. 
However the color_temp conversion is not correct in the scene attributes...
There is a huge difference in color_temp if I sent:

{ "state": "ON", "brightness": 120, "color_temp": 350 }

Or add a Scene with the same values and recall it:

{
    "scene_add": {
        "ID": 1,
        "state": "ON",
        "brightness": 120,
        "color_temp": 350
    }
}

I use the "color":{"x":0.447991567541816,"y":0.407559831563923} attribute instead, this value represents the correct color_temp

sjorge, 2020-11-10:
It might be worth filing the color_temp one as a separate issue/bug, AFAIK setting it via scene_add and directly should have the exact same behavior.

closed 2020-12-19 by github-actions[bot]