Koenkk/zigbee2mqtt #4917: TH1300ZB Thermostat Lacks Temperature Unit in HA Climate discovery

kbickar, 2020-11-09:
## What happened
The TH1300ZB thermostat is discovered by HomeAsssistant, but without a unit it defaults to the system unit.  Since this device always reports in C and my system defaults to F, it is showing inaccurate numbers.

Here is the discovery message:
```json
{
          "action_template": "{% set values = {'idle':'off','heat':'heating','cool':'cooling','fan only':'fan'} %}{{ values[value_json.running_state] }}",
          "action_topic": "zigbee2mqtt/0x500b91400002202f",
          "availability_topic": "zigbee2mqtt/bridge/state",
          "current_temperature_template": "{{ value_json.local_temperature }}",
          "current_temperature_topic": "zigbee2mqtt/0x500b91400002202f",
          "device": {
                    "identifiers": ["zigbee2mqtt_0x500b91400002202f"],
                    "manufacturer": "Sinope",
                    "model": "Zigbee smart floor heating thermostat (TH1300ZB)",
                    "name": "0x500b91400002202f",
                    "sw_version": "Zigbee2mqtt 1.14.1"
          },
          "json_attributes_topic": "zigbee2mqtt/0x500b91400002202f",
          "max_temp": "30",
          "min_temp": "7",
          "mode_command_topic": "zigbee2mqtt/0x500b91400002202f/set/system_mode",
          "mode_state_template": "{{ value_json.system_mode }}",
          "mode_state_topic": "zigbee2mqtt/0x500b91400002202f",
          "modes": [
                    "off",
                    "auto",
                    "heat"
          ],
          "name": "0x500b91400002202f_climate",
          "temp_step": 1,
          "temperature_command_topic": "zigbee2mqtt/0x500b91400002202f/set/occupied_heating_setpoint",
          "temperature_state_template": "{{ value_json.occupied_heating_setpoint }}",
          "temperature_state_topic": "zigbee2mqtt/0x500b91400002202f",
          "unique_id": "0x500b91400002202f_climate_zigbee2mqtt"
}
```
## What did you expect to happen
The discovery message should include the extra arg: `"temperature_unit": "C"`

## How to reproduce it (minimal and precise)
Pair the device. 

## Debug info
Zigbee2MQTT version: 1.14.1
Adapter hardware: CC2531
Adapter firmware version: latest

Koenkk, 2020-11-10:
You are using an old Zigbee2MQTT version, can you check with the latest one?

kbickar, 2020-11-10:
Looks like it's added in Zigbee2MQTT 1.16.1, thanks!

closed 2020-11-10 by kbickar