Koenkk/zigbee2mqtt #4918: Zigbee device are not receiving command anymore

sybux2000, 2020-11-09:
## What happened
I think that after update to release 1.16 my ZigBee device failed to received commands. In the log, I can see errors like : 
```
Zigbee2MQTT:error 2020-11-09 22:26:38: Publish 'set' 'state' to 'mydeviceID' failed: 'Error: Command mydeviceID/11 genOnOff.on
({},
{
    "timeout": 10000,
    "disableResponse": false,
    "disableRecovery": false,
    "disableDefaultResponse": false,
    "direction": 0,
    "srcEndpoint": null,
    "reservedBits": 0,
    "manufacturerCode": null,
    "transactionSequenceNumber": null
}) 
failed (SREQ '--> AF - dataRequest - {
    "dstaddr": 36657,
    "destendpoint": 11,
    "srcendpoint": 1,
    "clusterid": 6,
    "transid": 10,
    "options": 0,
    "radius": 30,
    "len": 3,
    "data": {
        "type": "Buffer",
        "data": [
            1,
            8,
            1
        ]
    }
}' 
failed with status '(0x02: INVALID_PARAM)' (expected '(0x00: SUCCESS)'))'
```
## What did you expect to happen
command to be run without errors :)

## How to reproduce it (minimal and precise)
click on buttons on ZigBee Device, send command thru HA to poweron/off a device

## Debug info
Zigbee2MQTT version:1.16.1
Adapter hardware: CC2531
Adapter firmware version: can't remember how to get this

Koenkk, 2020-11-10:
It seems your stick crashed. I recommend reflashing it with: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing which provides more stability

sybux2000, 2020-11-10:
I've just reflashed the ZigBee key. 
1st point : it's still working after flash :)
2nd point : I can see in zigbee add on the loading log with all devices
3rd point : I still can't send command but now I don't have any error messages :(

What can I provide you to try debuging the situation ?

Koenkk, 2020-11-10:
Can you provide the herdsman debug log when sending messages?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

sybux2000, 2020-11-10:
I can't really get debug information when for exemple powering on a lamp or click on a remote command button. But I've captured this for example :
`2020-11-10T21:39:36.461Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x001788010113fa1d:65524/11 (0,3,1)
2020-11-10T21:39:36.462Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":65524,"destendpoint":11,"srcendpoint":1,"clusterid":6,"transid":5,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,3,0]}}
2020-11-10T21:39:36.462Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,13,36,1,244,255,11,1,6,0,5,0,30,3,1,3,0,53]
2020-11-10T21:39:36.476Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2020-11-10T21:39:36.476Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2020-11-10T21:39:36.476Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2020-11-10T21:39:36.476Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2020-11-10T21:39:36.477Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-11-10T21:39:37.202Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,205,1,5,14]
2020-11-10T21:39:37.203Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,205,1,5,14]
2020-11-10T21:39:37.203Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [205,1,5] - 14
2020-11-10T21:39:37.203Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":205,"endpoint":1,"transid":5}
2020-11-10T21:39:37.203Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-11-10T21:39:37.204Z zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x001788010113fa1d:65524,205,3)
2020-11-10T21:39:37.204Z zigbee-herdsman:adapter:zStack:adapter Wait 2000ms
2020-11-10T21:39:39.207Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x001788010113fa1d:65524/11 (0,4,1)
2020-11-10T21:39:39.209Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":65524,"destendpoint":11,"srcendpoint":1,"clusterid":6,"transid":6,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,3,0]}}
2020-11-10T21:39:39.210Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,13,36,1,244,255,11,1,6,0,6,0,30,3,1,3,0,54]
2020-11-10T21:39:39.218Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2020-11-10T21:39:39.218Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2020-11-10T21:39:39.219Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2020-11-10T21:39:39.219Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2020-11-10T21:39:39.219Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-11-10T21:39:39.810Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,205,1,6,13]
2020-11-10T21:39:39.810Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,205,1,6,13]
2020-11-10T21:39:39.810Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [205,1,6] - 13
2020-11-10T21:39:39.811Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":205,"endpoint":1,"transid":6}
2020-11-10T21:39:39.811Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-11-10T21:39:39.811Z zigbee-herdsman:adapter:zStack:adapter Data confirm error (0x001788010113fa1d:65524,205,4)
2020-11-10T21:39:39.812Z zigbee-herdsman:controller:endpoint Command 0x001788010113fa1d/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'No network route' (205))
Zigbee2MQTT:error 2020-11-10 22:39:39: Publish 'set' 'state' to '0x001788010113fa1d' failed: 'Error: Command 0x001788010113fa1d/11 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'No network route' (205))'`
Hope it could help or tell me more action to do !

sybux2000, 2020-11-10:
Got a new debug frame. This is the result of a shut off a lamp :
`2020-11-10T21:55:01.045Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x001788010626a3e3:32389/11 (0,0,1)
2020-11-10T21:55:01.045Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":32389,"destendpoint":11,"srcendpoint":1,"clusterid":6,"transid":53,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,19,0]}}
2020-11-10T21:55:01.046Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,13,36,1,133,126,11,1,6,0,53,0,30,3,1,19,0,229]
2020-11-10T21:55:01.057Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2020-11-10T21:55:01.057Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2020-11-10T21:55:01.058Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2020-11-10T21:55:01.058Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2020-11-10T21:55:01.058Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-11-10T21:55:01.094Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,53,243]
2020-11-10T21:55:01.094Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,53,243]
2020-11-10T21:55:01.094Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,53] - 243
2020-11-10T21:55:01.095Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":53}
2020-11-10T21:55:01.095Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []`

Koenkk, 2020-11-11:
And can you still receive messages from devices in the network?

sybux2000, 2020-11-11:
I'm not sure of what you're asking.
In the log file, I can see payload such as :
'''Zigbee2MQTT:info  2020-11-11 18:15:55: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":[],"type":"groups"}'
Zigbee2MQTT:info  2020-11-11 18:15:55: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":[{"dateCode":"20190619","friendly_name":"Coordinator","ieeeAddr'''

Koenkk, 2020-11-11:
But also e.g. occupancy from an occupancy sensor? (or temperature from a temperature sensor?). 

For the devices that are not reachable I recommend to repair them.

sybux2000, 2020-11-11:
I've tried to repair a philips Hue dimmer switch but nothing can be see in the log about pairing device as I seen in the past when doing this operation.
The only info passing time to time in the log is : 
`Zigbee2MQTT:info  2020-11-11 22:20:08: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":[],"type":"groups"}'` followed by a response with all devices.
Is my system completly stuck ? should I reset it all ?

str3000, 2020-11-12:
I have the same problem with the Xiaomi button:  WXKG03LM
After restart I received only the first message (incl. action and click)
Afterwards the button is dead. I see in the zigbee2mqtt web log that no further message was send.
But, it looks like that other buttons works fine. perhaps it depends on the firmware of the button.

I make tests with two different coorinators.
I test with different legacy parameters.
homeassistant is enabled 
zigbee2mqtt@1.16.1 (on bare metal)
mqtt:  current version in docker
I see behavior ioBroker

Koenkk, 2020-11-12:
@sybux2000 I'm not sure what happend but a full reset is maybe the best thing to do. Reflash stick, remove database.db from the data directory and repair.

sybux2000, 2020-11-15:
Sorry for this late answer. I was looking the web for the process of erasing all when I found a post about distance between the USB stick and some Wifi device. My Pi has Wifi disabled but not the other one present in the same rack. So I've plug my USB stick to a long USB cable and now everything is nearly working ....
I said nearly because I can command all lamps from the addon UI, I can control them from HA but I can't add new device.
Another question about group. I've defined groups in the addon but I can't see them in HA. Is there a way to populate them ?
Thx you again in advance for your help.

closed 2020-12-24 by github-actions[bot]