Koenkk/zigbee2mqtt #4922: adding tradfri outlet to hassio: extra keys not allowed @ data['availability']

tim-vk, 2020-11-10:
## What happened

Paired Tradfri outlet through homeassistant to zigbee2mqtt. Confirmation is given that device is paired & device can be controlled through zigbee2mqtt frontend. But no device is added to homeassistant. Discovery is turned on in configuration.yaml.

The home assistant log shows the following errors: [Pastebin link](https://pastebin.com/AxRRKrnX).
Using a 3rd party MQTT explorer also shows this "availability" key.

## What did you expect to happen

The device will be added to home assistant after being paired.

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.16.1 running in docker.
Adapter hardware: CC2531
mosquitto: latest, running in docker
Homeassistant: latest, running docker

Koenkk, 2020-11-10:
What Home Assistant version are you running?

tim-vk, 2020-11-10:
... I feel like a fool:

Apparently i had an old image of homeassistant that kept being used with updates. Forcing an update to the actual latest version fixed it.

closed 2020-11-10 by tim-vk