Koenkk/zigbee2mqtt #4925: Tuya thermostatic valve - Exception in async_discover when dispatching

perseus177, 2020-11-10:
## What happened
I successfully paired second TRV to zigbee2mqtt  but it not displayed in MQTT integration 

## What did you expect to happen
Displayed in MQTT integration without any errors for use it in HA

## How to reproduce it (minimal and precise)
1. Pair TRV in zigbee2mqtt 

## Debug info
HA version: 112.2
Mosquitto broker:  5.1
Zigbee2MQTT version: 1.16.1
Adapter hardware: CC2531
Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190425,"transportrev":2},"type":"zStack12"}'

## Logs
`
    Exception in async_discover when dispatching 'mqtt_discovery_new_switch_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/0xbc33acfffe37e579/availability'}], 'command_topic': 'zigbee2mqtt/0xbc33acfffe37e579/set/valve_detection', 'device': {'identifiers': ['zigbee2mqtt_0xbc33acfffe37e579'], 'manufacturer': 'TuYa', 'model': 'Radiator valve with thermostat (TS0601_thermostat)', 'name': '0xbc33acfffe37e579', 'sw_version': 'Zigbee2MQTT 1.16.1'}, 'json_attributes_topic': 'zigbee2mqtt/0xbc33acfffe37e579', 'name': '0xbc33acfffe37e579_valve_detection', 'payload_off': 'OFF', 'payload_on': 'ON', 'state_off': 'OFF', 'state_on': 'ON', 'state_topic': 'zigbee2mqtt/0xbc33acfffe37e579', 'unique_id': '0xbc33acfffe37e579_valve_detection_zigbee2mqtt', 'value_template': '{{ value_json.valve_detection }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/switch.py", line 82, in async_discover config = PLATFORM_SCHEMA(discovery_payload) File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: extra keys not allowed @ data['availability']
    Exception in async_discover when dispatching 'mqtt_discovery_new_switch_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/0x60a423fffea0fc8f/availability'}], 'command_topic': 'zigbee2mqtt/0x60a423fffea0fc8f/set/window_detection', 'device': {'identifiers': ['zigbee2mqtt_0x60a423fffea0fc8f'], 'manufacturer': 'TuYa', 'model': 'Radiator valve with thermostat (TS0601_thermostat)', 'name': '0x60a423fffea0fc8f', 'sw_version': 'Zigbee2MQTT 1.16.1'}, 'icon': 'mdi:window-open-variant', 'json_attributes_topic': 'zigbee2mqtt/0x60a423fffea0fc8f', 'name': '0x60a423fffea0fc8f_window_detection', 'payload_off': 'OFF', 'payload_on': 'ON', 'state_off': 'OFF', 'state_on': 'ON', 'state_topic': 'zigbee2mqtt/0x60a423fffea0fc8f', 'unique_id': '0x60a423fffea0fc8f_window_detection_zigbee2mqtt', 'value_template': '{{ value_json.window_detection }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/switch.py", line 82, in async_discover config = PLATFORM_SCHEMA(discovery_payload) File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: extra keys not allowed @ data['availability']
    Exception in async_discover when dispatching 'mqtt_discovery_new_switch_mqtt': ({'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/0x60a423fffea0fc8f/availability'}], 'command_topic': 'zigbee2mqtt/0x60a423fffea0fc8f/set/valve_detection', 'device': {'identifiers': ['zigbee2mqtt_0x60a423fffea0fc8f'], 'manufacturer': 'TuYa', 'model': 'Radiator valve with thermostat (TS0601_thermostat)', 'name': '0x60a423fffea0fc8f', 'sw_version': 'Zigbee2MQTT 1.16.1'}, 'json_attributes_topic': 'zigbee2mqtt/0x60a423fffea0fc8f', 'name': '0x60a423fffea0fc8f_valve_detection', 'payload_off': 'OFF', 'payload_on': 'ON', 'state_off': 'OFF', 'state_on': 'ON', 'state_topic': 'zigbee2mqtt/0x60a423fffea0fc8f', 'unique_id': '0x60a423fffea0fc8f_valve_detection_zigbee2mqtt', 'value_template': '{{ value_json.valve_detection }}', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/switch.py", line 82, in async_discover config = PLATFORM_SCHEMA(discovery_payload) File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: extra keys not allowed @ data['availability']
    Exception in async_discover when dispatching 'mqtt_discovery_new_climate_mqtt': ({'action_template': "{% set values = {'idle':'off','heat':'heating','cool':'cooling','fan only':'fan'} %}{{ values[value_json.running_state] }}", 'action_topic': 'zigbee2mqtt/0xbc33acfffe37e579', 'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/0xbc33acfffe37e579/availability'}], 'away_mode_command_topic': 'zigbee2mqtt/0xbc33acfffe37e579/set/away_mode', 'away_mode_state_template': '{{ value_json.away_mode }}', 'away_mode_state_topic': 'zigbee2mqtt/0xbc33acfffe37e579', 'current_temperature_template': '{{ value_json.local_temperature }}', 'current_temperature_topic': 'zigbee2mqtt/0xbc33acfffe37e579', 'device': {'identifiers': ['zigbee2mqtt_0xbc33acfffe37e579'], 'manufacturer': 'TuYa', 'model': 'Radiator valve with thermostat (TS0601_thermostat)', 'name': '0xbc33acfffe37e579', 'sw_version': 'Zigbee2MQTT 1.16.1'}, 'hold_command_topic': 'zigbee2mqtt/0xbc33acfffe37e579/set/preset', 'hold_modes': ['schedule', 'manual', 'boost', 'complex', 'comfort', 'eco'], 'hold_state_template': '{{ value_json.preset }}', 'hold_state_topic': 'zigbee2mqtt/0xbc33acfffe37e579', 'json_attributes_topic': 'zigbee2mqtt/0xbc33acfffe37e579', 'max_temp': '30', 'min_temp': '5', 'mode_command_topic': 'zigbee2mqtt/0xbc33acfffe37e579/set/system_mode', 'mode_state_template': '{{ value_json.system_mode }}', 'mode_state_topic': 'zigbee2mqtt/0xbc33acfffe37e579', 'modes': ['auto'], 'name': '0xbc33acfffe37e579_climate', 'temp_step': 0.5, 'temperature_command_topic': 'zigbee2mqtt/0xbc33acfffe37e579/set/current_heating_setpoint', 'temperature_state_template': '{{ value_json.current_heating_setpoint }}', 'temperature_state_topic': 'zigbee2mqtt/0xbc33acfffe37e579', 'temperature_unit': 'C', 'unique_id': '0xbc33acfffe37e579_climate_zigbee2mqtt', 'platform': 'mqtt'},) Traceback (most recent call last): File "/usr/src/homeassistant/homeassistant/components/mqtt/climate.py", line 251, in async_discover config = PLATFORM_SCHEMA(discovery_payload) File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: extra keys not allowed @ data['availability']
    Exception in async_discover when dispatching 'mqtt_discovery_new_climate_mqtt': ({'action_template': "{% set values = {'idle':'off','heat':'heating','cool':'cooling','fan only':'fan'} %}{{ values[value_json.running_state] }}", 'action_topic': 'zigbee2mqtt/0x60a423fffea0fc8f', 'availability': [{'topic': 'zigbee2mqtt/bridge/state'}, {'topic': 'zigbee2mqtt/0x60a423fffea0fc8f/availability'}], 'away_mode_command_topic': 'zigbee2mqtt/0x60a423fffea0fc8f/set/away_mode', 'away_mode_state_template': '{{ value_json.away_mode }}', 'away_mode_state_topic': 'zigbee2mqtt/0x60a423fffea0fc8f', 'current_temperature_template': '{{ value_json.local_temperature }}', 'current_temperature_topic': 'zigbee2mqtt/0x60a423fffea0fc8f', 'device': {'identifiers': ['zig [... cut ...] le "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 272, in __call__ return self._compiled([], data) File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 594, in validate_dict return base_validate(path, iteritems(data), out) File "/usr/local/lib/python3.7/site-packages/voluptuous/schema_builder.py", line 432, in validate_mapping raise er.MultipleInvalid(errors) voluptuous.error.MultipleInvalid: extra keys not allowed @ data['availability']
`

drzony, 2020-11-10:
It seems that Siterwell and Saswell are using the same model ID, you need to add fingerprint to your device i.e. model ID and manufacturer ID

perseus177, 2020-11-10:
@drzony  thanks for quick answer, can you tell me where should I add this fingerprint ?  file + path 
Thanks

drzony, 2020-11-10:
I thought you added it already, but I made a mistake, it was other user that changed that. I'm looking into the error you posted. Are you using a fresh pull without any modifications?

perseus177, 2020-11-10:
Zigbee2MQTT version: 1.16.1  without modifications

drzony, 2020-11-10:
Ok, the problem is that your HA is too old, please update your HA

perseus177, 2020-11-10:
I will try update HA to the latest version of HA, if will help I will write here and I will close the ticket

perseus177, 2020-11-10:
Update HA to 0.117.5 fix the problem. I am closing the issue. @drzony  thanks

closed 2020-11-10 by perseus177