Koenkk/zigbee2mqtt #4926: Color_temp value in scenes not correct

Anjerlaan, 2020-11-10:
## What happened
I have been testing with Scenes, this is a nice option.
But there is a huge difference in color_temp if I sent to a GROUP:

`{ "state": "ON", "brightness": 120, "color_temp": 350 }`

Or add a Scene with the same values and recall it:

```
{
    "scene_add": {
        "ID": 1,
        "state": "ON",
        "brightness": 120,
        "color_temp": 350
    }
}
```

When I use the `"color": {"x":0.447991567541816,"y":0.407559831563923}` attribute instead, this value represents the correct color_temp

## Debug info
Zigbee2MQTT version: v.1.15.0 
Adapter hardware: CC26X2R1
Adapter firmware version: (zStack3x0 20200925)

Koenkk, 2020-11-12:
I don't understand the issue, can you provide a list of steps to reproduce this and clear indicate what the problem is?

Anjerlaan, 2020-11-12:
> I don't understand the issue, can you provide a list of steps to reproduce this and clear indicate what the problem is?

This happens if I send the GROUP command:

`{ "state": "ON", "brightness": 120, "color_temp": 350 }`

The info shows a color value of `{"x":0.447991567541816,"y":0.407559831563923}`

See Log:
```
18:48:50: Received MQTT message on 'zigbee2mqtt/Keuken_Spots_ct/set' with data '{ "state": "ON", "brightness": 120, "color_temp": 350 }'
debug 2020-11-12 18:48:50: Publishing 'set' 'brightness' to 'Keuken_Spots_ct'
debug 2020-11-12 18:48:50: Received Zigbee message from 'Keuken_Spot2', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2020-11-12 18:48:50: MQTT publish: topic 'zigbee2mqtt/Keuken_Spots_ct', payload '{"brightness":120,"color":{"x":0.447991567541816,"y":0.407559831563923},"color_temp":350,"state":"ON"}'
info  2020-11-12 18:48:50: MQTT publish: topic 'zigbee2mqtt/Keuken_Spot2', payload '{"brightness":120,"color":{"x":0.447991567541816,"y":0.407559831563923},"color_temp":350,"linkquality":63,"state":"ON","update_available":false}'
debug 2020-11-12 18:48:50: Received Zigbee message from 'Keuken_Spot1', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2020-11-12 18:48:50: MQTT publish: topic 'zigbee2mqtt/Keuken_Spots_ct', payload '{"brightness":120,"color":{"x":0.447991567541816,"y":0.407559831563923},"color_temp":350,"state":"ON"}'
info  2020-11-12 18:48:50: MQTT publish: topic 'zigbee2mqtt/Keuken_Spot1', payload '{"brightness":120,"color":{"x":0.447991567541816,"y":0.407559831563923},"color_temp":350,"linkquality":69,"state":"ON","update_available":false}'
debug 2020-11-12 18:48:50: Received Zigbee message from 'Keuken_Spot3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2020-11-12 18:48:50: MQTT publish: topic 'zigbee2mqtt/Keuken_Spots_ct', payload '{"brightness":120,"color":{"x":0.447991567541816,"y":0.407559831563923},"color_temp":350,"state":"ON"}'
info  2020-11-12 18:48:50: MQTT publish: topic 'zigbee2mqtt/Keuken_Spot3', payload '{"brightness":120,"color":{"x":0.447991567541816,"y":0.407559831563923},"color_temp":350,"linkquality":66,"state":"ON","update_available":false}'
info  2020-11-12 18:48:50: MQTT publish: topic 'zigbee2mqtt/Keuken_Spot1', payload '{"brightness":120,"color":{"x":0.447991567541816,"y":0.407559831563923},"color_temp":350,"linkquality":69,"state":"ON","update_available":false}'
info  2020-11-12 18:48:50: MQTT publish: topic 'zigbee2mqtt/Keuken_Spots_ct', payload '{"brightness":120,"color":{"x":0.447991567541816,"y":0.407559831563923},"color_temp":350,"state":"ON"}'
debug 2020-11-12 18:48:50: Publishing 'set' 'color_temp' to 'Keuken_Spots_ct'
info  2020-11-12 18:48:50: MQTT publish: topic 'zigbee2mqtt/Keuken_Spot2', payload '{"brightness":120,"color":{"x":0.447991567541816,"y":0.407559831563923},"color_temp":350,"linkquality":63,"state":"ON","update_available":false}'
info  2020-11-12 18:48:50: MQTT publish: topic 'zigbee2mqtt/Keuken_Spot3', payload '{"brightness":120,"color":{"x":0.447991567541816,"y":0.407559831563923},"color_temp":350,"linkquality":66,"state":"ON","update_available":false}'
info  2020-11-12 18:48:50: MQTT publish: topic 'zigbee2mqtt/Keuken_Spot1', payload '{"brightness":120,"color":{"x":0.447991567541816,"y":0.407559831563923},"color_temp":350,"linkquality":69,"state":"ON","update_available":false}'
info  2020-11-12 18:48:50: MQTT publish: topic 'zigbee2mqtt/Keuken_Spots_ct', payload '{"brightness":120,"color":{"x":0.447991567541816,"y":0.407559831563923},"color_temp":350,"state":"ON"}'
info  2020-11-12 18:48:50: MQTT publish: topic 'zigbee2mqtt/Keuken_Spot2', payload '{"brightness":120,"color":{"x":0.447991567541816,"y":0.407559831563923},"color_temp":350,"linkquality":63,"state":"ON","update_available":false}'
info  2020-11-12 18:48:50: MQTT publish: topic 'zigbee2mqtt/Keuken_Spot3', payload '{"brightness":120,"color":{"x":0.447991567541816,"y":0.407559831563923},"color_temp":350,"linkquality":66,"state":"ON","update_available":false}'
```

This happens when I add a Scene with the same values and recall it:

```
{
    "scene_add": {
        "ID": 1,
        "state": "ON",
        "brightness": 120,
        "color_temp": 350
    }
}
```

Log:
```
debug 2020-11-12 18:58:25: Received MQTT message on 'zigbee2mqtt/Keuken_Spots_ct/set' with data '{
    "scene_add": {
        "ID": 1,
        "state": "ON",
        "brightness": 120,
        "color_temp":350
    }
}'
debug 2020-11-12 18:58:25: Publishing 'set' 'scene_add' to 'Keuken_Spots_ct'
info  2020-11-12 18:58:26: MQTT publish: topic 'zigbee2mqtt/Keuken_Spot1', payload '{"brightness":120,"color":{"x":0.447991567541816,"y":0.407559831563923},"color_temp":350,"linkquality":72,"state":"ON","update_available":false}'
info  2020-11-12 18:58:26: MQTT publish: topic 'zigbee2mqtt/Keuken_Spots_ct', payload '{"brightness":120,"color":{"x":0.447991567541816,"y":0.407559831563923},"color_temp":350,"state":"ON"}'
info  2020-11-12 18:58:26: MQTT publish: topic 'zigbee2mqtt/Keuken_Spot2', payload '{"brightness":120,"color":{"x":0.447991567541816,"y":0.407559831563923},"color_temp":350,"linkquality":63,"state":"ON","update_available":false}'
info  2020-11-12 18:58:26: MQTT publish: topic 'zigbee2mqtt/Keuken_Spot3', payload '{"brightness":120,"color":{"x":0.447991567541816,"y":0.407559831563923},"color_temp":350,"linkquality":72,"state":"ON"," [... cut ...]
info  2020-11-12 18:58:50: MQTT publish: topic 'zigbee2mqtt/Keuken_Spot2', payload '{"brightness":120,"color":{"x":0,"y":0},"color_temp":350,"linkquality":75,"state":"ON","update_available":false}'
```
As you can see in the log the X and Y colors` {"x":0,"y":0}`, which causes a totally different color, although the color_temp is set to 350

As a work around I use the color attribute instead of the color_temp when adding a Scene.
And use the values `{"x":0.447991567541816,"y":0.407559831563923}` of the info  message after I send the command
`{ "state": "ON", "brightness": 120, "color_temp": 350 }`

Why is the value of  color attribute, when recallling a Scene, set to {"x":0,"y":0} ?

Koenkk, 2020-11-14:
Can you check if it works via `scene_store` instead of `scene_add`? https://www.zigbee2mqtt.io/information/scenes.html#scene_store

Anjerlaan, 2020-11-16:
> Can you check if it works via `scene_store` instead of `scene_add`? https://www.zigbee2mqtt.io/information/scenes.html#scene_store

With scene_store the recall is working correct, color values are equal to the normal set state command:
`{ "state": "ON", "brightness": 120, "color_temp": 350 }`
Only note is that the X color value is rounded to 3 digits

```
debug 2020-11-16 21:19:49: Received MQTT message on 'zigbee2mqtt/Keuken_Spots_ct/set' with data '{"scene_recall": 1}'
debug 2020-11-16 21:19:49: Publishing 'set' 'scene_recall' to 'Keuken_Spots_ct'
debug 2020-11-16 21:19:49: Received Zigbee message from 'Keuken_Spot3', type 'attributeReport', cluster 'lightingColorCtrl', data '{"currentX":29359}' from endpoint 1 with groupID 0
info  2020-11-16 21:19:49: MQTT publish: topic 'zigbee2mqtt/Keuken_Spots_ct', payload '{"brightness":120,"color":{"x":0.448,"y":0.407559831563923},"color_temp":350,"state":"OFF"}'
info  2020-11-16 21:19:49: MQTT publish: topic 'zigbee2mqtt/Keuken_Spot3', payload '{"brightness":120,"color":{"x":0.448,"y":0.407559831563923},"color_temp":350,"linkquality":63,"state":"OFF","update":{"state":"idle"},"update_available":false}'
debug 2020-11-16 21:19:49: Received Zigbee message from 'Keuken_Spot2', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2020-11-16 21:19:49: MQTT publish: topic 'zigbee2mqtt/Keuken_Spots_ct', payload '{"brightness":120,"color":{"x":0.447991567541816,"y":0.407559831563923},"color_temp":350,"state":"ON"}'
info  2020-11-16 21:19:49: MQTT publish: topic 'zigbee2mqtt/Keuken_Spot2', payload '{"brightness":120,"color":{"x":0.447991567541816,"y":0.407559831563923},"color_temp":350,"linkquality":66,"state":"ON","update":{"state":"idle"},"update_available":false}'
debug 2020-11-16 21:19:49: Received Zigbee message from 'Keuken_Spot2', type 'attributeReport', cluster 'lightingColorCtrl', data '{"currentX":29359}' from endpoint 1 with groupID 0
info  2020-11-16 21:19:49: MQTT publish: topic 'zigbee2mqtt/Keuken_Spots_ct', payload '{"brightness":120,"color":{"x":0.448,"y":0.407559831563923},"color_temp":350,"state":"ON"}'
info  2020-11-16 21:19:49: MQTT publish: topic 'zigbee2mqtt/Keuken_Spot2', payload '{"brightness":120,"color":{"x":0.448,"y":0.407559831563923},"color_temp":350,"linkquality":66,"state":"ON","update":{"state":"idle"},"update_available":false}'
debug 2020-11-16 21:19:49: Received Zigbee message from 'Keuken_Spot1', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2020-11-16 21:19:49: MQTT publish: topic 'zigbee2mqtt/Keuken_Spots_ct', payload '{"brightness":120,"color":{"x":0.447991567541816,"y":0.407559831563923},"color_temp":350,"state":"ON"}'
info  2020-11-16 21:19:49: MQTT publish: topic 'zigbee2mqtt/Keuken_Spot1', payload '{"brightness":120,"color":{"x":0.447991567541816,"y":0.407559831563923},"color_temp":350,"linkquality":63,"state":"ON","update":{"state":"idle"},"update_available":false}'
debug 2020-11-16 21:19:49: Received Zigbee message from 'Keuken_Spot1', type 'attributeReport', cluster 'lightingColorCtrl', data '{"currentX":29359}' from endpoint 1 with groupID 0
info  2020-11-16 21:19:49: MQTT publish: topic 'zigbee2mqtt/Keuken_Spots_ct', payload '{"brightness":120,"color":{"x":0.448,"y":0.407559831563923},"color_temp":350,"state":"ON"}'
info  2020-11-16 21:19:49: MQTT publish: topic 'zigbee2mqtt/Keuken_Spot1', payload '{"brightness":120,"color":{"x":0.448,"y":0.407559831563923},"color_temp":350,"linkquality":60,"state":"ON","update":{"state":"idle"},"update_available":false}'
debug 2020-11-16 21:19:49: Received Zigbee message from 'Keuken_Spot3', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
info  2020-11-16 21:19:49: MQTT publish: topic 'zigbee2mqtt/Keuken_Spots_ct', payload '{"brightness":120,"color":{"x":0.448,"y":0.407559831563923},"color_temp":350,"state":"ON"}'
info  2020-11-16 21:19:49: MQTT publish: topic 'zigbee2mqtt/Keuken_Spot3', payload '{"brightness":120,"color":{"x":0.448,"y":0.407559831563923},"color_temp":350,"linkquality":63,"state":"ON","update":{"state":"idle"},"update_available":false}'
debug 2020-11-16 21:19:49: Received Zigbee message from 'Keuken_Spot3', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":9}' from endpoint 1 with groupID 0
info  2020-11-16 21:19:49: MQTT publish: topic 'zigbee2mqtt/Keuken_Spots_ct', payload '{"brightness":9,"color":{"x":0.448,"y":0.407559831563923},"color_temp":350,"state":"ON"}'
info  2020-11-16 21:19:49: MQTT publish: topic 'zigbee2mqtt/Keuken_Spot3', payload '{"brightness":9,"color":{"x":0.448,"y":0.407559831563923},"color_temp":350,"linkquality":63,"state":"ON","update":{"state":"idle"},"update_available":false}'
debug 2020-11-16 21:19:49: Received Zigbee message from 'Keuken_Spot1', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":8}' from endpoint 1 with groupID 0
info  2020-11-16 21:19:49: MQTT publish: topic 'zigbee2mqtt/Keuken_Spots_ct', payload '{"brightness":8,"color":{"x":0.448,"y":0.407559831563923},"color_temp":350,"state":"ON"}'
info  2020-11-16 21:19:49: MQTT publish: topic 'zigbee2mqtt/Keuken_Spot1', payload '{"brightness":8,"color":{"x":0.448,"y":0.407559831563923},"color_temp":350,"linkquality":63,"state":"ON","update":{"state":"idle"},"update_available":false}'
[... cut ...]
info  2020-11-16 21:19:49: MQTT publish: topic 'zigbee2mqtt/Keuken_Spot2', payload '{"brightness":9,"color":{"x":0.448,"y":0.407559831563923},"color_temp":350,"linkquality":66,"state":"ON","update":{"state":"idle"},"update_available":false}'
```

Other thing I notice in the log is that:

- Brightness value is 120 (as it should), but there is also an attribute "Brightness" value of 8 or 9 at the last 6 rows of the log?
What is the difference?

- And the value of the data '{"currentLevel":X}' of the 3 bulbs in the group are not equal for all bulbs. 
data '{"currentLevel":8}' or data '{"currentLevel":9}', I thought that a group state is send all members set to the same state?

Koenkk, 2020-11-18:
The incorrect brightness occurs because the spots report this brightness themselves: `debug 2020-11-16 21:19:49: Received Zigbee message from 'Keuken_Spot1', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":8}' from endpoint 1 with groupID 0`

If the actual brightness is not 8 at this moment, this seems to be a bug inside the firmwares of the bulbs.

Anjerlaan, 2020-11-21:
So you say that a bug in the ikea bulbs firmware is causing the strange values with the scene_add, scene_store and recall functions? I can’t test it with other bulbs, because these are the only 3 with color temp I have.

I will continue to use the color attribute then instead of color_temp.

sjorge, 2020-11-26:
I can confirm that scene_add with color_temp is **not** working with Ikea trådfri GU10 WS and FLOAT panels.
Using color does work, which is odd as they don't seem to accept any RGB color values.

sjorge, 2020-11-26:
Some more testing... (with a Tradfri LED1537R6/LED1739R5 I had to spare)

```
mosquitto_pub -t zigbee2mqtt/0xec1bbdfffeb5154f/set/scene_add -m '{"ID": 22, "state": "OFF", "color_temp": "250"}'
mosquitto_pub -t zigbee2mqtt/0xec1bbdfffeb5154f/set/scene_recall -m 22
sleep 2
mosquitto_pub -t zigbee2mqtt/0xec1bbdfffeb5154f/set/state -m ON
sleep 2
mosquitto_pub -t zigbee2mqtt/0xec1bbdfffeb5154f/set/state -m OFF
sleep 2
mosquitto_pub -t zigbee2mqtt/0xec1bbdfffeb5154f/set/scene_add -m '{"ID": 23, "state": "OFF", "color_temp": "455"}'
mosquitto_pub -t zigbee2mqtt/0xec1bbdfffeb5154f/set/scene_recall -m 23
sleep 2
mosquitto_pub -t zigbee2mqtt/0xec1bbdfffeb5154f/set/state -m ON
```

This doesn't seem to work at all, while doing this using 'color' and a RGB value for a tradfri CWS bulb works fine...
De results in from the database.db

```json
"scenes": {
          "22_0": {
            "state": {
              "state": "OFF",
              "color_temp": "250"
            }
          },
          "23_0": {
            "state": {
              "state": "OFF",
              "color_temp": "455"
            }
          }
        }
```

This looks fine to be, I also tried with state ON, but that makes no difference.
The recall doesn't recall the color_temp (but it does recall the state!)

Lets try with scene_store!

```
mosquitto_pub -t zigbee2mqtt/0xec1bbdfffeb5154f/set -m '{"state": "ON", "color_temp": "250"}'
sleep 2
mosquitto_pub -t zigbee2mqtt/0xec1bbdfffeb5154f/set/state -m OFF
sleep 1
mosquitto_pub -t zigbee2mqtt/0xec1bbdfffeb5154f/set/scene_store -m 30
sleep 5
mosquitto_pub -t zigbee2mqtt/0xec1bbdfffeb5154f/set -m '{"state": "ON", "color_temp": "455"}'
sleep 2
mosquitto_pub -t zigbee2mqtt/0xec1bbdfffeb5154f/set/state -m OFF
sleep 1
mosquitto_pub -t zigbee2mqtt/0xec1bbdfffeb5154f/set/scene_store -m 31
```

```
        "scenes": {
          "22_0": {
            "state": {
              "state": "ON",
              "color_temp": "250"
            }
          },
          "23_0": {
            "state": {
              "state": "ON",
              "color_temp": "455"
            }
          },
          "30_0": {
            "state": {
              "state": "OFF",
              "brightness": 254,
              "color_temp": 250,
              "color": {
                "x": 0,
                "y": 0
              }
            }
          },
          "31_0": {
            "state": {
              "state": "OFF",
              "brightness": 254,
              "color_temp": 455,
              "color": {
                "x": 0,
                "y": 0
              }
            }
          }
        }
      }
```

Same results with as above when recalling (same done with state left to ON).
Having the color in there as 0,0 is odd, not sure if this an artifact of how z2m stores the scenes metadata or an actual state saved to the bulb.

sjorge, 2020-11-26:
Wanted to try a Hue CWS bulb to see if it supported color_temp (not sure) but I couldn't reset it :( https://github.com/Koenkk/zigbee2mqtt/issues/4968#issuecomment-734461425

So no more testing from me for now, I had a look at the scene_add code and it looks OK to me, we're writing the the correct field I think... not sure we should also specify the color_mode to 0x2 though... maybe we need to?

Anjerlaan, 2020-11-26:
> 
> 
> @Anjerlaan what bulbs are you using?

@sjorge 
3 x IKEA LED1537R6/LED1739R5

MattWestb, 2020-11-26:
IKEA is adding scenes with X and Y for their WS bulbs as shown from sniffing the GW:

```ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x11)
    Sequence Number: 37
    Command: Add Scene (0x00)
    Payload, String: 
        Group ID: 0x002a
        Scene ID: 0x00
        Transition Time: 1 seconds
        Length: 0
        String: 
        Extension field set 1
            Cluster: On/Off (0x0006)
            On/Off: 1
        Extension field set 2
            Cluster: Color Control (0x0300)
            Color X: 0,4580
            Color Y: 0,4100
        Extension field set 3
            Cluster: Level Control (0x0008)
            Level: 203
        Extension field set 4
            Cluster: Unknown (0x0102)
            Current Position Lift Percentage: 0
            Current Position Tilt Percentage: 0

```
And after it reading it for verifying the setting is made:

```
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x19)
    Sequence Number: 41
    Command: View Scene Response (0x01)
    Payload, String: 
        Scenes Status: Success (0x00)
        Group ID: 0x002a
        Scene ID: 0x00
        Transition Time: 1 seconds
        Length: 0
        String: 
        Extension field set 1
            Cluster: On/Off (0x0006)
            On/Off: 1
        Extension field set 2
            Cluster: Level Control (0x0008)
            Level: 203
        Extension field set 3
            Cluster: Color Control (0x0300)
            Color X: 0,4580
            Color Y: 0,4100
```

PS: The scenes is special initiated with manufacturer commands so only the color is changed then getting command from the 5 button remote and skipping the on/off, light level and blinds positions.
The example is for the first of the 3 "default" scene that is added in all IKEA bulbs for working with the 5 button remote and is identical with those that is created after firmware update of the bulbs.

sjorge, 2020-11-27:
Sending color seems to work for Trådfri and my CWS bulbs, I sadly don't have WS only bulbs from Hue or other brands to try with.

I still think something is broken, because the ZCL says we can set color_temp via scenes... i can take say Trådfri botching there firmware so there remotes can just send xy... but it would seem odd all brands would do this.

I'll try to setup my sniffer tomorrow and see what z2m is sending out for the scene_add, perhaps we're encoding something wrong.

MattWestb, 2020-11-27:
I think the IKEA WS scenes must being setted up / added with xy and not CT that its little of standard.
The remote is sending "manufacture patched" commands next and previous scene, therefore the scenes is transferring / setting up from controlling device to the lights then pairing (Lighting link networks dont have any coordinator) or added more of them later from the GW if having one.

The IKEA Zigbee 3 lights scenes is pretty good and standard then not using their GW and is certified, some things is added that is out of standard (true manufacture cluster) but that is OK in zigbee but the setting of colour is odd.

PS: Dont forgetting go to Lidl then there "Home Smart Home" is coming for getting little more zigbee toys like zigbee christmas tree lights and short RGBWW LED stripes ;-)

sjorge, 2020-11-27:
Some annoying side effects, the IKEA WS bulbs don't seem to update color_temp when they change color via a scene with color_xy...

Also looks like they only take a very specific set of colors, I figured them out by using the remote... then recording both color_temp and color_xy

The remotes are definitely doing odd stuff when they set the scene as the bulb seems to report color_temp afterwards.

MattWestb, 2020-11-27:
If you is adding groupe 0xff09 to the bulb and then adding new scenes to it you can using the remote for your scenes. If the remote dont get any groupe thru touchlink the pairing its sending the scene next and previous to groupe 0xff09.

sjorge, 2020-11-28:
I got a pcap here for the scene_add/recall with color_temp included: https://pkg.blackdot.be/cores/zigbee/scene_add_recall.pcapng

@Koenkk same netkey as usual.

sjorge, 2020-11-28:
[image]

I think we're not correctly adding the color_temp, as it is not present in the scene add payload, it does however list a blank x, y, hue, and saturation

sjorge, 2020-11-28:
I wonder if scenes have slightly modified clusters spec? As Color Loop * is not listed for cluster 0x0300 (color control) in the ZCL doc I have?

MattWestb, 2020-11-28:
Tyre save one scene and then veuve the scene and you can see how is stored.

MattWestb, 2020-11-28:
Is the scene stored / added with over off ??
That can being one logical problem I think.

sjorge, 2020-11-28:
There is no scene view support though, I also tried with scene_store and it doesn't recall color_temperature either.
I'm starting to suspect it might be impossible to do so via scenes.

This one was indeed with state off, but as mentioned above, on or off makes no difference. It works fine (state off) with color for CWS bulbs though (so you can change the color while they are off!, tested on trådfri, hue, innr, muller, and osram!)

MattWestb, 2020-11-28:
The extension for lights in zigbee 3:

[docs-15-0014-05-0plo-Lighting-OccupancyDevice-Specification-V1.0.pdf](https://github.com/Koenkk/zigbee2mqtt/files/5610805/docs-15-0014-05-0plo-Lighting-OccupancyDevice-Specification-V1.0.pdf)

Missing command 0x01 View scene is mandatory so it cant being moddes in the light.
Is the Z2M that not have implementing it ?

sjorge, 2020-11-28:
Yeah, It's z2m currently that does not implete the view/delete/delete_all

MattWestb, 2020-11-28:
Then you must doing some nice calculatings and sending the command raw and hoping its being right ;-))
ZHA dont have implementing the high level deCONZ have but not for the individual parameters that you can see from the wireshark sniffs.
Do you have some other commercial GW for testing ? 
I using my IKEA GW after sniffing its network key.

sjorge, 2020-11-28:
Nope, no gateways for testing.

z2m is handling the scene_add here, https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/toZigbee.js#L3699, but... currently i'm working under the assumption that:
- z2m is sending the wrong payload somehow
- scenes don't support color_temp

Still not sure which.

MattWestb, 2020-11-28:
IKEA GW / remote adding one scene to one WS bulb:

```ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x11)
    Sequence Number: 37
    Command: Add Scene (0x00)
    Payload, String: 
        Group ID: 0x002a
        Scene ID: 0x00
        Transition Time: 1 seconds
        Length: 0
        String: 
        Extension field set 1
            Cluster: On/Off (0x0006)
            On/Off: 1
        Extension field set 2
            Cluster: Color Control (0x0300)
            Color X: 0,4580
            Color Y: 0,4100
        Extension field set 3
            Cluster: Level Control (0x0008)
            Level: 203
        Extension field set 4
            Cluster: Unknown (0x0102)
            Current Position Lift Percentage: 0
            Current Position Tilt Percentage: 0

sjorge, 2020-11-28:
Yeah, setting XY works fine via scene_add.
I also figured out which XY values the IKEA WS supports:
```
    var color_payloads = {
        "cool": {
            "color": {"x":0.3818,"y":0.3797},
        },
        "neutral":{
            "color": {"x":0.458,"y":0.41},
        },
        "warm":{
            "color": {"x":0.5018,"y":0.4153},
        },
    };
```

I sadly have no non IKEA WS bulbs, all the others I own are RGB.

But if scenes with color_temp are not supported, we need to update scene_add to complain when passing color_temp. If they are supported, we need to look into fixing the payload.

sjorge, 2020-11-28:
@Koenkk how is the order of the extFields set determined? the listed order doesn't seem to match whats in "
Zigbee Cluster Library Specification v7.pdf" page 357 ?

MattWestb, 2020-11-28:
Its gets one thread with much dumps of setting up scenes on ikes CWS its like somthing "2 tyre getting the 5 button remote to work with scenes"

Koenkk, 2020-11-29:
@sjorge the order should match the one in the ZCL spec (so probably that is why it's not working)

sjorge, 2020-11-29:
> @sjorge the order should match the one in the ZCL spec (so probably that is why it's not working)

Right, so the color_temp payload we pack is probably wrong... 

I tried poking around with it but the slot where 'val' is doesn't even show up in the wireshark capture. If I shift it a slot forward it comes in the other listed attributes, but some give a malformed payload, probably because the value is out of range. Also not sure we should set the other fields to `0` ?

Koenkk, 2020-11-30:
So I found out what is happening here.

Started with wireshark and it looks that we include the color temperature values correctly as they can be seen in the payload , its a 2 byte number, see red circles, however they indeed do not show up in the "friendly name" list.

[image]

Then I found out that the colorTemperatureMireds extension field is something new in ZCL V7: (left V6, right V7):

[image]

This is (very likely) the reason why wireshark does not show it, it simply has not been updated with V7. Therefore many bulbs probably also don't support this. So in case your bulb doesn't you can do the following:
- For bulbs that support color, just set the color to the desired color temperature
- For bulbs that do not support color (e.g. TRADFRI GU10 spot), I've figured out you can set the color temperature via the color.
  - Fully yellow: {"scene_add": {"ID": 1,  "color": "#000000"}}
  - Fully white: {"scene_add": {"ID": 1,  "color": "#FFFFFF"}}

MattWestb, 2020-11-30:
`8657 Since there is a direct relation between ColorTemperatureMireds and XY, color temperature, if supported, is stored as XY in the scenes table.`

Way not setting the CT with XY and not the Mireds is stored in the scene table and dont need to knowing if the light is ZCL v7 compatible or not ?

sjorge, 2020-11-30:
So far from testing, I looks like none of:
- Hue
- Trådfri
- Mueller
- OSRAM
- Innr

Support it, that’s all I had for testing.
For trådfri there is a set of colors they take for there own scenes... I think I posted them above too.

~ sjorge

MattWestb, 2020-11-30:
IKEA GW can adding more then scene 0 to 2 with X Y.

```
ZigBee Cluster Library Frame
    Frame Control Field: Cluster-specific (0x11)
    Sequence Number: 66
    Command: Add Scene (0x00)
    Payload, String: 
        Group ID: 0x002a
        Scene ID: 0x03
        Transition Time: 1 seconds
        Length: 0
        String: 
        Extension field set 1
            Cluster: On/Off (0x0006)
            On/Off: 1
        Extension field set 2
            Cluster: Color Control (0x0300)
            Color X: 0,3790
            Color Y: 0,3753
        Extension field set 3
            Cluster: Level Control (0x0008)
            Level: 132
        Extension field set 4
            Cluster: Unknown (0x0102)
            Current Position Lift Percentage: 0
            Current Position Tilt Percentage: 0
```

It was called moods then i was doing the sniff but for some days its changed to scenes.

In deCONZ i have saved (not added) 6 scenes but its not the same.
[image]

MattWestb, 2020-11-30:
With the IKEA WS have smaller range then the maximum in ZCL but the range you is having in the warm and cold scenes. If I remember right its possible reading the attribute wat capacity one device is having. If trying going outside its range its not saving the scene.

sjorge, 2020-11-30:
@Koenkk we have code somewhere that maps Mirad to XY in z2m right?

I'm not sure how there is an easy way to do the fallback, but I guess we could always do it until we find bulbs that actually support it?

sjorge, 2020-12-01:
A quick dirty test shows,... that the lights (IKEA Trådfri) do seem to report colorTemperature correctly.

```
Zigbee2MQTT:debug 2020-12-01 14:45:07: Received MQTT message on 'zigbee2mqtt/office/lights/set' with data '{"scene_add": {"ID": 22, "state": "ON", "color": {"x":0.5018,"y":0.4153}}, "scene_recall": 22}'
Zigbee2MQTT:debug 2020-12-01 14:45:07: Publishing 'set' 'scene_add' to 'office/lights'
Zigbee2MQTT:info  2020-12-01 14:45:07: MQTT publish: topic 'zigbee2mqtt/office/lights/panel1', payload '{"brightness":64,"color_temp":250,"linkquality":60,"state":"ON","update":{"state":"idle"}}'
Zigbee2MQTT:info  2020-12-01 14:45:07: MQTT publish: topic 'zigbee2mqtt/office/lights', payload '{"brightness":64,"color":{"x":0.3818,"y":0.3797},"color_temp":250,"state":"ON"}'
Zigbee2MQTT:debug 2020-12-01 14:45:07: Publishing 'set' 'scene_recall' to 'office/lights'
Zigbee2MQTT:info  2020-12-01 14:45:07: MQTT publish: topic 'zigbee2mqtt/office/lights/panel2', payload '{"brightness":64,"color_temp":250,"linkquality":94,"state":"ON","update":{"state":"idle"}}'
Zigbee2MQTT:info  2020-12-01 14:45:07: MQTT publish: topic 'zigbee2mqtt/office/lights/panel3', payload '{"brightness":64,"color_temp":250,"linkquality":94,"state":"ON","update":{"state":"idle"}}'
Zigbee2MQTT:info  2020-12-01 14:45:07: MQTT publish: topic 'zigbee2mqtt/office/lights/panel4', payload '{"brightness":64,"color_temp":250,"linkquality":94,"state":"ON","update":{"state":"idle"}}'
Zigbee2MQTT:debug 2020-12-01 14:45:07: Received Zigbee message from 'office/lights/panel3', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":454}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-12-01 14:45:07: Received Zigbee message from 'office/lights/panel4', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":454}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-12-01 14:45:07: Received Zigbee message from 'office/lights/panel1', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":454}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-12-01 14:45:07: MQTT publish: topic 'zigbee2mqtt/office/lights', payload '{"brightness":64,"color":{"x":0.5018,"y":0.4153},"color_temp":250,"state":"ON"}'
Zigbee2MQTT:info  2020-12-01 14:45:07: MQTT publish: topic 'zigbee2mqtt/office/lights/panel1', payload '{"brightness":64,"color_temp":250,"linkquality":60,"state":"ON","update":{"state":"idle"}}'
Zigbee2MQTT:info  2020-12-01 14:45:07: MQTT publish: topic 'zigbee2mqtt/office/lights/panel2', payload '{"brightness":64,"color_temp":250,"linkquality":97,"state":"ON","update":{"state":"idle"}}'
Zigbee2MQTT:info  2020-12-01 14:45:07: MQTT publish: topic 'zigbee2mqtt/office/lights/panel3', payload '{"brightness":64,"color_temp":250,"linkquality":94,"state":"ON","update":{"state":"idle"}}'
Zigbee2MQTT:info  2020-12-01 14:45:07: MQTT publish: topic 'zigbee2mqtt/office/lights/panel4', payload '{"brightness":64,"color_temp":250,"linkquality":94,"state":"ON","update":{"state":"idle"}}'
Zigbee2MQTT:debug 2020-12-01 14:45:07: Received Zigbee message from 'office/lights/panel2', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":454}' from endpoint 1 with groupID 0
```

```
Zigbee2MQTT:debug 2020-12-01 14:46:52: Received MQTT message on 'zigbee2mqtt/office/lights/set' with data '{"scene_add": {"ID": 22, "state": "ON", "color": {"x":0.3818,"y":0.3797}}, "scene_recall": 22}'
Zigbee2MQTT:debug 2020-12-01 14:46:52: Publishing 'set' 'scene_add' to 'office/lights'
Zigbee2MQTT:info  2020-12-01 14:46:52: MQTT publish: topic 'zigbee2mqtt/office/lights/panel1', payload '{"brightness":64,"color_temp":454,"linkquality":42,"state":"ON","update":{"state":"idle"}}'
Zigbee2MQTT:info  2020-12-01 14:46:52: MQTT publish: topic 'zigbee2mqtt/office/lights', payload '{"brightness":64,"color":{"x":0.5018,"y":0.4153},"color_temp":454,"state":"ON"}'
Zigbee2MQTT:debug 2020-12-01 14:46:52: Publishing 'set' 'scene_recall' to 'office/lights'
Zigbee2MQTT:info  2020-12-01 14:46:52: MQTT publish: topic 'zigbee2mqtt/office/lights/panel2', payload '{"brightness":64,"color_temp":454,"linkquality":86,"state":"ON","update":{"state":"idle"}}'
Zigbee2MQTT:info  2020-12-01 14:46:52: MQTT publish: topic 'zigbee2mqtt/office/lights/panel3', payload '{"brightness":64,"color_temp":454,"linkquality":76,"state":"ON","update":{"state":"idle"}}'
Zigbee2MQTT:info  2020-12-01 14:46:52: MQTT publish: topic 'zigbee2mqtt/office/lights/panel4', payload '{"brightness":64,"color_temp":454,"linkquality":78,"state":"ON","update":{"state":"idle"}}'
Zigbee2MQTT:debug 2020-12-01 14:46:52: Received Zigbee message from 'office/lights/panel4', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":250}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-12-01 14:46:52: Received Zigbee message from 'office/lights/panel2', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":250}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-12-01 14:46:52: Received Zigbee message from 'office/lights/panel3', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":250}' from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2020-12-01 14:46:52: Received Zigbee message from 'office/lights/panel1', type 'attributeReport', cluster 'lightingColorCtrl', data '{"colorTemperature":250}' from endpoint 1 with groupID 0
[... cut ...]
Zigbee2MQTT:info  2020-12-01 14:46:57: MQTT publish: topic 'zigbee2mqtt/office/lights/panel4', payload '{"brightness":64,"color_temp":370,"linkquality":78,"state":"ON","update":{"state":"idle"}}'
Zigbee2MQTT:info  2020-12-01 14:46:57: MQTT publish: topic 'zigbee2mqtt/office/lights/panel2', payload '{"brightness":64,"color_temp":370,"linkquality":86,"state":"ON","update":{"state":"idle"}}'
Zigbee2MQTT:info  2020-12-01 14:46:57: MQTT publish: topic 'zigbee2mqtt/office/lights/panel1', payload '{"brightness":64,"color_temp":370,"linkquality":44,"state":"ON","update":{"state":"idle"}}'
```

Koenkk, 2020-12-01:
Looks good, can you make a PR? We  can work further from there.

sjorge, 2020-12-01:
Sure, not sure I will have time before next week. But I should be able to writing something up.

I’d love to do more testing but all test bulbs from other brands are CWS.

~ sjorge

Koenkk, 2020-12-03:
Should be fixed.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

sjorge, 2020-12-07:
Some interesting side effects after using this for a few days.

- trådfri WS bulbs immediately report currentX and currentY when off or on.
- trådfri WS bulbs only report colorTempersture when ON
- trådfri WS report wrong colorTempersture at a later point when OFF 
- trådfri WS with 1.x firmware report correct colorTemperature once turned on
- trådfri WS with 2.x firmware keeps reporting wrong colorTemperature after turning on

In all cases the bulb did change to the right color temp

- innr CWS doesn’t report currentX, currentY, and colorTempersture when ON but usually after a minute or 3
- innr CWS returns out of range colorTempersture values when OFF, but usually OK the next report, so it takes 2 att reports to be correct

It I have a bit more time I’ll see what these return with a get. I’d assume the same as the report but you never know.

(Only my FLOALT planets are still on a 1.x firmware as they lack updates)

MattWestb, 2020-12-08:
The GU10 WS 2.0 have those min and max attributes for color temperature:
```
Cluster Attributes
View and edit cluster attributes.
 
Attributes of the selected cluster
color_temp_physical_max (id: 0x400c)
 
Value
454
 

Cluster Attributes
View and edit cluster attributes.
 
Attributes of the selected cluster
color_temp_physical_min (id: 0x400b)
 
Value
250
```
Its reporting Color_temperature (0x0007), Current_X (0x0003) and Current_Y (0x0004).
Both then on and off but i dont knowing if the readings it in range or not.

My old E27 CWS (ZLL) is reporting the same for colour temperature min and max.
Current_X (0x0003) and Current_Y (0x0004) is reporting numbers then reading attributes then is on and off but i dont knowing if they is in range or not. Its not reporting Color_temperature (0x0007) but i think its in color mode (no TC in HA).

Cluster attributes is red thru ZHA.

sjorge, 2020-12-08:
I don't think currently the min/max range are even used without scene either, maybe they should though... that might prevent out of range values being set.

cc @Koenkk ?

I've not had time to do more testing with /get

MattWestb, 2020-12-08:
I think its better taking it in the calculating then its possible getting it from the device.
The CWS is RGB + W inside with simulated TC ?
[From FCC-ID](https://fccid.io/FHO-LED1624G9/Internal-Photos/Internal-Photos-3631436)

Koenkk, 2020-12-08:
The min/max color temperature is not used yet indeed.

sjorge, 2020-12-08:
> The min/max color temperature is not used yet indeed.

Should I file a feature request for that against zhc?

sjorge, 2021-01-01:
Been playing around with colorMode and xy vs hs a bit today, looks like for scenes we only write xy for now?

https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/toZigbee.js#L3542

[image]
Looks like field 3 and 4 can be set in that case, I'll add that to the todo list... which is rather long at this point haha.

All my bulbs are Innr and a Tint bulb, so I believe I should be able to test the effect on both kind of bulbs.

MattWestb, 2021-01-02:
Its looks that all manufactures is doing little home "cooking" in there devices and need little attention  for working well. The latest is the new LIDL WS / CT that is reporting that they is doing TC + color but is only color (perhaps the zigbee module is doing both but dont have the RGB LEDs implanted for saving time and money in development and production).

Goooood digging done !!

closed 2021-02-10 by github-actions[bot]

hildebrau, 2025-11-17:
@sjorge @MattWestb did this issue ever get officially solved?  I think I'm hitting similar things with some E11-N1EA bulbs from Sengled.  I have those and E21-N1EA Sengled bulbs which don't seem to exhibit the same Zigbee group scene recall issue.  

I'm running the latest z2m tag as of 12 hours ago.

You guys are way more in depth than I am so far, but I'm happy to try to help debug things.

When I store a scene in Zigbee2mqtt with the lights set at a dim blue set by HA, and then I restore the scene, the e11 bulb turns to a dim white.  

If I touch the exposes page on that bulb and just put my cursor in the color temp box and click out of it, then z2m sends a new signal to it and the bulb will turn to a light peach color instead and the x,y color definition will change.  Then I can set any of the color settings back to what it was and it goes dim blue.

If I then store the current fixed group settings into a scene and recall it, that bulb goes dim white again.

How can I see the saved settings of a group stored scene in Zigbee2mqtt?

So, this feels like a color temp related bug for scene restores in z2m and that's what led me to find this issue.

Thanks!