Koenkk/zigbee2mqtt #4927: Ledvance Led Stripe not detected / paired (same as OSRAM SMart+)

bsoufflet, 2020-11-11:
## What happened
I have a brand new Ledvance Smart+ flex 3p multicolor (zigbee version) and I am not able to pair it with zigbee2mqtt.

My installation :
Raspberry Pi 3 with zigbee2 mqtt installed using the default procedure. the CC2531 is connected to the RPI with a 50cm USB extension cable.

- My zigbee2mqtt is freshly installed with the latest version and with 1 aqara device already paired.
- No logging is shown at all regarding the Ledvance device.  (I tried this with no success)
- I tried to factory reset the device multiple times with no success (5sec ON - 5sec OFF 5 times). The strip quickly change colors and stay white on startup.

## What did you expect to happen
- I am expecting to see the pairing log in the zigbee2mqtt log but no logging is shown at all.

## Debug info
Log : https://pastebin.com/LKXYKT2j
Zigbee2MQTT version: 1.16.1 (commit #6b32f30)
Adapter hardware: CC2531
Adapter firmware version: 20190608

bsoufflet, 2020-11-15:
I just fix my problem by trying multiple channels. The channel 25 seems to work in my case and I was able to pair the device. Why is this not part of the documentation when pairing is not possible ? Thanks

Koenkk, 2020-11-16:
You can update the documentation by clicking the pencil icon here: https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/4058075208339.md

bsoufflet, 2020-11-16:
I think it is something which should be in the global  troubleshooting article when pairing is failing with nothing on the log of zigbee2mqtt no ?

closed 2020-11-16 by bsoufflet