Koenkk/zigbee2mqtt #4933: Tuya ZigBee 3.0 Smart Thermostat Heater TRV

jeviilladsen, 2020-11-11:
I have had some tiny level of success in reading the values, but i unfortunately have not the skills required to get it to work completely and in Homeassistant.

Hope somebody can help.

## Information about the device + link
Tuya ZigBee 3.0 Smart Thermostat Heater TRV
[interview__TZE200_fhn3negr.txt](https://github.com/Koenkk/zigbee2mqtt/files/5522957/interview__TZE200_fhn3negr.txt)

https://www.aliexpress.com/item/1005001665148698.html?spm=a2g0s.8937460.0.0.15c92e0e2yGK1I

## data/database.db entry of the device
{"id":3,"type":"EndDevice","ieeeAddr":"0xec1bbdfffe7827d3","nwkAddr":22399,"manufId":4098,"manufName":"_TZE200_fhn3negr","powerSource":"Battery","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,10,4,5,61184],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"!�='\u0012\"�='\u0012","65506":27,"modelId":"TS0601","manufacturerName":"_TZE200_fhn3negr","powerSource":3,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1605079371711}

insipiens, 2020-11-11:
Could be the same as _TZE200_ckud7u2l in the devices.js file - line 1787 in Dev branch.

If you change that entry to _TZE200_fhn3negr and see if it works (assuming you don't have other thermostats that the change will break, if so would need to make it an additional entry).

jeviilladsen, 2020-11-12:
I tried that, but if you look in my own converters file and compare it to the dp's in _TZE200_ckud7u2l you will see a great diff..
I have tried to let it act as _TZE200_ckud7u2l and that gave me weird readings.
[my_converters.js.txt](https://github.com/Koenkk/zigbee2mqtt/files/5527542/my_converters.js.txt)

insipiens, 2020-11-12:
yes, seems like different codes then.

The way Z2M is treating Tuya devices has changed so you might want to check out the new way of supporting these devices so a Pull request works. [https://www.zigbee2mqtt.io/how_tos/how_to_support_new_tuya_devices.html](url)

insipiens, 2020-11-12:
It is odd that it is also a Moes valve which their other type (HY368) works using different DP commands under the current Tuya TS0601 settings. 

However their room thermostat model used different DP codes for similar functions so there is evidence that they aren't consistent.

serrj-sv, 2020-11-24:
I user cc2531 sniffer to capture communication between Tuya Gateway and TZE200_fhn3negr.
This is what I managed to decode:
1. Data sent from Gateway to Device:
https://gist.github.com/serrj-sv/e6680647c438221b190a2b4d96805cc4
2. Data sent from Device to Gateway
will add soon

But I need some help to integrate this data into z2m

jeviilladsen, 2020-11-24:
Sounds very promising, unfortunately i do not know how to do this.. I'm hoping someone else can help. I really like this Thermostat.
@insipiens can you help ?

insipiens, 2020-11-24:
> 
> 
> I user cc2531 sniffer to capture communication between Tuya Gateway and TZE200_fhn3negr.
> This is what I managed to decode:
> 
>     1. Data sent from Gateway to Device [(zbee_aps.cluster == 0xef00) && (zbee_zcl.cs.cmd.id == 0x00)]:
>        https://gist.github.com/serrj-sv/e6680647c438221b190a2b4d96805cc4
> 
>     2. Data sent from Device to Gateway [(zbee_aps.cluster == 0xef00) && (zbee_zcl.cs.cmd.id == 0x02)]
>        will add soon
> 
> 
> But I need some help to integrate this data into z2m

Seems those already exist in Z2M 

For instance set current heating set point for the Moes room thermostat uses DP code 528 (0x0210) to specify the target temperature.

Check out this article and examine the existing set up of codes in common.js

https://www.zigbee2mqtt.io/how_tos/how_to_support_new_tuya_devices.html

serrj-sv, 2020-11-24:
Ok, I updated gists with all possible data sent to/from gateway/TRV

serrj-sv, 2020-11-24:
> For instance set current heating set point for the Moes room thermostat uses DP code 528 (0x0210) to specify the target temperature.

where I can see this in code? I checked  https://github.com/Koenkk/zigbee-herdsman-converters/blob/master/converters/common.js and there is no mention of "528". I can see only:

```moesHeatingSetpoint: 16,```

serrj-sv, 2020-11-26:
I created new device support request in zigbee-herdsman-converters, because it is more appropriate place: https://github.com/Koenkk/zigbee-herdsman-converters/issues/1803

closed 2021-01-03 by github-actions[bot]