Koenkk/zigbee2mqtt #4939: zigbee2mqtt 1.16.1 problem: TYZS1L is not able to change led strips color

rgolab, 2020-11-11:
## What happened
After upgrading `zigbee2mqtt` from `1.14.2` to the `1.16.1` my [TYZS1L](https://www.zigbee2mqtt.io/devices/TYZS1L.html) stopped changing my led strips color.
I'm able to change the state (ON/OFF) but I'm not able to change the color. In the `1.16.1` logs I can see:
```
Zigbee2MQTT:debug 2020-11-11 12:03:16: Received MQTT message on 'zigbee2mqtt/led_2/set' with data '{"state": "ON", "color": {"h": 225.0, "s": 100.0}}'
Zigbee2MQTT:debug 2020-11-11 12:03:16: Skipping state because of Home Assistant
Zigbee2MQTT:debug 2020-11-11 12:03:16: Publishing 'set' 'color' to 'led_2'
```
In the `1.14.2` version during color changing I can see in logs:
```
zigbee2mqtt:debug 2020-11-11 13:00:19: Received MQTT message on 'zigbee2mqtt/led_2/set' with data '{"state": "ON", "color": {"x": 0.508, "y": 0.239}}'
zigbee2mqtt:debug 2020-11-11 13:00:19: Skipping state because of Home Assistant
zigbee2mqtt:debug 2020-11-11 13:00:19: Publishing 'set' 'color' to 'led_2'
zigbee2mqtt:info  2020-11-11 13:00:20: MQTT publish: topic 'zigbee2mqtt/led_2', payload '{"linkquality":26,"state":"ON","color":{"x":0.508,"y":0.239},"color_temp":37,"brightness":254}'
```

## What did you expect to happen
When I upgrade `zigbe2mqtt` to the `latest` version my TYZS1L devices are able to change color of my led strips.

## How to reproduce it (minimal and precise)
Downgrade `zigbee2mqtt` to the `1.14.2` version.

## Debug info
Zigbee2MQTT version: 1.15.0 - 1.16.1
Adapter hardware: CC2531
Adapter firmware version: zStack12

Burninf, 2021-01-10:
+ same trouble

D3M80L, 2021-01-19:
Can confirm, that the color can't be changed, using 1.17.0.1

Koenkk, 2021-01-20:
Can you provide the data/database.db entry of your device?

Burninf, 2021-01-20:
{"id":12,"type":"Router","ieeeAddr":"0xbc33acfffefe26ce","nwkAddr":28351,"manufId":4098,"manufName":"_TZ3000_obacbukl","powerSource":"Mains (single phase)","modelId":"TS0503A","epList":[1,242],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0503A","manufacturerName":"_TZ3000_obacbukl","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}},"lightingColorCtrl":{"attributes":{"currentHue":0,"currentSaturation":254}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1611159774549}

D3M80L, 2021-01-20:
{"id":3,"type":"Router","ieeeAddr":"0x842e14fffe13f46a","nwkAddr":32797,"manufId":4098,"manufName":"_TZ3000_obacbukl","powerSource":"Mains (single phase)","modelId":"TS0503A","epList":[1,242,254],"endpoints":{"1":{"profId":260,"epId":1,"devId":258,"inClusterList":[0,3,4,5,6,8,768,4096],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0503A","manufacturerName":"_TZ3000_obacbukl","powerSource":1,"zclVersion":3,"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":""}},"genOnOff":{"attributes":{"onOff":1}},"lightingColorCtrl":{"attributes":{"currentHue":0,"currentSaturation":254}}},"binds":[],"configuredReportings":[],"meta":{}},"242":{"profId":41440,"epId":242,"devId":97,"inClusterList":[],"outClusterList":[33],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}},"254":{"epId":254,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[],"configuredReportings":[],"meta":{}}},"appVersion":70,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1611093517514}

Koenkk, 2021-01-20:
Can you check if this is fixed?

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

D3M80L, 2021-01-20:
Just checked that manually by sending a message (I put a comment on the fix you provided).
The device supports colors, and the fix you provided does not support 'color'.
Haven't checked all the possibilities, but for instance brightness works and also the RGB color command.
Hope to find more time to play more with the device soon.

Koenkk, 2021-01-21:
@D3M80L thanks, fixed

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

D3M80L, 2021-01-22:
Few more observations:
* I can't get pure colors on that device (i.e. red, green or blue) - I get only just kind of red with white, don't know if this is a limitation or a bug in the calculations
* no color picker is available in Home Assistant, the only entities available in HA are: link quality and on/off switch
[image]

Koenkk, 2021-01-23:
Good, removed the color temp stuff. Assuming this can be closed now.

closed 2021-01-23 by Koenkk