Koenkk/zigbee2mqtt #4942: last_seen and linkquality not stored in state.json and frontend LQI column N/A after refresh

Nonsense1112, 2020-11-11:
## What happened
Since later versions, the last_seen and linkquality properties are not stored in state.json file. Because of this, the LQI column in the frontend is showing N/A on all devices when the page is refreshed.

I've made some changes based on #2175 and I moved data from
https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/controller.js#L303-L327
to
https://github.com/Koenkk/zigbee2mqtt/blob/dev/lib/controller.js#L279
and modified `messagePayload.last_seen =` & `messagePayload.linkquality =` to `payload.last_seen =` & `payload.linkquality =`
Can this changes be permanent on next releases?

## What did you expect to happen
I want to store the linkquality propriety in state.json so that the frondend can pull the data and populate the LQI column on page refresh.

## How to reproduce it (minimal and precise)
Open the frontend, wait a bit so that the LQI column populate with some new data and after, refresh the page, the N/A values are shown.

## Debug info
Zigbee2MQTT version:
1.16.1
Adapter hardware: zzh
Adapter firmware version: 20201026

Koenkk, 2020-11-12:
I cannot reproduce this on 1.16.1:

[image]

Nonsense112, 2020-11-12:
Do you have the data for linkquality in state.json? After i modified the files and added the linkquality in state.json it also works for me. It seems that it pulls the LQI values from there.

Nonsense112, 2020-11-12:
It seems that the realtime LQI is 162, when you refresh the page it pulls an old value of LQI 89 from state.json (just a guess). I notice that on my setup, and when I deleted the old linkquality keys from state.json, they didn't get rewritten.

Koenkk, 2020-11-13:
I understand the issue now, fixed!

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Nonsense112, 2020-11-15:
Thank you!
It's working fine now.
However, can the changes I mentioned in the issue description be made permanent so that the linkquality and the last_seen are available in state.json? Maybe this will be useful for others.

Koenkk, 2020-11-15:
@Nonsense112 linkquality and last_seen should not be kept in the state.json (this was done before) however we now store these values in zigbee-herdsman (underlying zigbee library) instead of zigbee2mqtt.

Nonsense112, 2020-11-16:
Thank you, i understand.

closed 2020-11-16 by Nonsense112